raspberrypi/firmware #723: vc_dispmanx_resource_create returns valid handle even if no gpu memory is available

kapetanos, 2017-01-17:
I am trying to detect whether a resource can be allocated on the gpu memory or not (low gpu memory).
It seems that vc_dispmanx_resource_create always returns a "valid" handle even if there is no available gpu memory. If this "valid" handle is used to write_data to the resource the write_data call hangs forever.

You can have a look at this [gist](https://gist.github.com/kapetanos/4a2d195d07546a71be48ac79b33a1027) that demonstrates the issue. If the writing part is omitted then the code  "creates" up to 64 full hd resources even on a 32mb split !

Is there any other reliable way to detect if there is enough gpu memory for the resource?

popcornmix, 2017-01-17:
It does look like dispmanx_resource_create should be returning 0 when the resource fails to allocate. 
Can you test this firmware which should correct this:
https://drive.google.com/uc?id=0B-6zmEDJwxZEbzV0OTU0aWcyV0E&export=download

kapetanos, 2017-01-18:
It seems to be working as it should. When will this change make it to the official repo release?

popcornmix, 2017-01-18:
It will be in next rpi-update firmware (likely within the next week).
It will be in next apt-get firmware, but that's harder to predict. Likely within the next month or two.

kapetanos, 2017-01-18:
Great, thanks!

kapetanos, 2017-01-18:
Just an edge case I noticed on the new firmware. After using the same gist to create as much resources possible, the first dispmanx resource created is not freed when the process exits. This happens only if I hit the maximum resources possible. This seems to be happening once and not for every run.

For example:

```
# vcgencmd cache_flush; vcdbg reloc

Relocatable heap version 4 found at 0x37000000
total space allocated is 108M, with 108M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

free list at 0x3d401d20
100M free memory in 1 free block(s)
largest free block is 100M bytes

0x37000000: free 100M
[   4] 0x3d401d40: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3d401d60, d0rual) 'ILCS VC buffer pool'
[   3] 0x3d401f80: used 8.0M (refcount 1 lock count 8, size  8355840, align 4096, data 0x3d402000, d1rual) 'ARM FB'
[   2] 0x3dbfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3dbfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3dbfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3dbff000, d1rual) 'camera fast alloc arena'
small allocs not requested

# ./gist
...
...
...
Resource id: 1610612888
Resource: 25
gist: dispmanx.c:31: main: Assertion `resource' failed.
Aborted

# vcgencmd cache_flush; vcdbg reloc

Relocatable heap version 4 found at 0x37000000
total space allocated is 108M, with 108M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

free list at 0x3d401d20
96M free memory in 2 free block(s)
largest free block is 84M bytes

0x37000000: free 84M
[   5] 0x3c411c40: used 4.0M (refcount 1 lock count 0, size  4177920, align   32, data 0x3c411c60, d1rual) 'dispmanx_resource'
0x3c80dc80: free 12M
[   4] 0x3d401d40: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3d401d60, d0rual) 'ILCS VC buffer pool'
[   3] 0x3d401f80: used 8.0M (refcount 1 lock count 8, size  8355840, align 4096, data 0x3d402000, d1rual) 'ARM FB'
[   2] 0x3dbfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3dbfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3dbfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3dbff000, d1rual) 'camera fast alloc arena'
small allocs not requested
```

popcornmix, 2017-01-18:
Good spot. Actually unrelated to the first bug, but a failed allocation does overwrite an entry in the list of host allocated resources causing a memory leak when host app closes without releasing resources.
Can you test this: https://drive.google.com/uc?id=0B-6zmEDJwxZEQlQxU2ZxM0dsOFU&export=download

kapetanos, 2017-01-18:
I had a quick look and it seems it's resolved. Thanks again!

closed 2017-01-18 by kapetanos