raspberrypi/firmware #724: CM3 starts with UART0 disabled

lurch, 2017-01-18:
When I flash Raspbian Lite (2017-01-11) to a CM1 in my CMIOv3, and then reboot with a USB-TTL cable connected to GPIOs 14 and 15, I see the bootup messages both on a HDMI display and in a serial terminal window.

But when I try the same with the CM3, I see the bootup messages on a HDMI display but **not** in the serial terminal window. If I add `dtparam=uart0=on` to `/boot/config.txt`on the CM3 and reboot, then I *do* get bootup messages in the serial terminal.

According to https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L125 the default for uart0 should _already_ be on, so (taking a wild stab in the dark) I'm guessing that the firmware is detecting the BCM2837, and then disabling uart0 by default, just like it does for the RPi 3B. However for the CM3 wouldn't it make sense for the firmware to default to uart0 enabled, just like it does for the CM1?

lurch, 2017-01-18:
Haha, after a bit of reading around I've just found @pelwell confirming the same thing on the forums ;)
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=141248&start=50

Ruffio, 2017-03-16:
@pelwell is this 'working as designed' ?

pelwell, 2017-03-16:
Yes, as of https://github.com/Hexxeh/rpi-firmware/commit/48099e1634c0d845e6619f1a18df84b895370dd4.

Ruffio, 2017-03-16:
So I guess that it can be closed then...

pelwell, 2017-03-16:
That's usually left to the reporter of the issue...

lurch, 2017-03-16:
Ah sorry, I hadn't realised that this had been fixed in firmware. I'll update, test and close either this evening or over the weekend.

lurch, 2017-03-26:
Sorry for taking so long to get back to this! I can confirm that this now works as expected "out of the box" in the latest version of Raspbian Jessie Lite.

closed 2017-03-26 by lurch