raspberrypi/firmware #725: Limited memory on RPi3 is back again

DougieLawson, 2017-01-22:
https://github.com/raspberrypi/firmware/issues/649 is back.

```
root@challenger:~# free
             total       used       free     shared    buffers     cached
Mem:        117972      80216      37756       2372       7300      28432
-/+ buffers/cache:      44484      73488
Swap:       102396      47888      54508
root@challenger:~# cat /proc/meminfo
MemTotal:         117972 kB
MemFree:           37672 kB
MemAvailable:      32980 kB
```

```
root@challenger:~# vcgencmd get_mem arm
arm=128M
root@challenger:~# vcgencmd get_mem gpu
gpu=128M
root@challenger:~#
```

DougieLawson, 2017-01-22:
Linux challenger 4.9.4-v7+ #949 SMP Mon Jan 16 18:25:36 GMT 2017 armv7l GNU/Linux
That's with https://github.com/Hexxeh/rpi-firmware/commit/7c1a883209a9da9178f08b7cf189694efe49dadc

Legend2015, 2017-01-22:
I am having the same problem too. When I run rpi-update without the "BRANCH=next" it reverts to the right memory value and my system runs smooth again. But then i can't boot from my USB device anymore. :(

DougieLawson, 2017-01-22:
```BRANCH=master rpi-update``` which installs https://github.com/Hexxeh/rpi-firmware/commit/a12b699d642096023a2b3009bb88ca82ad0baef4 boots from USB for me and gets the right memory allocation.

popcornmix, 2017-01-22:
ping @ghollingworth

Legend2015, 2017-01-22:
For me it booted from USB after doing `BRANCH=master rpi-update` , but just once, after that it refused to boot again from USB. I loaded a previous version of the update from the next branch: `sudo BRANCH=next rpi-update 6c702d55e2637404f03cd6886cf8e8a30fd62eaa` and everything started working fine for me.

ghollingworth, 2017-01-22:
Is this reproducible?

Are you sure you didn't do something to break it between the two boots?

DougieLawson, 2017-01-22:
Nothing bad happened between the two boots (other than BRANCH=next rpi-update).

Give me five minutes and I'll try it again.

Legend2015, 2017-01-22:
In my case, after doing the standard rpi-update from the master branch, the Pi won't boot from the USB. I booted up using a working image from a SD card and then copied over start* , fixup* and bootcode.bin from the next branch on the boot folder of the USB and the Pi booted up from the USB but showed 120MB total memory again. That's when i tried `sudo BRANCH=next rpi-update 6c702d55e2637404f03cd6886cf8e8a30fd62eaa` and got back the full memory

DougieLawson, 2017-01-22:
```
root@challenger:~# uname -a; vcgencmd get_mem arm; vcgencmd get_mem gpu
Linux challenger 4.9.4-v7+ #949 SMP Mon Jan 16 18:25:36 GMT 2017 armv7l GNU/Linux
arm=128M
gpu=128M
root@challenger:~# uptime -s
2017-01-22 21:14:43
root@challenger:~# uptime
 21:16:31 up 1 min,  3 users,  load average: 3.06, 1.47, 0.55
root@challenger:~#
``` 

Still borken.

BenChud, 2017-01-22:
Same problem here - upgraded firmware and kernel from master to next branch on rpi2 and total memory drops to 256mb. Reverting firmware back to master and leaving kernel on next restores full 1gb.

ghollingworth, 2017-01-23:
Please try again with the newly released next branch firmware...

Legend2015, 2017-01-23:
Ok, just updated. Works fine. Thanks a lot for the quick turnaround.

DougieLawson, 2017-01-24:
Thanks, my RPi3 is booted OK from a USB stick running 4.9.5-v7+

Ruffio, 2017-01-26:
@DougieLawson so this issue can be closed?

ED6E0F17, 2017-02-08:
I have missing memory on Pi2 with mainline firmware:

Feb  8 2017 19:08:31 
Copyright (c) 2012 Broadcom
version d6903a3d6a6dbcc3c03e8ec6facb1af2883951c9 (clean) (release)

ED6E0F17, 2017-02-08:
Cancel that - I appear to be missing fixup.dat after upgrading.

popcornmix, 2017-02-09:
> EDIT: My mistake - it works a lot better when you download ALL the files......

Yes, ``fixup*.dat`` is used for implementing the arm/gpu memory split, so with that missing you just get a basic (reduced) memory split.

Closing as I believe this is fixed.

closed 2017-02-09 by popcornmix