raspberrypi/firmware #726: avoid_pwm_pll=1 changes mmc clock

portets, 2017-01-22:
Specifically,  dtparam=sd_overclock=100 with avoid_pwm_pll=1 changes the real mmc clock to 83.3.

pelwell, 2017-01-22:
The sdhost interface, which is what Pis now use to drive the SD card, doesn't use the MMC clock (I wish it did) - instead it has to make do with the system clock. What you have observed doesn't sound like a bug - it just suggests that your core frequency is 250MHz.

What other settings do you have, and why do you think you should be able to hit 100MHz?

portets, 2017-01-22:
So `/sys/kernel/debug/mmc0/ios 2>/dev/null | awk '{printf("%0.3f MHz", $3/1000000)}'` will report an inaccurate frequency with respect to the actual SD interface? With core_freq=500 and  dtparam=sd_overclock=100 it reports 100MHz without avoid_pwm_pll=1, and 83.3 with.

pelwell, 2017-01-22:
The sdhost clock is the same as the core clock - try measuring the core clock instead:
```
vcgencmd measure_clock core
```
The sdhost interface can only run at integer divisions of the core clock, so the requested speed is only a target (and never exceeded).

portets, 2017-01-22:
In both cases `vcgencmd measure_clock core` gives 500Mhz, but `sudo cat /sys/kernel/debug/mmc0/ios` gives an "actual clock" of 100MHz or 83.3MHz depending on `avoid_pwm_pll`. So it sounds like using `dtparam=sd_overclock=100` with `avoid_pwm_pll=1` is changing the the integer divisor from 5 to 6?

Running sd benchmarks seems to correlate. Read speeds are about 20% slower when adding `avoid_pwm_pll=1`, with no other config changes.

pelwell, 2017-01-23:
I don't see the same effect. Using a Pi3 with this config.txt:
```
pi@raspberrypi:~$ cat /boot/config.txt
enable_uart=1
dtparam=sd_overclock=100
core_freq=500
force_turbo=1
avoid_pwm_pll=1
```
I get:
```
pi@raspberrypi:~$ sudo cat /sys/kernel/debug/mmc0/ios
clock:          50000000 Hz
actual clock:   100000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
pi@raspberrypi:~$ vcgencmd measure_clock core
frequency(1)=500000000
```
Note that running with force_turbo=1 is not recommended, particularly without a heat sink, but in this case it is required in order to keep the core running at 500MHz (core_freq only sets the turbo frequency, and the Pi will only go into turbo mode when under load).

It's probably worth checking that your core isn't being throttled, either for voltage or thermal reasons. `vcgencmd get_throttled` should return `throttled=0x0` on a healthy Pi.

Finally, the "actual clock" displayed in the mmc0 metrics is the maximum of the turbo/non-turbo values - it is not updated when the Pi switches into and out of turbo so may not reflect reality, but your throughput measurements will not lie.

portets, 2017-01-23:
Huh, I can't seem to replicate it any more. User error maybe, not sure. force_turbo was always on so I don't know what it was. Although:
`core_freq=600
dtparam=sd_overclock=100
force_turbo=1
avoid_pwm_pll=1`
gives me an "actual clock" of 50MHz instead of 100MHz. And it reports core_freq not being throttled.

This issue can be closed otherwise.

pelwell, 2017-01-23:
600MHz is very high - many Pi's won't be able to achieve stability at that frequency.
Having said that, I did get mine to boot (with an over_voltage of 6) and found that actual_clock is still 100MHz.

These are the active config.txt settings I used:
```
enable_uart=1
dtparam=audio=off
dtparam=i2c_arm=on
dtparam=spi=on
core_freq=600
over_voltage=6
dtparam=sd_overclock=100
avoid_pwm_pll=1
force_turbo=1
```
And the results:
```
pi@raspberrypi:~$ sudo cat /sys/kernel/debug/mmc0/ios
clock:          50000000 Hz
actual clock:   100000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
pi@raspberrypi:~$ vcgencmd measure_clock core
frequency(1)=600000000
```

pelwell, 2017-01-23:
And for the record, `avoid_pwm_pll` is no longer recognised by the firmware, so it can't have caused whatever issue it was that you saw. I've removed it from our list of configuration settings.

closed 2017-01-23 by pelwell

popcornmix, 2017-01-23:
FYI the setting was removed here: https://github.com/Hexxeh/rpi-firmware/commit/260bc9c7589b3359485fc02fed8f56d4c5eaad9a