raspberrypi/firmware #727: dma-ranges property in device tree

kettenis, 2017-01-23:
I'm working in Raspberry Pi support for OpenBSD. One of the remaining issues I'm dealing
with is the dma address translation. While I can special-case the Pi in my code, I'd prefer
a more general solution. I noticed that the device tree in upstream Linux advertises the
address translation by including a "dma-ranges" property on the /soc simple-bus node:

    dma-ranges = <0xc0000000 0x00000000 0x3f000000>;

Would it be possible to include that node in the device trees that come with the firmware?

pelwell, 2017-01-23:
The `dma-ranges` property is already present in rpi-4.9.y builds (available now via `sudo BRANCH=next rpi-update`,  or you can browse the files here: https://github.com/raspberrypi/firmware/tree/next/boot), and we will be switching to the 4.9 kernel imminently.

kettenis, 2017-01-23:
Great. I'm hitting issue #725 with the new firmware, so I haven't been able to test this yet. But once that gets resolved, we should be in great shape.

closed 2017-01-23 by kettenis