raspberrypi/firmware #732: High HDMI pixel clocks (via hdmi_pixel_freq_limit) no longer working

Phlamethrower, 2017-01-28:
With the latest firmware, it seems to be impossible to make use of screen modes which require the pixel frequency limit to be increased (via the hdmi_pixel_freq_limit config.txt setting).

`vcgencmd get_config` shows that the custom limit is being read from the config file, but if I use hdmi_timings to select a screen mode which requires a pixel frequency which is higher than the default limit then no picture appears on my monitor. Custom modes which are lower than the default frequency limit work fine, and the same OS version works fine with older firmware. I've seen this with a Pi 1 and a Pi 3.

A bit of experimentation with different firmware versions suggests that it's commit https://github.com/raspberrypi/firmware/commit/3c3ff70bf9f8c0c8e4bc3fd7b5d08df53233ad4d which has caused the breakage.

My config.txt:

```
disable_overscan=1
kernel=RISCOS.IMG
framebuffer_swap=0
ramfsfile=CMOS
ramfsaddr=0x508000
max_framebuffer_width=2560
max_framebuffer_height=1440
hdmi_pixel_freq_limit=260000000
```

I'm performing a mode switch at runtime by setting hdmi_timings via vcgencmd and then switching mode using dispmanx (as per https://github.com/raspberrypi/firmware/issues/637), e.g. hdmi_timings of 2560 1 48 32 80 1440 0 3 5 33 0 0 0 59 0 241500000 3 no longer works. However a forum user looks like they've run into the same issue when setting a custom mode on boot via hdmi_cvt in config.txt: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=172057&p=1101111

popcornmix, 2017-01-31:
Can you test with latest rpi-update firmware? Hopefully this is fixed.

Phlamethrower, 2017-01-31:
Latest firmware seems to be working fine. Thanks!

closed 2017-01-31 by Phlamethrower