raspberrypi/firmware #733: vc.ril.image_encode:out:0 SUPPORTED_ENCODINGS may be incorrect

waveform80, 2017-02-02:
I've been trying to make the MMAL component connection logic in picamera a bit more intelligent by having it query (and use) the supported encodings on ports that it's connecting (in older firmwares this isn't feasible as several ports raised MMAL_EINVAL upon querying this parameter, but in newer firmwares this all seems to be sorted out so the next release of picamera uses it when possible and substitutes hard-coded tables when it's not).

Unfortunately, one of the SUPPORTED_ENCODINGS lists looks ... a little strange. In the `vc.ril.image_encode` component I'd expect to see things like JPEG, GIF, and PNG supported on the output, but querying it reveals (using the current HEAD of picamera):

```python
>>> from picamera import mmalobj as mo, mmal
>>> enc = mo.MMALImageEncoder()
>>> [mmal.FOURCC_str(f) for f in enc.inputs[0].supported_formats]
['RGB2', 'RGB3', 'RGBA', 'BGRA', 'I420', 'I422', 'YUYV', 'YVYU', 'VYUY']
>>> [mmal.FOURCC_str(f) for f in enc.outputs[0].supported_formats]
['MP2V', 'MP2V', 'H264', 'H264', 'VP7 ', 'VP7 ', 'VP6 ', 'VP6 ']
```

The input port looks reasonable (some RGB formats, some YUV formats) but the output port is weird. Each encoding's doubled up and they're all video encodings? I'll include a hard-coded override for this in the next picamera release but it'd be nice if the underlying firmware could report the right things at some point (no urgency; I'll have to include an override anyway for people on older firmwares).

For reference the firmware on the box above is:

```
Dec  9 2016 15:15:20 
Copyright (c) 2012 Broadcom
version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)
```

6by9, 2017-02-02:
Almost certainly the framework isn't using the right IL domain.

The list of formats comes back as either a enum OMX_IMAGE_CODINGTYPE (https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/khronos/IL/OMX_Image.h#L54), or an enum OMX_VIDEO_CODINGTYPE (https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/khronos/IL/OMX_Video.h#L62), both of which are zero based.
Comparing them JPEG=MPEG2=2, PNG=H264=7, TGA=VP6=0x7F000001, PPM=VP7=0x7F000002.

The duplication appears to be due to the way the list of supported encodings works within that component. JPEG is listed twice as it can take YUV420 or YUV422 natively. PNG and TGA are listed twice as it takes RGB24 or RGBA32. PPM is listed twice as it can take RGB24 or RGB565. I'd need to look to see why GIF and BMP aren't listed, as they should be.
Again it shouldn't be a big job to remove the duplicates.

I'll dig into this tomorrow.

6by9, 2017-02-03:
My guess was right. I have a fix that will get pushed out at some point soon (probably early next week).
The list of supported settings I now get is
```
Input port
mmal: : Supported encoding 0 is RGB2
mmal: : Supported encoding 1 is RGB3
mmal: : Supported encoding 2 is BGR3
mmal: : Supported encoding 3 is RGBA
mmal: : Supported encoding 4 is BGRA
mmal: : Supported encoding 5 is I420
mmal: : Supported encoding 6 is I422
mmal: : Supported encoding 7 is YUYV
mmal: : Supported encoding 8 is YVYU
mmal: : Supported encoding 9 is VYUY
mmal: : Supported encoding 10 is UYVY

Output port
mmal: : Supported encoding 0 is GIF 
mmal: : Supported encoding 1 is BMP 
mmal: : Supported encoding 2 is JPEG
mmal: : Supported encoding 3 is PNG 
mmal: : Supported encoding 4 is PPM 
mmal: : Supported encoding 5 is TGA 
```
There was an error on the input list as well, as reading the wrong domain for the output port.

waveform80, 2017-02-03:
Excellent news! Many thanks for the quick fix :)

6by9, 2017-02-08:
Fix released if you do ```sudo rpi-update```.
Don't do this on a critical machine - that is the cutting edge branch so may have issues.

6by9, 2017-10-10:
@waveform80 If you're happy would you close the issue please?

waveform80, 2017-10-10:
Sure thing :)

closed 2017-10-10 by waveform80