raspberrypi/firmware #735: Support Open WF composition api's in firmware

tarkshya, 2017-02-07:
Currently open WF api's are not supported in firmware.  In the file 

https://github.com/raspberrypi/userland/blob/master/interface/khronos/wf/wfc_client_ipc.c

inside this function
VCOS_STATUS_T wfc_client_ipc_init(void);

the following call 

`vchiq_status = vchiq_open_service(wfc_client_ipc_vchiq_instance, &vchiq_params, &wfc_client_ipc.service);`

returns failure.

vchiq_open_service in turn calls create_service (on the userland side)

Debugging the VCHIQ driver shows that create_service (kernel side) fails, probably because the service either being not present or not enabled in the firmware.

Thanks
Tarkshya

popcornmix, 2017-02-07:
This is not an API we support.

tarkshya, 2017-02-07:
Just out of curiosity, why its not supported? Any plans to support them in future?

this -->
https://www.raspberrypi.org/forums/viewtopic.php?t=5621

The above link states that dispmanx is being deprecated in favour of khronos libs, OpenWF implementation is there, not sure how much complete though. So seems the intention is there.

Could you let us know (if possible) about the timelines?

popcornmix, 2017-02-07:
Well the commend was from nearly 5 years ago and said:
"Yes, I think OpenWF is what will be used, but don't quote me on it."

But that is not what happened. OpenWF didn't become a popular standard and no one uses it on the Pi.

tarkshya, 2017-02-07:
Another question, does supporting it in the firmware take a big effort, why not simply enable the service in firmware and allow other developers to figure it out (on how to work with it) ?

popcornmix, 2017-02-08:
The service is enabled. It was part of the firmware tree for a previous Broadcom device.
But it has never been used on Raspberry Pi so my guess is it won't work.
But feel free to try using it, but if it doesn't work we won't go to great lengths to fix it.

tarkshya, 2017-02-11:
did a basic testing with the following code (actually modified the vchiq_test.c)

```

#define WFC_IPC_CONTROL_FOURCC()   VCHIQ_MAKE_FOURCC('W','F','C','I')  
#define WFC_IPC_VER_CURRENT     8
#define WFC_IPC_VER_MINIMUM     5

static VCHIQ_STATUS_T
func_clnt_callback(VCHIQ_REASON_T reason, VCHIQ_HEADER_T *header,
   VCHIQ_SERVICE_HANDLE_T service, void *bulk_userdata)
{
return VCHIQ_SUCCESS;
}

int main()
{
   VCHIQ_INSTANCE_T instance;
   VCHIQ_SERVICE_HANDLE_T service;
   VCHIQ_SERVICE_PARAMS_T service_params;
   VCHIQ_STATUS_T status;

   status = vchiq_initialise(&instance);
   
   if (status != VCHIQ_SUCCESS)
      return 0;

   status = vchiq_connect(instance);

   if (status != VCHIQ_SUCCESS)
      return 0;
  
   memset(&service_params, 0, sizeof(service_params));
   service_params.fourcc = WFC_IPC_CONTROL_FOURCC();
   service_params.callback = func_clnt_callback;
   service_params.userdata = (void*)1;
   service_params.version = WFC_IPC_VER_CURRENT;
   service_params.version_min = WFC_IPC_VER_MINIMUM;

   VCHIQ_STATUS_T status = vchiq_open_service(instance, &service_params, &service);
   
   if (status != VCHIQ_SUCCESS)
   {
      vcos_log_error("%s: could not open vchiq service: %d", VCOS_FUNCTION, vchiq_status);
      return VCHIQ_ERROR;
   }

  return 0;
}
```

I am getting VCHIQ_ERROR as the status for vchiq_open_service, previous calls to vchiq_XXXX are all successful. Could you tell me why this is happening? got the VER_CURRENT, VER_MINIMUM, FOURCC form the OpenWF userland code.

Electron752, 2017-02-17:
@tarkshya - Out of curiosity, what are you trying to accomplish in general?  Just curious because if you are trying to create a general userland to userland IPC mechanism, it seems to me that Linux already has solutions for this.  You can use pipes, unix sockets, shared memory(if I understand correctly), or even IP sockets.  Are you trying to support kernel to userland messaging?

What what I understand of vchiq, nothing in the driver itself would prevent short circuiting the firmware and redirect the messages to another userland client.  I just don't completely understand what the purpose of doing that would be since other mechanisms already exist.

tarkshya, 2017-02-18:
@Electron752 nothing to do with linux IPC. This is related to the Open Windowing foundation composition library for displaying things on the screen. Raspi has a userland library for this, they had plans to replace dispmanx with this, but probably it didnt make it. Open WF is (was) mostly used in low power systems, not common in full blown OS. The userland library communcates with the linux VCHIQ driver. Here that is not working (for OpenWF), trying to get it to work. Waiting for the experts to remove my ignorance regarding its functionality.

pelwell, 2017-02-20:
I'd be lying if I said there was no WFC-related code in the firmware, but from what I can see those bits are used to implement EGL. There is no host-facing API (i.e. no VCHIQ server), which is why your test returns an error and you should stop trying.

As @popcornmix said previously, "This is not an API we support." You may not have noticed, but there is a gradual shift to running more graphics code on the ARM, as spearheaded by Eric Anholt's VC4 driver. This is because the ARMs are more powerful than the VPU, and by doing so we can reduce the reliance on closed-source firmware code and perhaps improve performance, reliability and features at the same time.

tarkshya, 2017-02-20:
@pelwell Thanks for the confirmation. I am aware of Eric Anholt's VC4 driver, which is DRI based and not particularly suitable for bare metal (or non-linux based OS's), unless the core (VPU stuff of eric's code) is separated out (am i misunderstanding something?). 

It all started when i though of studying/porting the VCHIQ driver to bare metal and comparatively seems less work if we could get the VCOS layer, threads sorted out (the original VCHIQ android source reveals some info related to that). Once thats done, userland should be fairly straightforward. Hence my experiments.

pelwell, 2017-02-20:
The VC4 reference wasn't a suggestion, just an indication of the fact that you are swimming against the current.

Good luck.

closed 2017-02-20 by pelwell

popcornmix, 2017-02-20:
The dispmanx API is widely used and unlikely to be removed completely, so that may be something to consider.
Otherwise you have a framebuffer you can fill with pixels. Any sort of composition library can be run on the arm side.

tarkshya, 2017-02-20:
@pelwell  learning how to swim the hard way :-)

@popcornmix had some success with that - composition with raw framebuffer, thought of getting the GPU to do it, hence the enthusiasm.