raspberrypi/firmware #736: Failed to boot when display_rotate=1 or display_rotate=3

hxhxhx88, 2017-02-07:
I have updated to the latest firmware using 
    sudo rpi-update

If I set display_rotate=1 or display_rotate=3, the booting process will stop with nothing shows on the screen. 

If I set display_rotate to other values, such as 2 or 0x10000, the pi can boot, BUT the display will not be affected, i.e. they all behave like display_rotate=0!

Followings are results of running some commands:

tvservice -s:
    state 0x120009 [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

fbset:
    mode "1920x1080"
        geometry 1920 1080 1920 1080 32
        timings 0 0 0 0 0 0 0
        accel true
        rgba 8/16,8/8,8/0,0/0
    endmode

My config.txt reads:

# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# display_rotate=1

# uncomment if you get no picture on HDMI for a default "safe" mode
# hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
# disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
# hdmi_force_hotplug=0

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
# config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
dtoverlay=lirc-rpi,gpio_in_pin=23,gpio_out_pin=22

# Additional overlays and parameters are documented /boot/overlays/README

start_x=1
gpu_mem=128

# Enable audio (added by raspberrypi-sys-mods)
dtparam=audio=on
dtoverlay=vc4-kms-v3d

Please help!

popcornmix, 2017-02-07:
You can't use ``display_rotate`` with ``dtoverlay=vc4-kms-v3d``
If using vc4-kms-v3d you should be able to get a similar behaviour with xrandr

hxhxhx88, 2017-02-07:
@popcornmix it works, thanks!!!

closed 2017-02-07 by popcornmix