raspberrypi/firmware #738: Wifi unstable after 4.4.39

djtm, 2017-02-09:
Compiling the kernel from source, the Wifi has become unstable on the Rpi3b after
kernel 4.4.39(-v7+)
Hardware	: BCM2709
Revision	: a02082

Any hints for debugging?

pelwell, 2017-02-09:
Can you elaborate on the instability?

Do you have a commit ID where the trouble began? git bisect makes this less painful than you might think.

djtm, 2017-02-09:
The wifi goes off and does not reconnect. I can still use kodi, but without network.

The painful part is that it takes a while for the system to lose the wifi connection each time. That's why it took me so long to find the issue. It appeared between hours and a day or two.

popcornmix, 2017-02-09:
I assume this is built in wifi on Pi3?
Report the most recent commit you believe to be working well and the oldest commit you believe to be working badly.
Test the commit in between these two and test until you are happy if it is good or bad.

pelwell, 2017-02-09:
```
git bisect start
git bisect good <hash of recent known good commit>
git bisect bad <hash of failing commit>
[... git selects a new commit ]
build and test
if it works:
  git bisect good
if it doesn't work
  git bisect bad
[... git selects another commit, or exits and tells you which commit broke it ]
git bisect reset  # Return to previous HEAD
```

djtm, 2017-02-14:
Happily, the problems disappeared again in the most recrent kernel version.

djtm, 2017-02-14:
Ha, and then it was gone again.... :(
See, the problem is that there is a day or more between two bisects. So it would still take a very long time... It would probably make more sense to manually look for wifi-related changes.

JamesH65, 2017-03-23:
Is this still an issue? 
Does dmesg say anything about a kevent being dropped?

djtm, 2017-03-27:
I had not noted the interface is inside a bridge. I've recently compiled the newest git kernel. I'm still waiting for the first issue there.

I get these errors.
```
Mar 22 22:26:23 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Wed Mar 22 22:27:53 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 22 22:26:23 raspberrypi kernel: [1376716.045889] bond0: link status definitely down for interface wlan0, disabling it
```

I do get a lot of this in general:
`wpa_supplicant[563]: wlan0: WPA: EAPOL-Key Replay Counter did not increase - dropping packet`

JamesH65, 2017-04-20:
Is this still an issue? Does this issue seem to be relevent/similar #673? Do you have the ability to test the patch in that issue?

djtm, 2017-04-21:
Yes it is. Thank you I'll check that! :+1:

djtm, 2017-05-07:
The patch that applied didn't really help much, unfortunately.

I'm starting to think it might be another issue entirely, though. It's so tough to debug.

JamesH65, 2017-05-08:
OK, thanks for the report, I am fairly sure there are other issues in the wifi stack somewhere, its finding a way to easily replicate the issue that is difficult! Your dmesg log is different o others received - are you doing anything unusual in your setup?

djtm, 2017-05-11:
Not sure.

Pi v. 3
runs from USB2 HD
previously attached to ethernet
once HD attached only works with Wifi.
Services include apple timemachine, dlna, syncthing.

Right now I have a feeling it might be power related after all. I've had a long stable period by disabling the hard drive's power saving features with hdparm. I have a feeling the problem might occur when the drive power back up. How that's related to the change in the kernel, I'm not sure.

pelwell, 2017-05-11:
Try `vcgencmd get_throttled` - if it returns anything other than zero then your Pi is either underpowered or overheating.

djtm, 2017-05-11:
Hm, that's not it, either, right now: `throttled=0x0`
However my theory is that the HD spins down or goes to standby, then gets reactivated, there is a power spike, this power spike causes wifi to disconnect. And wifi generally does not reconnect for me.

I've noticed this recently:
```
dmesg|grep -C 3 -i warn
126-[    0.761059] Multiprocessor Interrupt Enhancement - disabled
127-[    0.763228] OTG VER PARAM: 0, OTG VER FLAG: 0
128-[    0.765429] Dedicated Tx FIFOs mode
129:[    0.767907] WARN::dwc_otg_hcd_init:1053: FIQ DMA bounce buffers: virt = 0xba904000 dma = 0xfa904000 len=9024
130-[    0.772352] FIQ FSM acceleration enabled for :
131-Non-periodic Split Transactions
132-Periodic Split Transactions
133-High-Speed Isochronous Endpoints
134-Interrupt/Control Split Transaction hack enabled
135-[    0.783704] dwc_otg: Microframe scheduler enabled
136:[    0.783752] WARN::hcd_init_fiq:459: FIQ on core 1 at 0x8058e4c4
137:[    0.786054] WARN::hcd_init_fiq:460: FIQ ASM at 0x8058e834 length 36
138:[    0.788372] WARN::hcd_init_fiq:486: MPHI regs_base at 0xbb87a000
139-[    0.790720] dwc_otg 3f980000.usb: DWC OTG Controller
140-[    0.793033] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
141-[    0.795389] dwc_otg 3f980000.usb: irq 62, io mem 0x00000000
```

djtm, 2017-05-11:
These are voltage, temperature, and clock statistics:
[image]
 statistics from last week.
[image]
[image]

djtm, 2017-05-11:
Unusual might be that I'm using channel bonding since ethernet stopped working.

JamesH65, 2017-05-12:
What do you mean by ethernet stopping working?

djtm, 2017-05-13:
Ha. For some reason ethernet was at least unreliable for a while. So I switched to wifi. Then I thought, why dont bond the two. Might even speed things up. Now, after further debugging, I found out that the RPi3's ethernet just doesn't work with bonding. Only wifi worked inside bonding.

Now I've disabled bonding again and I'm back to a working ethernet. Let's see how this goes. :)

iurly, 2017-08-31:
@djtm Were you able to track down this issue in the end?
I'm having the exact same issue (it disconnects after a day or two) and does not automatically reconnect.
See my post on the forum:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=183683&p=1162692#p1162692

djtm, 2017-08-31:
> 183683&p=1162692#p1162692

Sorry. I switched to Ethernet with a script in cron to check and reconnect
on disconnect.

iurly, 2017-08-31:
I see. My idea for a workaround was also to use monit to check availability and trigger a scan on wpa_cli.

JamesH65, 2018-06-29:
Please update to the latest kernel which may contain a fix for this issue.

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-06-29 by JamesH65

djtm, 2018-07-07:
I think this was a power issue.