raspberrypi/firmware #739: recent firmwares turn on TV every 10 minutes

nullr0ute, 2017-02-09:
There seems to be a recent regression where the TV turns on automatically every 10 mins presumably by a CEC call. 

The TV is a Sony KD-43X8309C with a Raspberry Pi 3 connected. 

Firmware dated "2016-12-09 15:12" (git 6d45dcf) works fine
Firmware dated "2017-01-31 17:41" (git 72b44d8) turns TV on every 10 mins or there abouts. Latest firmware from 8th Feb (git db5fd5e) has the same issues, 

I've not tried any firmware between Dec 9th and Jan 31st but there was CEC changes on Dec 30  which were reverted on Jan 4th, and then some more on the 14th and a few since then.

popcornmix, 2017-02-09:
Please test the firmware prior to https://github.com/raspberrypi/firmware/commit/72b44d850dc8c2307cf0dccea00928702e16bc12
You can get it with:
```
sudo rpi-update bb35a389edd88e96f7a587ecb34ef5bf9167bcab
```
I'd like to know if CEC cleanups 11-12 are responsible.

When switching firmware please power off TV from the mains for a minute before testing as it may remember state from the previous build.

nullr0ute, 2017-02-09:
This is on Fedora, I'll test that firmware. I normally turn the TV off and unplug/replug the cable but moving back to the Dec 9th firmware fixes it every time.

nullr0ute, 2017-02-11:
The "2017-01-27 19:12" b846de8 firmware works as expected with the TV staying off so it does look like the CEC cleanups 11-12 are responsible

popcornmix, 2017-02-11:
Could you test this firmware which includes cleanup 11 but no 12.
https://drive.google.com/uc?id=0B-6zmEDJwxZENXJoaEpacVNROWM&export=download

nullr0ute, 2017-02-11:
That seems fine with about 20 mins of testing

nullr0ute, 2017-02-12:
I can confirm that firmware is good

popcornmix, 2017-02-12:
Thanks. Useful information. I'll check with CEC guy tomorrow what to do next.

popcornmix, 2017-02-13:
CEC is off for the next couple of days so I've reverted CEC cleanup 12 so you should be able to switch back to latest rpi-update firmware.
When CEC guy is back we may have a new firmware to test.

nullr0ute, 2017-02-13:
One thing I did notice just now with that test firmware is that the "raspberry" name isn't displayed on the TV when I change inputs. I can re-test the pre 11/12 firmware to see if it's there, or the newer firmware too. Just thought I'd note the difference.

LeaseOnLife, 2017-02-13:
I understand the reversal is temporary. As a note though, reversing CEC cleanup 12 killed my VGA display (using a HDMI-VGA converter): Black screen during boot, and wrong display resolution and timing when X is running. As it was before the CEC cleanup. I originally reported it here: https://github.com/anholt/linux/issues/56

popcornmix, 2017-02-14:
> One thing I did notice just now with that test firmware is that the "raspberry" name isn't displayed on the TV when I change inputs

Can you test the preceding firmware to double check this?
We should report the OSD name as "raspberry" with either firmware, but the timing will change in cleanup 12. It sounds like without 12 the cec is not working as intended, as possibly with 12 it is working as intended (but your TV is doing something unwanted).

Can you grab a CEC with each of the two firmwares? Add ``logging_level=0x230000`` to config.txt and reboot. You can grab the log with:
```
sudo vcdbg log msg 2>&1 | pastebinit
```
And post the links here. Leave the Pi long enough to do the bad thing (i.e. power on the TV) before grabbing the log.

Power off TV from mains for a minute after switching firmware to make sure it is not remembering state form the previous firmware.

popcornmix, 2017-02-20:
@nullr0ute Any chance of grabbing the logs requested?

Ruffio, 2017-04-03:
@nullr0ute has this been fixed and can be closed?

nullr0ute, 2017-04-03:
Yes, I believe so, I was testing a firmware from mid march over the weekend and it appears to be good now. Was actually on my list to update this today once I checked a few things.

Ruffio, 2017-04-03:
@nullr0ute can you close it then?

nullr0ute, 2017-04-03:
Sorry, didn't know if there was a specific process used

closed 2017-04-03 by nullr0ute