raspberrypi/firmware #740: latest start_x.elf fixup_x.elf only support mem up to 256MB?

jeffersonchua, 2017-02-10:
Setting these two parameters in /boot/config.txt caused mem to become 256MB on pi3. It used be 1GB on 4.8.0 ... didn't realized till now ... somewhere after 4.8.0, it became 256MB.

start_file=start_x.elf
fixup_file=fixup_x.elf

Jeff.

pelwell, 2017-02-10:
Try:
```
start_file=start_x.elf
fixup_file=fixup_x.dat
```
(Note the file extension on the second line)

Better still, delete both lines and replace them with:
```
start_x=1
```

jeffersonchua, 2017-02-11:
Phil,

> fixup_file=fixup_x.dat

> (Note the file extension on the second line)

Silly me. Sorry, didn't see that!

> Better still, delete both lines and replace them with:

> start_x=1

That worked.

Thank you for spotting the mistake!

Jeff.

closed 2017-02-11 by popcornmix