raspberrypi/firmware #741: Odd audio_render (to local) interaction with GPIO pin 1,23,24 and 26

dividuum, 2017-02-10:
I've been informed that [my software](https://info-beamer.com/pi) has a problem playing a video with audio track once GPIO 1, 23, 24 or 26 is switched to pwm mode (gpio mode $pin pwm). I'm using `ilclient.c`. It seems like the `EmptyBufferDone` callback for the `audio_render` component is called for the first ~14 buffers submitted but then stops. The audio gets stuck and the video waits and gets stuck as well. This only happens then I'm rendering to `local` (instead of `hdmi`), so I guess there's something going on behind the scenes that I don't understand. Any hints of what's going on?

Right now I don't have a smaller test case.

```
Pi3

Dec  9 2016 15:13:48 
Copyright (c) 2012 Broadcom
version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)

Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
```

pelwell, 2017-02-10:
Those GPIO numbers don't make sense either has Broadcom GPIO numbers or as header pins - can you choose one of those numbering schemes (I'd prefer the former).

dividuum, 2017-02-10:
Sorry, I've absolutely no experience with using the GPIOs at all. I don't even have any hardware to connect to them. I just got the report that the problem is triggered once `gpio mode 1 pwm` is run and I noticed that it also happend for 23, 24 and 26. What's the correct terminology in that case? From `gpio` it seems like `pin`? 
```
root@raspberrypi:~# gpio mode
Usage: gpio mode pin mode
```

I found [this picture](http://wiringpi.com/wp-content/uploads/2013/03/gpio1.png). Assuming I'm using the WiringPi PIN number scheme, it looks like that ends up being BCM GPIO 18?

pelwell, 2017-02-10:
We recommend raspi-gpio - it uses the Broadcom numbering scheme, and should be installed by default; if not, `sudo apt-get install raspi-gpio`.

dividuum, 2017-02-10:
Thanks. I'll try that later and see if that indeed maps to GPIO 18.

pelwell, 2017-02-10:
Can you get your Pi into a failing state then upload the output from `raspi-gpio get`?

dividuum, 2017-02-10:
Will do. Thanks.

dividuum, 2017-02-10:
```
root@raspberrypi:~# raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=0 alt=  func=INPUT
  GPIO 03: level=1 fsel=0 alt=  func=INPUT
  GPIO 04: level=0 fsel=0 alt=  func=INPUT
  GPIO 05: level=0 fsel=0 alt=  func=INPUT
  GPIO 06: level=0 fsel=0 alt=  func=INPUT
  GPIO 07: level=0 fsel=0 alt=  func=INPUT
  GPIO 08: level=0 fsel=0 alt=  func=INPUT
  GPIO 09: level=0 fsel=0 alt=  func=INPUT
  GPIO 10: level=0 fsel=0 alt=  func=INPUT
  GPIO 11: level=0 fsel=0 alt=  func=INPUT
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=0 fsel=0 alt=  func=INPUT
  GPIO 15: level=1 fsel=0 alt=  func=INPUT
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=2 alt=5 func=PWM0
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=0 alt=  func=INPUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=0 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=0 fsel=4 alt=0 func=PWM0
  GPIO 41: level=0 fsel=4 alt=0 func=PWM1
  GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
  GPIO 43: level=0 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=1 fsel=0 alt=  func=INPUT
  GPIO 45: level=1 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=4 alt=0 func=SDA0
  GPIO 47: level=1 fsel=4 alt=0 func=SCL0
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
```

As suspected, it turns GPIO 18 to PWM.

pelwell, 2017-02-10:
Assuming that mapping PWM0 to GPIO18 is a static configuration, you could try unmapping PWM0 from GPIO40 `raspi-gpio set 40 ip` to see if it avoids the problem. There are some functions (I2C, for example) that don't work if multiply-mapped, but it's not immediately obvious why PWM might be one of those.

dividuum, 2017-02-10:
That does indeed help. Now there is only a single GPIO with PWM0 and GPIO 40 looks like this:

```
  GPIO 40: level=0 fsel=0 alt=  func=INPUT
```

Playback works again with `local` output. Any idea how all of that might be related?

pelwell, 2017-02-10:
None of the software is aware which pins are being driven, so although I don't know enough to say why mapping the PWM pins twice is bad, the empirical evidence is strong that it is something to be avoided.

The location of the audio output pins is controlled by Device Tree (the board-specific .dtb file). The audremap overlay moves the audio pins to GPIOs 12 and 13, but it could be parameterised and extended to also cover GPIOs 18 and 19. 

The appropriate action by the users may depend on why they are moving PWM0 elsewhere - are they wanting to use PWM0 for something other than sound? But an immediate fix for them would be to unmap PWM0 from its original location (and presumably PWM1 if they are also moving that). I would recommend using an overlay to perform the remapping, but that's up to them.

dividuum, 2017-02-10:
I think that information helps a lot. Thanks for that. Closing for now.

closed 2017-02-10 by dividuum