raspberrypi/firmware #742: CM3 ignores IO expander pull settings in dt-blob

rps102, 2017-02-14:
Creating a ticket so this topic has a number to refer to:
  https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=174059

The two issues are
* The defaults pull settings in the firmware conflict with the discrete resistor on IO0 of FXL6408
* Adding termination="pull_up" or termination="pull_down" for any of the pins on the IO expander has no effect, should modify defaults in register 0xB and 0xD

Propose defaults should be
  reg 0x3 = 0x02, reg 0xB = 0xFC
so all 6 unused inputs are pulled to valid logic level, with no pull downs on IO0 and IO1.

6by9, 2017-02-14:
I have a fix already that applies the termination as specified on the blob. There was an rpi-update yesterday, so I'll be rolling up all my work (including that patch) and pushing internally in a couple of days time to pick up the next one.
If you're really lucky it may include the mailbox service to set direction/polarity/termination dynamically as well.

rps102, 2017-02-14:
Wow - that was quick. It took me nearly as long to remember my GitHub password!
Cheers,
Sprow.

6by9, 2017-02-15:
Latest rpi-update firmware should solve this.

It also includes the mailbox update, and I've just pushed an updated app to https://github.com/6by9/rpi3-gpiovirtbuf that adds a 'c' command.
```rpi3-gpiovirtbuf c <GPIO> <DIR> <ACT_LOW> <TERM_EN> <PULL_UP> <STATE>```
DIR: 0=in, 1=out.
POLARITY: 0=active high, 1=active low (inverted).
TERM_EN: 0=no termination, 1=termination enabled
PULL_UP: 0=pull down, 1=pull up. Only if TERM_EN=1 and DIR=0
STATE: state to set output to. Only if DIR=1

rps102, 2017-02-18:
Having tested it, the new message seems to do as your gpiovirtbuf example illustrates.

I notice the response code for RPI_FIRMWARE_SET_GPIO_CONFIG is always 0, indicating it returns nothing (unlike some of the other property SETters). RPI_FIRMWARE_GET_GPIO_CONFIG does set b31 of the response code, as expected.

closed 2017-02-18 by rps102

6by9, 2017-02-18:
> I notice the response code for RPI_FIRMWARE_SET_GPIO_CONFIG is always 0, indicating it returns nothing (unlike some of the other property SETters). RPI_FIRMWARE_GET_GPIO_CONFIG does set b31 of the response code, as expected.

I'd need to check with @popcornmix on that. The mailbox code was a little cryptic on what it was returning and when/why/how.

popcornmix, 2017-02-19:
@rps102 Can you post what words you are submitting to mailbox interface and what words are returned?