raspberrypi/firmware #743: Firmware/VCHIQ cache line size negotiation issue

Electron752, 2017-02-17:
I've been toying with the idea of getting vchiq to load without a device tree node.   I have some code that isn't ready to share out that treats the firmware as a virtual bus that other drivers get loaded onto instead of a pure device tree.   The idea would be to make things more like PCI on PCs or USB.  The firmware is queried to configure things rather then the device tree.  The current drivers I'm looking at are vchiq, sound, and thermal.

At this point, one issue I'm finding is that the cache line size isn't being detected properly by the firmware if the vchiq node is missing from the device tree.  On a RPI 3, the correct cache line size is 64.   If I hardcode in 64, vchiq_test -p, vchiq_test -c, and vchiq_test -b all work.  But vchiq_test -f doesn't work due to data corruption issues. I believe the issue is that if the device tree node is missing, then the firmware is assuming the device is a RPI 2 or earlier.

1. Is this a known issue with the firmware if the device tree node is missing?
2. Does the concept of loading drivers based on firmware properties have any appeal?   At the moment, I'm mostly looking at how to get vchiq to load in pure upstream without needing to document the device tree settings and go through the process of pretending the driver is multi-platform.

pelwell, 2017-02-17:
> Is this a known issue with the firmware if the device tree node is missing?

Yes, this is expected behaviour. An DTB which doesn't include the cache line size property is assumed to belong to a kernel with a vchiq driver that doesn't recognise the property, and hence is using the value of 32 at all times (which appears to work in most cases even when it is wrong...)

> Does the concept of loading drivers based on firmware properties have any appeal?

The idea of inventing a new mechanism at a time when we are getting quite close to upstream doesn't immediately appeal.

> At the moment, I'm mostly looking at how to get vchiq to load in pure upstream without needing to document the device tree settings

If that level of documentation is unattractive, think about the documentation required for a DT-replacement.

Electron752, 2017-02-17:
That makes sense.  Maybe it isn't as difficult to get the docs in for the device tree as I was told.  Anyway, it's good I asked.

Electron752, 2017-02-27:
The concept of loading these firmware based drivers without a DT has been given the green light in upstream.  I was wondering if it would be possible to get the firmware fixed to use the correct cache size on the RPI 3 even if the vchiq node is not in the tree?

https://github.com/raspberrypi/firmware/issues/755

Ruffio, 2017-03-16:
@pelwell what you say?

Electron752, 2017-03-16:
Actually, I'm still thinking about this.  I'm wondering if making a change like this makes sense anyway.  Perhaps it's better just to leave things alone for awhile.

closed 2017-03-25 by Electron752