raspberrypi/firmware #745: Invalid destination MAC address in TFTP RRQ

vvg77ripe, 2017-02-18:
When booting from network, RPi 3 successfully loads bootcode.bin from TFTP server, then bootcode is attempting to load other files, but with zero destination mac address, and RRQ packets have not been received by TFTP server.

I've tried last master version of bootcode.bin, last "next" and some older versions with no luck.

There is a dump:
11:59:07.446234 b8:27:eb:f9:92:f7 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 10.23.16.13 tell 10.23.16.10, length 46                                                                                                                                   
11:59:07.446251 38:d5:47:1c:aa:34 > b8:27:eb:f9:92:f7, ethertype ARP (0x0806), length 42: Reply 10.23.16.13 is-at 38:d5:47:1c:aa:34, length 28                                                                                                                                      
11:59:07.446375 b8:27:eb:f9:92:f7 > 38:d5:47:1c:aa:34, ethertype IPv4 (0x0800), length 63: 10.23.16.10.49152 > 10.23.16.13.69:  21 RRQ "bootcode.bin" octet                                                                                                                         
11:59:07.469377 b8:27:eb:f9:92:f7 > 38:d5:47:1c:aa:34, ethertype IPv4 (0x0800), length 62: 10.23.16.10.49152 > 10.23.16.13.69:  20 RRQ "bootsig.bin" octet                                                                                                                          
11:59:09.589113 b8:27:eb:f9:92:f7 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 71: 10.23.16.10.49154 > 10.23.16.13.69:  29 RRQ "autoboot.txt" octet tsize 0                                                                                                                 
11:59:11.155156 b8:27:eb:f9:92:f7 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 69: 10.23.16.10.49155 > 10.23.16.13.69:  27 RRQ "config.txt" octet tsize 0                                                                                                                   
11:59:12.155515 b8:27:eb:f9:92:f7 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 71: 10.23.16.10.49156 > 10.23.16.13.69:  29 RRQ "recovery.elf" octet tsize 0                                                                                                                 
11:59:12.451694 38:d5:47:1c:aa:34 > b8:27:eb:f9:92:f7, ethertype ARP (0x0806), length 42: Request who-has 10.23.16.10 tell 10.23.16.13, length 28                                                                                                                                   
11:59:12.451806 b8:27:eb:f9:92:f7 > 38:d5:47:1c:aa:34, ethertype ARP (0x0806), length 60: Reply 10.23.16.10 is-at b8:27:eb:f9:92:f7, length 46                                                                                                                                      
11:59:13.451956 b8:27:eb:f9:92:f7 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 68: 10.23.16.10.49157 > 10.23.16.13.69:  26 RRQ "start.elf" octet tsize 0                                                                                                                    
11:59:15.158692 b8:27:eb:f9:92:f7 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 68: 10.23.16.10.49158 > 10.23.16.13.69:  26 RRQ "fixup.dat" octet tsize 0

JamesH65, 2018-06-29:
@ghollingworth Are any of the recent bootcode.bin changes likely to have sorted this out?

ghollingworth, 2018-06-29:
There have been a lot of changes...  Probably worth reconfirming

Gordon
-- 
--
Director of Software Engineering, Raspberry Pi (Trading) Limited
t:  +44 (0) 1223 322633
m: +44 (0) 7450 946289
e: gordon@raspberrypi.org
w: www.raspberrypi.org