raspberrypi/firmware #746: USB boot failure when USB camera also attached

pishotta, 2017-02-20:
Hi,

Following instructions from

> https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

I got my RPi-3 to boot from a USB stick (a SanDisk Ultra 64) under updated and upgraded Raspbian Jessie. It works just fine, exactly as described, except it won't boot when I have either of my USB Logitech webcams plugged in.  Sounds like a timing problem, but...

Notes and things I have tried:

1. I added "rootdelay=10" to /boot/cmdline.txt
2. I used the correct "root=PARTUUID=xxx-xxx..." syntax in cmdline.txt
3. I added "program_usb_timeout=1" to config.txt
4. I used the UUID of the /root file system in fstab instead of the /dev/sda2 designation
5. It boots OK with keyboard, mouse, an additional USB hub attached, and/or other additional memory sticks in the USB sockets
6. Using a powered USB hub to connect the camera (in case a power issue) didn't help
7. Juggling what's plugged into what USB slot doesn't help either
8. Neither dmesg, syslog, nor messages show anything obvious (it probably never gets that far)
9. The webcams work OK if I plug either in after the boot completes

But absolutely nothing happens on the screen when a webcam is plugged in (trying two different ones). I get one flash of the activity light on the USB boot device, but nothing thereafter. Works perfectly without a camera plugged in. Seems like specifically the camera is fouling up the hunt for a boot device.

cmdline.txt

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=0065dda2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes elevator=deadline rootwait rootdelay=30

blkid

> /dev/sda1: LABEL="BOOT" UUID="FFD8-3B23" TYPE="vfat" PARTUUID="0065dda2-01"
/dev/sda2: UUID="75af61a7-08a9-4e74-aae5-570cd4c2072e" TYPE="ext4" PARTUUID="0065dda2-02"

fstab

<pre>proc               /proc     proc     defaults     0     0
# /dev/sda1        /boot     vfat     defaults     0     2
UUID=FFD8-3B23     /boot     vfat     defaults     0     2
# /dev/sda2        /         ext4     defaults,noatime   0   1
UUID=75af61a7-08a9-4e74-aae5-570cd4c2072e   /   ext4   defaults,noatime   0   1
# a swapfile is not a swap partition, no line here
# use  dphys-swapfile swap[on|off]  for that
</pre>

lsusb

> Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 0409:0058 NEC Corp. HighSpeed Hub
Bus 001 Device 006: ID 0781:5583 SanDisk Corp.
Bus 001 Device 007: ID 046d:0809 Logitech, Inc. Webcam Pro 9000
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I note that "lsusb -v" does show the webcam to be the most complicated device with gobs of interfaces.

Like I said, looks like the presence of the usb webcam is either slowing or blocking the finding of the bootable usb mass storage device 006, but I'm out of ideas as to how to deal with this.  AFAIK, it is only the presence of a webcam that causes this to happen.  Perhaps there are additional options?  At the very least, I thought I should bring it to your attention so that if this is a systemic issue and there is a solution to what may become a more common problem, it can be referenced in the documentation.

-- Fred P.

ghollingworth, 2017-02-21:
Can you please post up the response to the following:

lsusb -d 046d:0809 -v

I've tried it here with a webcam and it works fine...

One thing that might be worth trying, copy just bootcode.bin onto an SD card (nothing else on there) and try again...  I've fixed a couple of minor issues with flash drives that might be causing this, if that still breaks then try adding a file "TIMEOUT" to the SD card.  Doesn't matter what it contains just so long as the file exists.

pishotta, 2017-02-21:
Hi,

Here are the lsusb results.  Pretty big and largely inscrutable (to me):  [lsusb_results.txt](https://github.com/raspberrypi/firmware/files/790685/lsusb_results.txt)

bootcode.bin (alone) on the first vfat partition of an  SD card didn't help.
touch TIMEOUT didn't help either.

Bummer when you can't reproduce the problem on your end, but one wonders what that means?

-- Fred P.

ghollingworth, 2017-02-22:
Do you also have a hub connected there or is that part of the camera?

I would guess that the problem here is the hub causing an issue when trying to boot...

pishotta, 2017-02-22:
The KB and mouse sit on a hub, but I have tried it with all hubs disconnected, and it makes no difference.  It will not boot with only the camera and USB drive directly connected.

However...

I tried copying the whole /boot partition to the SD card, and then it boots with everything connected to the USBs?!?  After the boot, it shows mounts only on the USB drive /dev/sda partitions, not the /dev/mmcblk0p1 partition.
<pre>
root@rpi-3green:~# mount
/dev/sda2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=443280k,nr_inodes=110820,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=22,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/sda1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=89580k,mode=700)

root@rpi-3green:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       59472212 3289944  53138124   6% /
devtmpfs          443280       0    443280   0% /dev
tmpfs             447888       0    447888   0% /dev/shm
tmpfs             447888    6436    441452   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             447888       0    447888   0% /sys/fs/cgroup
/dev/sda1          94759   33344     61415  36% /boot
tmpfs              89580       0     89580   0% /run/user/0
</pre>
I guess I could live with that, but I am concerned about where it actually pulled the running kernel from.  The SD card or the USB drive?  Possibly/probably (and unfortunately) the SD card.  Any easy way to check that besides deleting things until it fails?

-- Fred P.

isipka, 2018-07-31:
Having the same issue as described with Logitech USB camera and a USB stick booting. The boot from USB stick goes fine without the USB cam plugged in. The USB camera works correctly when plugged int after USB booting. Any advice ?

ghollingworth, 2018-07-31:
Can you put the bootcode.bin onto an SD card (on its own nothing else), and try again?

isipka, 2018-08-01:
Copied bootcode.bin from USB to SD Card, no boot 
- (USB plugged in, WebCam plugged in, SD Card in) - no boot
- (WebCam plugged in, SD Card in) - no boot
Copied latest bootcode.bin from GitHub firmware to SD Card, no boot 
- (USB plugged in, WebCam plugged in, SD Card in) - no boot
- (WebCam plugged in, SD Card in) - no boot

ghollingworth, 2018-08-02:
OK,

I assume that with the SD card in and without the webcam then it does boot?  You haven't shown a case above where it does boot, are you sure the USB is setup correctly?

Otherwise, run the following command on a Pi on the bootcode.bin

sed -e "s/BOOT_UART=0/BOOT_UART=1/" -i bootcode.bin

You can check that it exists in your version of bootcode.bin and that the operation worked by doing:

strings bootcode.bin | grep "BOOT_UART"

This will then output more useful debug information to the UART 115200-8-n-1

Gordon

isipka, 2018-08-02:
> I assume that with the SD card in and without the webcam then it does boot? 
I will check this and get back

> You haven't shown a case above where it does boot, are you sure the USB is setup correctly?
(USB plugged in, no WebCam, no SD Card) - boots ok, the camera also works if I plug it after boot 

Unfotunatelly I don't have USB to TTL Serial Cable so that I can read UART, any other approach to debug ?

MrCalumet, 2018-12-17:
Exactly the same problem. Any solution ?
Thanks 👍

ghollingworth, 2018-12-18:
If you've got a serial cable then plug that in and follow instructions above to get debug out

nurtext, 2018-12-28:
Same problem here. Logitech C525 with Raspberry Pi 3 Model B+
Any chance of getting this issue solved?

ghollingworth, 2019-01-03:
Just as soon as I can get some debug:

Question 1: Does the green LED do anything with the webcam plugged in?  Or does it stay dark forever?
Question 2: If it does flash then follow the instructions above for debug information

MrCalumet, 2019-01-03:
For me, i'm giving up. I leave an sd card with berryboot and run raspian on a pidrive.
Now, no problem with the logitech webcam at start up.

Insane88, 2019-01-23:
Exactly the same issue... and need it for octopi... for remote reboot after plugins and updates..

ghollingworth, 2019-01-24:
Not going to bother asking the same questions again...  But suffice to say, this won't be re-opened until someone has bothered to answer them

isipka, 2019-02-04:
> Just as soon as I can get some debug:
> 
> Question 1: Does the green LED do anything with the webcam plugged in? Or does it stay dark forever?
It stays dark forever
> Question 2: If it does flash then follow the instructions above for debug information
Well I got the cable, I get the boot log from serial terminal without the camera pulugged in. I tried to get the terminal reading using serial cable but the device doesn't boot when usb and camera are plugged in.

MagnusThome, 2019-03-26:
Have the exact same issue (also Logitech webcam, 525)

With the cam plugged in the green LED doesn't light up what so ever . 

@ghollingworth What can I do to help?

pishotta, 2019-03-27:
I don’t believe this has ever been resolved since I originally filed it years ago.  My best guess is that the camera confuses the firmware as to the proper usb boot device, so it mistakenly tries to boot from the camera.  Firmware not my specialty, though.

llabas, 2019-07-03:
Hi,  I have the same issue with Octoprint using two different logitech models.

When plugged into the USB hub or the PI directly, there is no flashing of the green light on an RPI 3.

The red light just stays on also.

I bought a USB to TTL serial cable so if someone can point me to how to get the debug, I can get it and attach it.

llabas, 2019-07-04:
I've attached the USB to TTL Serial cable and have an open terminal on it.  Verified by getting a console login after boot without the logitech webcams on.

sed -e "s/BOOT_UART=0/BOOT_UART=1/" -i bootcode.bin was done and verified.  

Shutdown, added the camera and added power back.

Nothing on the console.  No green light activity and the red light just stays on.  I'm going to leave it connected like this for a bit to see if there is something that will timeout.
[image]

pishotta, 2019-07-04:
The green light is for network activity. It never gets far enough in the boot process to activate that circuitry. -- Fred P.Sent from a device the make or model of which I refuse to reveal.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/raspberrypi/firmware/issues/746?email_source=notifications\u0026email_token=ACROKDLDWRL6YRQE22FJGN3P5U6L5A5CNFSM4DA2SWU2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZF7JSI#issuecomment-508294345",
"url": "https://github.com/raspberrypi/firmware/issues/746?email_source=notifications\u0026email_token=ACROKDLDWRL6YRQE22FJGN3P5U6L5A5CNFSM4DA2SWU2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZF7JSI#issuecomment-508294345",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

llabas, 2019-07-04:
Removed the webcam and it boots.   Partial console debug below to show it does boot and I am capturing it.

Raspberry Pi Bootcode

USB MSD boot
Read File: config.txt, 1644 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1644
Read File: start_x.elf, 3774980 (bytes)
Read File: fixup_x.dat, 9752 (bytes)
MESS:00:00:14.884329:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:14.889054:0: brfs: File read: 1644 bytes
MESS:00:00:15.116637:0: HDMI:EDID error reading EDID block 0 attempt 0
MESS:00:00:15.122710:0: HDMI:EDID error reading EDID block 0 attempt 1
MESS:00:00:15.128955:0: HDMI:EDID error reading EDID block 0 attempt 2
MESS:00:00:15.135205:0: HDMI:EDID error reading EDID block 0 attempt 3
MESS:00:00:15.141455:0: HDMI:EDID error reading EDID block 0 attempt 4
MESS:00:00:15.147705:0: HDMI:EDID error reading EDID block 0 attempt 5
MESS:00:00:15.153955:0: HDMI:EDID error reading EDID block 0 attempt 6
MESS:00:00:15.160206:0: HDMI:EDID error reading EDID block 0 attempt 7
MESS:00:00:15.166455:0: HDMI:EDID error reading EDID block 0 attempt 8
MESS:00:00:15.172706:0: HDMI:EDID error reading EDID block 0 attempt 9
MESS:00:00:15.178718:0: HDMI:EDID giving up on reading EDID block 0
MESS:00:00:15.649308:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:15.653921:0: HDMI:Setting property pixel encoding to Default
MESS:00:00:15.659521:0: HDMI:Setting property pixel clock type to PAL
MESS:00:00:15.665683:0: HDMI:Setting property content type flag to No data
MESS:00:00:15.672283:0: HDMI:Setting property fuzzy format match to enabled
MESS:00:00:15.679058:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:15.883125:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:15.890571:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined

pishotta, 2019-07-04:
The problem is certainly reproducible. The firmware group will need to address it for it to be resolved.  -- Fred P.Sent from a device the make or model of which I refuse to reveal.

nurtext, 2019-07-04:
Thanks for debugging. So what are the next steps here? Having the same issue as mentioned long time ago in a comment above… What needs to be done to get this fixed once and for all?

nurtext, 2019-07-04:
@pishotta Btw: You closes this issue in 2017. I think it should be re-opened to be recognized as a bug that still exists.

pishotta, 2019-07-04:
Will do, @nurtext .

llabas, 2019-07-05:
I've left the serial cable attached (used a different ground so I can have my cooling fan on) so I'm ready to assist whenever. 

Two cameras have the issue that I own.  Won't boot when attached.  To get them to work I watch the console and plug them in after the debug starts showing the PI 3 booting.

1) C270
2) LZ211SX

My old Orbit AF does not exibit the behavior.  It boots without issue.

llabas, 2019-07-14:
Did some more testing tonight and I can confirm that my PI4 is not affected. 

Plugged my B910 into it and the Pi4 booted with no issues and was recognized.

kevinmce, 2019-07-29:
Unfortunatly I seem im having this issue on the PI4 but with not a USB cam, instead a printer control board the SKR1.3. When removing the SKR1.3 USB the pi boots normally.

pelwell, 2019-07-29:
This thread is about USB booting, which the Pi 4 doesn't currently support. Unless you are somehow USB booting your Pi 4, you have a different issue.

pelwell, 2019-07-29:
I would firstly be suspicious about power consumption and voltage levels - make sure your power supply is adequate, remembering that the Pi4 draws more current than the Pi3 and 3+, and that boot time can be the time of peak current consumption. You could try plugging in the printer after booting to see how it behaves (this is not a solution, obviously), or attaching it via a powered hub.

kevinmce, 2019-07-30:
Apologies, I missed the fact that this thread was specifically for USB booting. In my case the pi4 is booting from the SD card.

An update in my case is that I was able to come up with a solution in the form of switching the power the SKR board from internal to USB. And so now it seems that the SKR USB only becomes active after the Pi4 has powered on and so not preventing the pi4 from booting.

Again apologies if my issue is off topic.

llabas, 2019-10-23:
Is this a dead thread and issue?  Any plans to fix it?

pishotta, 2019-10-23:
Not fixed yet, as far as I know, but not exactly pinned down yet, either.  Seems to work with some USB cameras, but I think most/all failures have revolved around Logitech cameras, which do present a pretty complex device picture to the Pi, which I think confuses the Pi about what the boot device is.  Just a hunch.  

 

Don’t know if it’s still a problem with Buster.  There was a lot of low level re-writing with that upgrade.  Personally, I threw in the towel and moved on.

 

-- Fred P.

pishotta, 2019-10-23:
And I don’t know if anybody has tried it with Buster on a Pi-4.

isipka, 2019-12-08:
Hi, did anybody from the formware group investigate the issue any further ?
Still having the same issue.

spout, 2020-01-14:
@isipka same issue with Logitech C270 :
- Wont boot on USB with C270 plugged
- Boots on SD only + C270
- Swapped ports, etc...
- Boot on USB with another webcam
- Same problem with only bootcode.bin on SD card

llabas, 2020-01-21:
@pishotta My understanding is the Pi 4 does not support USB booting yet.

schneemaier, 2020-04-11:
I have a similar issue on an older Raspberry Pi with PXE boot.
When i try to boot the Pi with the Logitech, Inc. HD Pro Webcam C920 (046d:082d) attached, it does not boot. I have an SD card in the Pi with the bootcode.bin file. Here is the output from the console, when the webcam is attached during boot:

Raspberry Pi Bootcode                                                                                                                                
                                                                                                                                                     
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0                                                                          
Trying USB                                                                                                                                           
Hub device found at addr 4, enumerating HUB                                                                                                          
Initialise hub                                                                                                                                       
Found 5 ports, multi_tt = 1                                                                                                                          
Setting interface 0                                                                                                                                  
Enabling PORT POWER on port 1                                                                                                                        
Enabling PORT POWER on port 2                                                                                                                        
Enabling PORT POWER on port 3                                                                                                                        
Enabling PORT POWER on port 4                                                                                                                        
Enabling PORT POWER on port 5                                                                                                                        
Waiting for devices to respond to reset                                                                                                              
Found device on port 1                                                                                                                               
Found highspeed device                                                                                                                        
Device found: type = Ethernet adapter, addr = 5                                                                                                      
Found device on port 3                                                                                                                               
Found highspeed device                                                                                                                               
Trying booting from MSD device addr 0                                                                                                                
Retrieving the config descriptor                                                                                                                     
Failed to read short config descriptor -2                                                                                                            
Failed initialising mass storage device                                                                                                              
Trying booting from MSD device addr 0                                                                                                                
Retrieving the config descriptor                                                                                                                     
Failed to read short config descriptor -2                                                                                                            
Failed initialising mass storage device                                                                                                              
Trying booting from MSD device addr 41                                                                                                               
Retrieving the config descriptor
.....

It seems that the Pi is trying to boot from the Webcam for some reason.

And this is the log when the webcam is not attached:

Raspberry Pi Bootcode                                                                                                                                
                                                                                                                                                     
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0                                                                          
Trying USB                                                                                                                                           
Hub device found at addr 4, enumerating HUB                                                                                                          
[... cut ...]
Found device on port 1                                                                                                                               
Found highspeed device                                                                                                                               
Device found: type = Ethernet adapter, addr = 5                                                                                                      
Trying booting from Ethernet device addr 5                                                                                                           
....

capi92, 2020-04-13:
Same here C270 on octoprint distrib

llabas, 2020-04-13:
This issue has been open for 3 years now, is there any hope or should it just be marked "Will not fix"?

JamesH65, 2020-04-13:
Difficult to tell. If the guy who knows all about the USB on PI0-3 has time, then maybe, but we are all pretty busy.

ghollingworth, 2020-04-14:
I've rebuilt bootcode.bin with a possible fix for this, I think I understand what the issue is (which is to do with the interface descriptor not being the first thing to follow the config descriptor)

https://drive.google.com/file/d/1QHEFqh_dS5A7Tyxm_UvjZGD9-3X28gzv/view?usp=sharing

Please copy this to a blank,  FAT32 formatted SD card and try again...

schneemaier, 2020-04-15:
Thanks for the updated bootcode.bin. Unfortunately it has the same issue as before:

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 4
Found highspeed device
Trying booting from MSD device addr 0
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 0
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 41
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 9
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 1
Retrieving the config descriptor

ghollingworth, 2020-04-17:
OK, added some extra debug to the bootcode.bin, can you try again?

[bootcode.zip](https://github.com/raspberrypi/firmware/files/4491714/bootcode.zip)

schneemaier, 2020-04-17:
Thanks. Here is the result with Logitech, Inc. HD Pro Webcam C920 (046d:082d) attached:
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Set address 4
Num devices = 1, addr = 4
get_config_descriptor 41 bytes
device class = 9
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Set address 5
Num devices = 2, addr = 5
get_config_descriptor 39 bytes
device class = 255
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Set address 6
Num devices = 3, addr = 6
get_config_descriptor 3452 bytes
device class = 239
Trying booting from MSD device addr 0
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 0
Retrieving the config descriptor
Failed to read short config descriptor -2
Failed initialising mass storage device
Trying booting from MSD device addr 41
Retrieving the config descriptor

this is with an other type of webcam when the boot works(also logitech):
Raspberry Pi Bootcode
Read File: config.txt, 1798

Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Set address 4
Num devices = 1, addr = 4
get_config_descriptor 41 bytes
device class = 9
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Set address 5
Num devices = 2, addr = 5
get_config_descriptor 39 bytes
device class = 255
Device found: type = Ethernet adapter, addr = 5
Found device on port 4
Found highspeed device
Set address 6
Num devices = 3, addr = 6
get_config_descriptor 1433 bytes
device class = 239
Device found: type = Ethernet adapter, addr = 6
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:dc:06:4c
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
ARP request: my IP addr 192.168.10.131 my eth_addr b8:27:eb:dc:06:4c ARP for ad9
Done ARP for 192.168.10.9 got 00:25:31:04:0a:37
Read File: config.txt, 1798 (bytes)

ghollingworth, 2020-04-20:
OK, I'm guessing the problem is related to the size of the config descriptor...  Will try only reading a smaller amount of it...

Gordon

schneemaier, 2020-04-28:
Gordon, did yo made any progress on this issue? Thanks

ghollingworth, 2020-04-29:
Can you try this bootcode.bin?

[bootcode.zip](https://github.com/raspberrypi/firmware/files/4550607/bootcode.zip)

schneemaier, 2020-04-30:
Thanks! It works, i tried all the USB webcams i have at home and it can boot even when 4 of them attached at the same time.

Here is the boot log:
Raspberry Pi Bootcode

Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0
Trying USB
Set address 4
Num devices = 1, addr = 4
get_config_descriptor 41 bytes
device class = 9
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Set address 5
Num devices = 2, addr = 5
get_config_descriptor 39 bytes
device class = 255
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Set address 6
Num devices = 3, addr = 6
get_config_descriptor 200 bytes
device class = 239
Device found: type = Ethernet adapter, addr = 6
Found device on port 3
Found highspeed device
Set address 7
Num devices = 4, addr = 7
get_config_descriptor 200 bytes
device class = 239
Device found: type = Ethernet adapter, addr = 7
Found device on port 5
Found highspeed device
Set address 8
Num devices = 5, addr = 8
get_config_descriptor 200 bytes
device class = 239
Device found: type = Ethernet adapter, addr = 8
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:XX:XX:XX
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
ARP request: my IP addr 192.168.YYY.XXX my eth_addr b8:27:eb:XX:XX:XXc ARP for addr 192.168.YYY.XXX
Done ARP for 192.168.YYY.XXX got 00:25:31:XX:XX:XX
Read File: config.txt, 1798 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 1798

pishotta, 2020-04-30:
Does this mean the problem is fixed?!?!?  I've been following this issue for years since I initially observed the phenomenon.  Out of curiosity, what was the problem, some kind of memory overrun when loading config info?

Is the current fix to simply substitute the modified bootcode.bin for the one in /boot?  And any idea when the new version will work its way into the official Raspbian distro?

Thanks for your persistence, @ghollingworth .

ghollingworth, 2020-05-01:
Yes, the issue was with a buffer overrun when reading the descriptors.  Difficult to debug from logging because it just didn't make any sense, (which is why I went looking for it!)

Of course it is not possible to fix this in the bootROM (for obvious reasons), but if you put the bootcode.bin onto an empty SD card it'll work.

Shouldn't take too long to make it's way through the build system, keep an eye out on the commit logs.  Just checking I've not broken anything and seeing whether there is anything I can fix whilst it's fresh in my head!

Gordon

capi92, 2020-09-07:
Thanks for your work. 
When is it going to be part of the release ? can't see anything in commit logs.

jimmyeao, 2020-09-08:
Same issue here. I wonder if the bootloader code could be adjusted to use a specific port for booting?

capi92, 2020-11-07:
> Thanks for your work.
> When is it going to be part of the release ? can't see anything in commit logs.

Any news ?

ghollingworth, 2020-11-07:
Already fixed and in bootcode.bin

closed 2020-11-07 by ghollingworth

pishotta, 2020-11-08:
Thanks! -- Fred P.

hlipka, 2021-03-23:
I'm still seeing this problem when trying to boot from USB with a C270 camera connected. I tried the latest bootcode (retrieved with rpi-update), the last bootcode file attached above, and the revision 7eff9f6 on 1 May 2020 (https://github.com/raspberrypi/firmware/blob/7eff9f6774bb43bfd61e749a0b45ffddc98c2311/boot/bootcode.bin ) which is the one that should have fixed it.
Does this work only when booting into USB via a SD card? (Because thats what I wanted to avoid)

nitronixxx, 2021-08-13:
Hi, i don't now how to load the bootcode.bin in the RPI3B+ to solve my USB Webcam problem. I must make à copy of the bootcode.bin in a SD card, boot on, then shutdown and then already start with my USB SDD und my USB webcam?

[6 more replies]