raspberrypi/firmware #748: device tree overlay: when and how does the kernel load the module

nikiwaibel, 2017-02-24:
more a question then an issue.

if there is a fully modular linux kernel and we have a proper set of overlays in config.txt. can someone describe (or point a link) to the things that happen on the kernel side to load the modules related to the device tree?

i am asking, because i wonder when this happens. the kernel may not see the filesystem with the modules when it is loaded and run. later, when the filesystems are mounted, the modules may appear.

pelwell, 2017-02-24:
Please ask this question as a new topic in the Raspberry Pi Device Tree forum and I may have a go at answering. I don't have a potted answer to hand, and writing one will probably involve some research, so I'd like it to be seen by more people and not generate more inbox spam for the other firmware maintainers.

closed 2017-02-24 by nikiwaibel

nikiwaibel, 2017-02-24:
@pelwell thanks for the forum hint.

https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=175433

pelwell, 2017-02-24:
I've bookmarked the post and I will put together an answer, it just might take a while.