raspberrypi/firmware #749: bootcode.bin cannot load files when loaded from a GPT drive

feherneoh, 2017-02-24:
I'm using commit 1a717bb3efc7f4751d8c1839f8d48111a89958f8 from the next branch.
I'm trying to boot from an external hard drive, that is partitioned as GPT. When I have another USB device connected (can be anything, mouse, keyboard, game controller), bootcode loads from the hard drive's EFIESP partition, the rainbow splash appeares, but the green led starts blinking, and nothing more happens. I tried disabling the rainbow screen in config.txt, but it did not have any effect, so seems like it's not loaded at all.

Drive is a Western Digital My Passport Ultra (1TB), partitioned using Windows' diskpart.
`Bus 004 Device 002: ID 1058:0837 Western Digital Technologies, Inc.`
No ext4 partition was added yet.

Partition table shown by gdisk:
```
Command (? for help): p
Disk /dev/sdc: 1953458176 sectors, 931.5 GiB
Logical sector size: 512 bytes
Disk identifier (GUID): XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
Partition table holds up to 128 entries
First usable sector is 34, last usable sector is 1953458142
Partitions will be aligned on 8-sector boundaries
Total free space is 4029 sectors (2.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1              34          262177   128.0 MiB   0C01  Microsoft reserved ...
   2          264192         1288191   500.0 MiB   EF00  EFI system partition
   3         1288192      1953456127   930.9 GiB   0700  Basic data partition

Command (? for help):
```

Device is a Raspberry Pi 3 Model B V1.2

Also note that bootcode.bin won't get loaded if the only attached USB peripheral is the external HDD.

popcornmix, 2017-02-24:
Ping @ghollingworth

feherneoh, 2017-03-12:
I'll play the impatient guy here, so ping @popcornmix @ghollingworth
Offtopic, is it a known error that the bootrom seemingli cannot read Kingston flashdrives? (not bootcode, the actual bootrom)

ghollingworth, 2017-03-12:
There are some drives that will not boot properly, it's known about and if it works with just bootcode.bin on the SD card then it's already been found!

Unfortunately it's still going to take some time to get around to this, I think it should be fixable quite easily though, the code does support GPT (the fact that you get the splash screen means it's a problem with bootcode.bin)

Gordon

JamesH65, 2017-12-21:
@feherneoh @ghollingworth Any status to report on this?

JamesH65, 2018-06-29:
@ghollingworth Ping - can be closed?

ghollingworth, 2018-06-29:
Unfortunately to support guid partitions, it needs code for both
bootcode.bin and start.elf to be written.

It's quite far down the list of properties
-- 
--
Director of Software Engineering, Raspberry Pi (Trading) Limited
t:  +44 (0) 1223 322633
m: +44 (0) 7450 946289
e: gordon@raspberrypi.org
w: www.raspberrypi.org

satmandu, 2019-12-03:
Is loading start.elf and the rest of the bootloader files from the first fat32 GUID partition supported on the 4B without Hybrid GPT/MBR hackery on either the SD card or via usb boot (as planned.)

The reason I'm asking is that ubuntu 20.04 is getting their zfs installer setup to format gpt partitions and it would be useful to not have to use different tooling to support an image for RPI boot disks.