raspberrypi/firmware #752: firmware replaces console=ttyAMA0 with console=ttyS0 even when ttyS0 is disabled

stapelberg, 2017-02-26:
I am running Linux 4.9.9 (upstream, not the Raspberry Pi version) on my Raspberry Pi 3.

I have patched the upstream devicetree files to provide UART0 on GPIO 14/15 and disable UART1 entirely (which is what the pi3-disable-bt overlay does):
```diff
--- a/arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dts
+++ b/arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dts
@@ -18,6 +18,19 @@
 	};
 };
 
+&gpio {
+	uart0_pins: uart0_pins {
+		brcm,pins = <14 15>;
+		brcm,function = <BCM2835_FSEL_ALT0>;
+		brcm,pulls = <0 2>;
+	};
+};
+
+&uart0 {
+	pinctrl-names = "default";
+	pinctrl-0 = <&uart0_pins>;
+};
+
 &uart1 {
-	status = "okay";
+	status = "disabled";
 };
```

My `cmdline.txt` reads `console=ttyAMA0,115200 root=/dev/mmcblk0p2 elevator=deadline rootwait`. 

## What I see

I get no kernel boot output on UART0. However, `/dev/ttyAMA0` works after boot.

Running `cat /proc/cmdline`, I can see that the firmware replaced `console=ttyAMA0,115200` with `console=ttyS0,115200`, even though I disabled UART1. This is happening even when `enable_uart=0` is set in `config.txt`.

## What I expected

I would have expected no replacements to occur.

## Workaround

As a workaround, I’m setting `console=ttyAMA0,115200` in the devicetree bootargs property, but that’s not a good solution: I need to update the dtb file whenever I need to switch my console setting, which is a rather elaborate procedure (as opposed to editing `cmdline.txt`).

## Theory/question

My theory is that the firmware makes the decision as to whether the `console=` parameter should be replaced based on some aspect which differs on my system. Could you clarify how the firmware detects UART1 to be disabled in that regard? Does it not just look at the `status` property in the device tree?

pelwell, 2017-02-27:
There is something that the pi3-disable-bt overlay does that your hacked dtb doesn't; the source is [here](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/pi3-disable-bt-overlay.dts) - can you spot it?

The Pi serial port handling now uses the concepts of primary and secondary UARTs - the primary UART is the one on which you would start a console, the secondary is the other one (used for Bluetooth on a Pi3). If the firmware determines that uart1 (ttyS0) is the primary it will replace uses of ttyAMA0 on the command line with ttyS0.

The primary/secondary role assignments are determined by the DT aliases serial0 and serial1 - serial0 is the primary, serial1 is the secondary. If you swap the aliases over in your dtb the rewriting will stop. Raspbian uses serial0 in its command line, making it immune to remapping.

Having said that, I hope you are prepared to update your dtb with kernel changes - using overlays would insulate you from that.

stapelberg, 2017-03-01:
Thanks for the explanation!

> The primary/secondary role assignments are determined by the DT aliases serial0 and serial1 - serial0 is the primary, serial1 is the secondary. If you swap the aliases over in your dtb the rewriting will stop. 

I set up the correct aliases (?) but the firmware still rewrites my cmdline:

```
# hd /proc/device-tree/aliases/serial0
00000000  2f 73 6f 63 2f 73 65 72  69 61 6c 40 37 65 32 30  |/soc/serial@7e20|
00000010  31 30 30 30 00                                    |1000.|
00000015
# hd /proc/device-tree/aliases/serial1
00000000  2f 73 6f 63 2f 73 65 72  69 61 6c 40 37 65 32 31  |/soc/serial@7e21|
00000010  35 30 34 30 00                                    |5040.|
00000015
# cat /proc/cmdline
earlyprintk console=tty1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa02082 bcm2709.serial=0x95d0f70b smsc95xx.macaddr=B8:27:EB:D0:F7:0B bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  console=ttyS0,115200 root=/dev/mmcblk0p3 elevator=deadline rootwait
# mount /dev/mmcblk0p1 /tmp/mnt
# cat /tmp/mnt/cmdline.txt 
console=ttyAMA0,115200 root=/dev/mmcblk0p3 elevator=deadline rootwait
```

Changing cmdline.txt to read `console=serial0,115200` has the same effect, i.e. it’ll be replaced with `console=ttyS0,115200`.

Even swapping the `serial0` and `serial1` aliases doesn’t change anything. It’s as if the firmware didn’t read them.

My updated patch is:
```diff
--- a/arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dts
+++ b/arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dts
@@ -16,8 +16,26 @@
 			gpios = <&gpio 47 0>;
 		};
 	};
+
+	aliases {
+		serial0 = &uart0; /* primary */
+		serial1 = &uart1; /* secondary */
+	};
+};
+
+&gpio {
+	uart0_pins: uart0_pins {
+		brcm,pins = <14 15>;
+		brcm,function = <BCM2835_FSEL_ALT0>;
+		brcm,pulls = <0 2>;
+	};
+};
+
+&uart0 {
+	pinctrl-names = "default";
+	pinctrl-0 = <&uart0_pins>;
 };
 
 &uart1 {
-	status = "okay";
+	status = "disabled";
 };
```

When converting the dtb file to dts, I see:
```
        aliases {
                ethernet = "/soc/usb@7e980000/usb1@1/usbether@1";
                serial0 = "/soc/serial@7e201000";
                serial1 = "/soc/serial@7e215040";
        };
```

Here’s my device tree as [dtb](https://github.com/raspberrypi/firmware/files/810070/rpi-3-b.dtb.gz) and [txt](https://github.com/raspberrypi/firmware/files/810071/rpi-3-b.txt.gz).

> Having said that, I hope you are prepared to update your dtb with kernel changes - using overlays would insulate you from that.

Yeah, I’ll eventually switch to an overlay. First trying to understand how the mechanism works in detail, for which directly modifying the dtb is easier.

pelwell, 2017-03-01:
If you also add aliases for uart0 and uart1 it should work - the firmware is doing a string comparison on the uart0 and serial0 aliases. If they are different it swaps the UARTs.

I would consider changing the logic to make it more upstream-friendly, provided it doesn't affect the downstream use cases - suggestions welcome.

stapelberg, 2017-03-01:
Thanks, adding the uart0 and uart1 aliases did the trick.

closed 2017-03-01 by stapelberg