raspberrypi/firmware #753: USB audio stuttering since update to 4.9 kernel (USB packet management regression?)

trulio, 2017-02-26:
Hi,

i just wanted to raise that there is more audio stuttering using mopidy since 4;9 update.

I'm using an synchronous USB DAC (USB adaptive mode).

see this product description if needed. (http://hifimediy.com/DACs/ready-made-dacs/Sabre-9018-DAC)

before i had stuttering a few seconds when restarting OS and it was stabilizing after a few seconds but now it does not stabilize.

It could be an USB regression with packet losses.

Regards

kristofejro, 2017-03-19:
I have someting similar but with USB camera. Picture is stuttering (jams)
As far as I remember it is happening since 4.9.14

kristofejro, 2017-03-20:
With 4.9.16 it is still stuttering.

LichtiMC, 2017-04-24:
Same problem here with Hifiberry Amp+ (https://www.hifiberry.com/products/ampplus/).
Pulseaudio ("squeezelite -o pulse") gives stuttering sound, whereas alsa ("squeezelite -o hw:CARD=sndrpihifiberry,DEV=0") works without stuttering.

So I guess it's pulseaudio related?

Here are the log files with latest stable kernel (https://github.com/Hexxeh/rpi-firmware/commit/9108b7f712f78cbefe45891bfa852d9347989529):
[pulseaudio_playing.txt](https://github.com/raspberrypi/firmware/files/950719/pulseaudio_playing.txt)
[pulseaudio_start.txt](https://github.com/raspberrypi/firmware/files/950720/pulseaudio_start.txt)
[squeezelite.txt](https://github.com/raspberrypi/firmware/files/950718/squeezelite.txt)

With latest experimental kernel (rpi-update without parameter) the playback looks good in log-files, but I have no sound output at all. (Even mplayer doesn't output anything, no matter if alsa or pulse)
[pulseaudio_start1.txt](https://github.com/raspberrypi/firmware/files/950809/pulseaudio_start1.txt)
[squeezelite1.txt](https://github.com/raspberrypi/firmware/files/950810/squeezelite1.txt)
[pulseaudio_playing1.txt](https://github.com/raspberrypi/firmware/files/950811/pulseaudio_playing1.txt)

trulio, 2017-05-08:
OK i will double the audio buffer in Mopidy audio parameters to see if it changes something

trulio, 2017-05-08:
I confirm the problem is still present. Some USB packets seems to be lost.

P33M, 2017-05-08:
Can I get a `sudo lsusb -v` listing with the broken USB audio device(s) plugged in to the pi?

trulio, 2017-05-08:
here it is
[lsusb_verbose_content.txt](https://github.com/raspberrypi/firmware/files/983533/lsusb_verbose_content.txt)

trulio, 2017-05-08:
forget the comment closing.
of course the thread is still living

Lesita, 2017-05-08:
Can. Using this command to get a list of USB devices.

在2017-5-8 下午9:08之上，2017-5-8 下午9:08写道：

Reopened #753.

P33M, 2017-05-08:
A quick survey of all my various USB audio devices reveals a lack of Isochronous Adaptive endpoints among them. I will have to buy one of these.

trulio, 2017-05-08:
lesita, i don't understand your request. 
what are you expecting from me?

P33M, 2017-05-31:
I have a HifimeDIY DAC on order and will investigate.

trulio, 2017-05-31:
Hi needed, i can make additional tests with following devices : 
dragonfly 1.2 (Asynchronous)
HiFimeDIY Sabre U2 Asynchronous USB DAC
HiFime 9018 Asynchronous High resolution USB DAC

P33M, 2017-06-01:
@trulio Speaker-test works OK for me with a Hifimediy Sabre UAE23. What do I need to install/run from a clean Raspbian install in order to replicate your setup?

trulio, 2017-06-01:
Hi here is the hard part of the job.

mine is not the exact reference with same adaptive chip so we do not have exactly the same env.

Anyway the bad news is that you need to run long time test to get the issue on current kernel (4.9.30)
i would say that every one hour you get it during a few minutes and it can not be removed even by pausing of skipping to next track 
i think the problem can be more easily reproduced with first rpi-update releases of 4.9 kernel
as they appear less and less with the last release of the kernel

my configuration is the following one 
Raspbian Jessie Lite with mopidy
install raspbian lite and just type apt-get install mopidy
to test it on mopidy, you can just stream a radio (no need of a premium spotify account)
just provide the url via android mopidy-radio app for example
i provide my mopidy.conf file and

[conf files.zip](https://github.com/raspberrypi/firmware/files/1045727/conf.files.zip)
 my asound.conf file

but if you want a premium spotify account for spotify tests, i can share with you one i will create on my family account (i will recommand to use the iris extension installable using sudo pip install Mopidy-Iris or more simple an mpd client recommanded on mopidy page)

thank for your efforts and your great job.

Regards, 

Trulio

trulio, 2017-08-10:
Hi, 
what's up with the usb management issue? could you reproduce it?
I rolled back my kernel to last 4.4 availlable (4.4.50) to avoid the issue.
Regards,
Trulio

trulio, 2017-09-14:
Hi guys,

i just wanted to mention that the stuttering issue disappeared since 4.9.46 (and may be 4.9.45 but i 'm not sure of this).

I also wanted to thank you for your hard work and your efforts.
This bug report maybe have been received as a selfish request about a rare issue that only concerned my usage, but i really wanted to contribute to the raspberry pi foundation project and help them make a better product.
I'm well aware of the difficulty of the issue i reported and I've felt really disappointed not been able to bring you useful data about how to track the issue.

Thank you for this great tool, that, i really think, is helping to make the world better.

Trulio

popcornmix, 2017-09-14:
Is it possible 4.9.43 fixed the issue? It could be the same problem as https://github.com/raspberrypi/linux/issues/2134

@kristofejro any improvement for you?

trulio, 2017-09-14:
I can just give an approximate time of when i didn't perceive the issue anymore.
I think i've "rpi-updated" 2 or 3 times(with a real firmware upgrade) and i don't reproduce the issue since more or less two weeks.
I don't think i was exactly in the same conditions of "raspberrypi/linux#2134" issue because 4.9 kernel always had the issue (I'm sure of it).
I can also add that the initial glitches i had when starting usb audio from a rebooted raspberry with 4.4 kernel before stabilization are fewers (but i need to practice additional subjective tests to confirm this). Anyway this case seems to be easier to reproduce than the one that was occurring every 30 minutes.

JamesH65, 2018-06-29:
Please update to the latest kernel which may contain a fix for this issue.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2018-10-29 by JamesH65