raspberrypi/firmware #754: Unable to netboot :(

Baanin, 2017-02-27:
I've a RPI 3B. Installed a SD with newer Raspbian release
Updated the rom to the next BRANCH.
Inserted program_usb_boot_mode=1 in /boot/config.txt
Rebooted.
Checking that vcgencmd otp_dump | grep 17: = 17:3020000a : OK
Everything seems to be ok at this point.
but... It just don't work.
Checked for bootpc packet on my server, nothing, except the Lan connector that is powered up, blinking time to time. But can't see any DHCP request on my server via tcpdump
Tried to set it in "LABO" mode : Rpi as a server and Rpi as a client directly connected via ethernet wire : Nothing,I just can't see the DHCP Request.
If I put back the SD card in my pi, I can see the DHCP request :( 
So I figure that my
So I'm asking your help, maybe i'm missing something ?

ghollingworth, 2017-02-27:
I would suggest first formatting an SD card and copying the bootcode.bin from the fat partition (or copy it from the firmware/boot/ directory)

Make sure it is the only file on there and try booting again...  If it now works then it's a known problem...

Gordon

Baanin, 2017-02-27:
My bad so if I didn't see the known bug. Just tested with the bootcode.bin on the SD and yes it's working :'(

Could you link the conversation about the known issue if I want to track it ?

Thanks for Helping Gordon

Baanin, 2017-03-01:
Ok, it's booting with only the bootcode.bin only in the SD card, but not as it supposed to be. I mean that in the documentation, as the kernel files are not seeked on the folder tftp/serial/kernel.img but directly in tftp/kernel.img, so we are unable to use multiple system or it's just that i'm too dumb to understand the way it works ?

ghollingworth, 2017-03-01:
I assume you're using dnsmasq and its tftp server, if so you should be able to 

tail -f /var/log/daemon.log

To see the log from the tftp server, you should see it first try to access

serial/start.elf

If that file exists then it will prepend 'serial/' in front of all subsequent accesses

Gordon

Baanin, 2017-03-02:
Using dnsmasq and it's ftp server, I see that the file isn't requested in
serial/start.elf.

I've created the folder with the script in documentation/hardware/
raspberrypi/bootmodes/

But as you can see in the next log, the file isn't requested in
serial/start.elf. Maybe I'm missing something ?

ar  2 09:04:27 dnsmasq-dhcp[1743]: 653460281 sent size: 32 option: 43
vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Mar  2 09:04:27 dnsmasq-tftp[1743]: fichier /tftpboot/autoboot.txt non
trouvé
Mar  2 09:04:27 dnsmasq-tftp[1743]: envoyé /tftpboot/config.txt à
10.60.212.124
Mar  2 09:04:27 dnsmasq-tftp[1743]: fichier /tftpboot/recovery.elf non
trouvé

Baanin, 2017-03-02:
And if I remove all files in /tftpboot except the folder with the serial number : 

Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 la gamme DHCP disponible est : 10.60.212.101 -- 10.60.212.150
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 Classe de vendeur ('Vendor Class') : PXEClient:Arch:00000:UNDI:002001
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 DHCPDISCOVER(eth0) b8:27:eb:63:b4:9d
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 options: eth0
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 DHCPOFFER(eth0) 10.60.212.124 b8:27:eb:63:b4:9d
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 options demandées : 43:vendor-encap, 60:vendor-class, 67:bootfile-name,
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 options demandées : 128, 129, 130, 131, 132, 133, 134, 135, 66:tftp-server,
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 options demandées : 1:netmask, 3:router
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 serveur suivant : 10.60.212.200
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  1 option: 53 message-type  2
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option: 54 server-identifier  10.60.212.200
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option: 51 lease-time  12h
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option: 58 T1  6h
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option: 59 T2  10h30m
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option: 28 broadcast  10.60.212.255
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size: 21 option: 66 tftp-server  Raspberry Pi Boot
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option:  3 router  10.60.212.1
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  4 option:  1 netmask  255.255.255.0
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size: 17 option: 97 client-machine-id  00:9d:b4:63:ba:9d:b4:63:ba:9d:b4:63:ba:9d...
Mar  2 10:27:33 dnsmasq-dhcp[2414]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Mar  2 10:27:33 dnsmasq-tftp[2414]: fichier /tftpboot/autoboot.txt non trouvé
Mar  2 10:27:33 dnsmasq-tftp[2414]: fichier /tftpboot/config.txt non trouvé
Mar  2 10:27:33 dnsmasq-tftp[2414]: fichier /tftpboot/recovery.elf non trouvé
Mar  2 10:27:33 dnsmasq-tftp[2414]: fichier /tftpboot/start.elf non trouvé
Mar  2 10:27:33 dnsmasq-tftp[2414]: fichier /tftpboot/fixup.dat non trouvé

ghollingworth, 2017-03-02:
Yes I think I've missed out the serial number check when booting with bootcode.bin only.  Will have a look into this

Baanin, 2017-03-02:
thanks, i'll close when you push the modification :)

Baanin, 2017-03-08:
Hello Gordon, any news ?

Baanin, 2017-03-14:
Hello Gordon, 

Do you have an estimation of when you can push the modification ?

Baanin, 2017-03-24:
Hello @ghollingworth,

Nearly 1 month since we discussed it, do you know if the patch is pushed ?

Ruffio, 2017-04-03:
@ghollingworth, has the modification been pushed? Any ETA?

pelwell, 2017-04-18:
Can you download and test [this bootcode.bin](https://drive.google.com/open?id=0B_P-i4u-SLBXcEpRbDhSaVNJQ3c)? It should only alter behaviour in the case where bootcode.bin is present on the SD card and the failover to net booting.

Baanin, 2017-04-18:
[image]

Sounds to be working for me :) Ty

Baanin, 2017-04-18:
After some testing, it's working fine, but i'm not sure, it seems that if the serial/start.elf is not found, it doensn't seek for the start.elf at the root tftp.

But for me it's working as I needed to :)

pelwell, 2017-04-18:
> After some testing, it's working fine, but i'm not sure, it seems that if the serial/start.elf is not found, it doensn't seek for the start.elf at the root tftp.

Really? It works for me:
```
Apr 18 13:23:30 dnsmasq-tftp[32453]: file /tftpboot/xxxxxxxx/start.elf not found
Apr 18 13:23:30 dnsmasq-tftp[32453]: file /tftpboot/autoboot.txt not found
Apr 18 13:23:30 dnsmasq-tftp[32453]: file /tftpboot/config.txt not found
Apr 18 13:23:30 dnsmasq-tftp[32453]: file /tftpboot/recovery.elf not found
Apr 18 13:23:31 dnsmasq-tftp[32453]: sent /tftpboot/start.elf to w.x.y.z
Apr 18 13:23:31 dnsmasq-tftp[32453]: sent /tftpboot/fixup.dat to w.x.y.z
```

Baanin, 2017-04-18:
My bad, didn't checked all log file.

It's working very well, thanks

popcornmix, 2017-04-19:
The fix should be in latest rpi-update firmware

pelwell, 2017-04-20:
Closing after positive feedback.

closed 2017-04-20 by pelwell