raspberrypi/firmware #757: RealVNC is broken - tells me it's un-licenced.

DougieLawson, 2017-03-01:
https://github.com/Hexxeh/rpi-firmware/commit/5eb58b416697071f9187aa85cea23004ca0270f4 breaks RealVNC.

If I revert to https://github.com/Hexxeh/rpi-firmware/commit/52241088c1da59a359110d39c1875cda56496764 it's back working.

I'll try https://github.com/Hexxeh/rpi-firmware/commit/3e1c3e9ba2b948e6023114b9633223363da78679 to see if it's the 4.9 kernel (don't think so) or the firmware (hence why I opened the issue here).

DougieLawson, 2017-03-01:
It works OK if I revert to https://github.com/Hexxeh/rpi-firmware/commit/3e1c3e9ba2b948e6023114b9633223363da78679

sconemad, 2017-03-02:
Thanks for reporting this @DougieLawson. I've investigated this and it would appear to be an issue in RealVNC Server. We'll do our best to have a fix for before the 4.9 kernel becomes the default.

DougieLawson, 2017-03-02:
Thanks.

Give me an alpha or beta test version. I've got 12 out of 14 raspberries running 4.9 kernels that I can test on.

dividuum, 2017-03-03:
Random throught since I also ran into "problems" with 4.9: Maybe RealVNC checks the Pi serial by reading  `/sys/module/[bcm2708|bcm2709]/parameters/serial`? Those seem to have disappeared with 4.9. So `/proc/cpuinfo` seems to be the best way to get the serial number?

DougieLawson, 2017-03-03:
@dividuum I've got a 4.9.11-v7+ #971 system and ```sys/module/[bcm2708|bcm2709]/parameters/serial``` doesn't exist on there, but RealVNC is working.

sconemad, 2017-03-09:
A new version of the realvnc-vnc-server package (version 6.0.2.26367) has just been made available in the repository, which should fix this problem.

DougieLawson, 2017-03-10:
Thank you - it's working on 4.9.13+ #974 now.

Ruffio, 2017-03-16:
@DougieLawson so this can be closed now?

DougieLawson, 2017-03-16:
RealVNC fixed this with an update to their package.

closed 2017-03-16 by DougieLawson