raspberrypi/firmware #758: PiZeroW device tree

procount, 2017-03-02:
Where is bcm2708-rpi-0-w.dtb ? It is not being pulled into the NOOBS build.

pelwell, 2017-03-02:
There is currently a temporary solution in place based on the B+ DTS (like Zero). The next build is likely to contain the dedicated DTB and the firmware to load it.

procount, 2017-03-02:
So if I understand you correctly, the current firmware will use the B+ DTB on the Pi0W? 
I ask because I built my fork of NOOBS and bcm2708-rpi-0-w.dtb was not present, but it is in the downloadable NOOBS zip file. Currently, I don't have a Pi0W to test it on, so I wanted to make sure it would work. I guess copying bcm2708-rpi-0-w.dtb would do no harm and probably would not be used until the next firmware is released?

pelwell, 2017-03-02:
The bcm2708-rpi-0-w.dtb in the NOOBS zip is there in error and almost certainly is not being used (`sudo vcdbg log msg |& grep dtb` will tell you for certain).  Sorry for any confusion.

procount, 2017-03-02:
No worries. Thanks.

PythonSmith, 2017-03-18:
I think this is somehow related:
I'm using Raspbian Jessie Light from the 2017-03-02 on the RPiZero W.
It boots fine however the file /sys/firmware/devicetree/base/name says the device is a RPiB Rev 2.
It also can't find it's Wifi-Adapter wlan0 (ifconfig -a, iwconfig)

pelwell, 2017-03-18:
What is the Revision number in /proc/cpuinfo?

PythonSmith, 2017-03-18:
`pi@raspberrypi:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 000f
Serial		: 00000000f46969e7
`

pelwell, 2017-03-18:
Interesting. I don't have the code in front of me but I would guess that your board revision hasn't been programmed - f is the default value. Or perhaps it is an unrecognised number that the firmware has sanitised. Again, I'm going to have check the source.

popcornmix, 2017-03-18:
@PythonSmith can you add to config.txt
``force_board_rev=Can you add 0xa020d0``

and check if things change.

pelwell, 2017-03-18:
This will only work if your boardrev is currently unprogrammed.

PythonSmith, 2017-03-18:
The first value didnt work, bootloop with periodic ACT blink and had a lightning on the screen
The second value just blinks randomly and does blackscreen after the first color screen

PythonSmith, 2017-03-18:
I just found this table: http://elinux.org/RPi_HardwareHistory
0x9000c1 also didn't work

PythonSmith, 2017-03-19:
Is there any work around?
I tried loading "bcm2708-rpi-0-w.dtb" with U-Boot however it just hangs at Booting the kernel.
"bcm2708-rpi-b.dtb" is working without problems. I'm really happy that I have this new Pi however without its Wifi capability it is pretty useless for me. (Because it is just a Pi Zero)

pelwell, 2017-03-19:
Can you run `vcgencmd otp_dump` and post the output?

PythonSmith, 2017-03-19:
pi@raspberrypi:~ $ sudo vcgencmd otp_dump > otp_dump.txt

```
08:00000000
09:00000000
10:00000000
11:00000000
12:00000000
13:00000000
14:00000000
15:00000000
16:24280000
17:1020000a
18:1020000a
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00000000
28:f46969e7
29:0b969618
30:00000000
31:00000000
32:00000000
33:00000000
34:00000000
35:00000000
36:00000000
37:00000000
38:00000000
39:00000000
40:00000000
41:00000000
42:00000000
43:00000000
44:00000000
45:00000000
46:00000000
47:00000000
48:00000000
49:00000000
50:00000000
51:00000000
52:00000000
53:00000000
54:00000000
55:00000000
56:00000000
57:00000000
58:00000000
59:00000000
60:00000000
61:00000000
62:00000000
63:00000000
64:00000000
65:00000000
66:00000000
```

pelwell, 2017-03-19:
Thanks. As we suspected, for some reason your Pi has not been programmed correctly at manufacturing. Fortunately this is a very rare occurrence; unfortunately it has happened to you.

Your best course of action is to return it to your supplier as faulty. You will need a Return Merchandise Authorisation - you can point them at this thread if they need confirmation of the fault.

Ruffio, 2017-04-03:
@procount, @pelwell Can this issue be closed?

closed 2017-04-03 by pelwell

Jelmer-H, 2018-04-24:
I have exactly the same issue witt the same output for `vcgencmd otp_dump` with my brand new Pi Zero W. Is this a faulty batch? is there any way to program the unit manually instead of requesting an RMA?

pelwell, 2018-04-24:
Although technically possible, reprogramming in the field would require us to divulge some secret keys, so unfortunately the answer is effectively no.

blucmal, 2018-08-19:
Another with same issue will try noobs then post back

acassis, 2018-08-29:
Please help us, explain how to flash a firmware in the WiFi chip to fix it.

pelwell, 2018-08-30:
This is nothing to do with the WiFi chip - the OTP that stores the identity of the BCM2835 (B, B+, Zero, ZeroW etc.) and multiple keys has not been programmed. This is unfortunately a return-to-manufacturer fault.

Pricem586, 2019-01-01:
I have the same issue but mine has line 28 and 29 slightly different:
28: ac07a139
29: 53f85ec6
Is mine faulty too?

pelwell, 2019-01-01:
Rows 28 and 29 are the serial number and its inverse, so we would expect them to be different. This also highlights that some OTP programming has been successful, but not all, meaning that your Pi is indeed faulty (underbaked, really).

Goritmi, 2019-03-13:
I've also experiencing the same problem... But my lines at 16,27,28,29 and 30 are different...
16: 6c280000
27: 00005050
28: 6efca8b0
29: 9103574f
30: 00900c1

StephenTurley, 2019-09-09:
I'm having the same problem. Guess I have to return it. =(

StephenTurley, 2019-09-09:
.08:00000000
09:00000000
10:00000000
11:00000000
12:00000000
13:00000000
14:00000000
15:00000000
16:24280000
17:1020000a
18:1020000a
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00000000
28:90cdf785
29:6f32087a
30:00000000
31:00000000
32:00000000
33:00000000
34:00000000
35:00000000
36:00000000
37:00000000
38:00000000
39:00000000
40:00000000
41:00000000
42:00000000
43:00000000
44:00000000
45:00000000
46:00000000
47:00000000
48:00000000
49:00000000
50:00000000
51:00000000
52:00000000
53:00000000
54:00000000
55:00000000
56:00000000
57:00000000
58:00000000
59:00000000
60:00000000
61:00000000
62:00000000
63:00000000
64:00000000
65:00000000
66:00000000

For reference*

pelwell, 2019-09-10:
Yes, unfortunately.

autarctechelias, 2019-09-17:
Just got two pi zero w's from conrad about a week ago. And yep, no wifi and missing otp data.
08:00000000
09:00000000
10:00000000
11:00000000
12:00000000
13:00000000
14:00000000
15:00000000
16:24280000
17:1020000a
18:1020000a
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00005050
28:e97f1f75
29:1680e08a
30:009000c1
31:00000000
32:00000000
33:00000000
34:00000000
35:00000000
36:00000000
37:00000000
38:00000000
39:00000000
40:00000000
41:00000000
42:00000000
43:00000000
44:00000000
45:00000000
46:00000000
47:00000000
48:00000000
49:00000000
50:00000000
51:00000000
52:00000000
53:00000000
54:00000000
55:00000000
56:00000000
57:00000000
58:00000000
59:00000000
60:00000000
61:00000000
62:00000000
63:00000000
64:00000000
65:00000000
66:00000000

mrouillard, 2019-11-19:
Dear @pelwell, would you mind looking at my data below? I fear that I have the same issue as @PythonSmith (for the second time in a row, a brand new PiZeroW I just received):
cpuinfo:
`
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000005d48d8be
Model		: Raspberry Pi Zero W Rev 1.1
`
vcgencmd otp_dump:
`
08:00000000
09:00000000
10:00000000
11:00000000
12:00000000
13:00000000
14:00000000
15:00000000
16:24280000
17:1020000a
18:1020000a
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00005050
28:5d48d8be
29:a2b72741
30:009000c1
31:00000000
32:00000000
33:00000000
34:00000000
35:00000000
36:00000000
37:00000000
38:00000000
39:00000000
40:00000000
41:00000000
42:00000000
43:00000000
44:00000000
45:00000000
46:00000000
47:00000000
48:00000000
49:00000000
50:00000000
51:00000000
52:00000000
53:00000000
54:00000000
55:00000000
56:00000000
57:00000000
58:00000000
59:00000000
60:00000000
61:00000000
62:00000000
63:00000000
64:00000000
65:00000000
66:00000000
`

pelwell, 2019-11-19:
@mrouillard Your OTP looks fine to me. The key indicator is row 30 which holds the revision code. Yours says 900c1, which is a rev 1.1 Zero W, but /proc/cpuinfo already told you that.

What symptoms are you seeing to make you think there was a problem?

mrouillard, 2019-11-19:
Thanks for the quick answer!
The symptom was being able to list wifi networks using "sudo iwlist wlan0 scan" but never connecting to any of them despite two of them being set in my wpa_supplicant.conf.

However, in the meantime I found an issue in the wpa_supplicant.conf file that I had put on the sdcard: one SSID was enclosed by ” instead of straight double quotes. Argh. I feel ashamed.
So actually, I have only one faulty Pi Zero W (the oldest I had). The brand new one is fine. My bad, sorry for the trouble, and thanks again for your reactivity.

Ben1701A, 2019-12-23:
Lost all wireless capabilities on my Pi 0 W just today.

Had to type this one out, didn’t have a USB-Ethernet on hand.
`08-15: 00000000
16: 24280000
17: 1020000a
18: 1020000a
19-26: ffffffff
27: 00005050
28: bb635847
29: 449ca7b8
30: 009000c1
31-66: 00000000`

pelwell, 2019-12-23:
That also looks normal.

Ben1701A, 2019-12-23:
Ok, Thanks for the fast reply. Will continue my journey for a solution.

jkober, 2021-11-27:
Hello, my raspberry pi zero does not recognize the camera "raspberry pi camera rev 1.3" I have raspbian lite installed

pelwell, 2021-11-30:
This issue was closed nearly 2 years ago. Open a new issue, explaining what you have done, what you are expecting to happen and what actually happens.