raspberrypi/firmware #759: git pull problem.

MrEngman, 2017-03-03:
I have a clone of the `raspberrypi/firmware` repo. Just tried git pull to update it and it's come up with an odd problem that I don't know how to fix.

git pull initially shows
```
git pull
remote: Counting objects: 3, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/raspberrypi/firmware
 + 78bcdea...78c4983 master     -> origin/master  (forced update)
```
but then the nano editor opens with the following text.
```
  GNU nano 2.2.6                                         File: .git/MERGE_MSG

Merge branch 'master' of https://github.com/raspberrypi/firmware

# Please enter a commit message to explain why this merge is necessary,
# especially if it merges an updated upstream into a topic branch.
#
# Lines starting with '#' will be ignored, and an empty message aborts
# the commit.
```
What's happening and why? Is something broken? I usually `git pull` every time there is a new commit and haven't seen this before.

pelwell, 2017-03-03:
I did a force push to tidy up a commit to the wrong location via the GUI, but had to wait for the time it took to clone the repo. `git reset --hard origin/master` should fix it.

MrEngman, 2017-03-03:
Thanks, looks like it's fixed. `ls -l firmware` showed
```
-rw-r--r-- 1 pi pi 20195 Mar  3 13:56 bcm2708-rpi-0-w.dtb
drwxr-xr-x 3 pi pi  4096 Mar  3 16:50 boot
drwxr-xr-x 3 pi pi  4096 May  2  2014 documentation
drwxr-xr-x 2 pi pi  4096 Mar  2 11:12 extra
drwxr-xr-x 3 pi pi  4096 May  2  2014 hardfp
drwxr-xr-x 4 pi pi  4096 Feb 27 00:57 modules
drwxr-xr-x 3 pi pi  4096 May  2  2014 opt
-rw-r--r-- 1 pi pi   897 Mar  2 11:12 README
```
so bcm2708-rpi-0-w.dtb looks to be in the wrong place. After running the command you suggest it now shows
```
drwxr-xr-x 3 pi pi 4096 Mar  3 16:50 boot
drwxr-xr-x 3 pi pi 4096 May  2  2014 documentation
drwxr-xr-x 2 pi pi 4096 Mar  2 11:12 extra
drwxr-xr-x 3 pi pi 4096 May  2  2014 hardfp
drwxr-xr-x 4 pi pi 4096 Feb 27 00:57 modules
drwxr-xr-x 3 pi pi 4096 May  2  2014 opt
-rw-r--r-- 1 pi pi  897 Mar  2 11:12 README

```
So presumably it's fixed and file `bcm2708-rpi-0-w.dtb` is now in the ./boot directory so should be in the right place.

popcornmix, 2017-03-05:
> I usually git pull every time there is a new commit and haven't seen this before

Using ``git pull --rebase`` is recommended. It would cope with this situation, and doesn't have any downsides for the normal situation.

JamesH65, 2017-03-25:
@MrEngman Can this be closed?

MrEngman, 2017-03-25:
Of course.

closed 2017-03-25 by MrEngman