raspberrypi/firmware #760: V2 Camera: Firmware possibly not doing a full reset of camera on soft reboots

Electron752, 2017-03-08:
I'm doing some testing of the camera with the upstream version, and I sometimes see issues with connecting to the camera(V2).

A sequence like this is essentially 100% reliable:
1. Power up RPI 3(32 bit mode) 
2. Connect to camera or query camera info with mmal applications in userlands
3. modprobe bcm2835-v4l2
4. Connect with application like cheese or ffmpeg

But if I do a soft reset of the RPI(normal sudo reboot), I will often see a complete machine lockup on the modprobe of the V4L driver next time I connect. But as I say, pulling the plug and leaving it off for awhile seems to work quite well.

I'm wondering if the firmware isn't always completely doing a full reset of the camera on reboots. I'm also wondering if running some of the mmal applications from userland isn't sometimes forcing a better reset then would happen if V4L was loaded without any other steps.

Here is my firmware information:
[    3.163532] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-02-20 14:01

Here are the camera entries I added to config.txt:
start_x=1
gpu_mem=256

6by9, 2017-03-08:
It's pulling the power from the camera module - that's what one of the GPIO control lines does.
You can't get a more comprehensive reset than that.
Now admittedly on a Pi3 the GPIO is on the I2C expander, but that is reset to a default state (camera powered down) at boot.

MMAL from userland will create the camera component as it runs, and destroy it on exit.
V4L2 is "persistent" as the kernel hasn't gone away as the client. It gives you more chance for memory leaks, but otherwise should be as resilient.

Electron752, 2017-03-08:
Electronics hardware is my secondary area of interest, so I know from experience that sometimes pulling the power can be worse then a hard reset pin. Especially if the power isn't pulled long enough.

As you know, the camera probably has some capacitors and when the power is pulled they are probably going to drain exponentially(so they never completely discharge).  If the voltage doesn't drop enough or the camera doesn't have a proper BOR component, then it may actually be only half reset and in a scrambled state.

6by9, 2017-03-08:
I've just been through 4 soft boots with no issues (I used qv4l2, but it makes no difference what the client app is), and it's not something I have ever seen before.

Having studied the schematics it all looks correct. There is a PWDN signal to the sensor itself, and that is controlled off a logical place to provide a clean reset.

Is this on a clean kernel tree or with modifications?
There is a logged issue at https://github.com/raspberrypi/linux/issues/1447 of a kernel oops which has only ever shown up against ArchLinux. If you've rebuilt the kernel then it may be that, but you've not given us anything to go on.

At the moment you've hypothesised an issue based on no real evidence.

Electron752, 2017-03-08:
This is with a clean upstream tree(4.11+).  Other have made some small modifications, but they have mostly been cosmetic and formating issues.

Let's leave the issue open for a bit and I'll see if I can provide clearer and more consistent reproduction instructions for the issue.

In the meantime, if I want to get some clear logging on the lockup I'll probably need to attach a serial cable.  Do you have any brief recommendations for anything to look for in terms of logging?  I can add logging to the camera or vc04_services, but since the entire RPI seems to lock up I think it might be happening in the firmware when it initializes the camera.

Electron752, 2017-03-08:
OK, I have some more specific information on what is happening. The following conditions must ALL be meet to hit the issue:
1. The RPI(RPI 3 in my case) MUST be network booted(bootcode.bin only on SD card)
2. NO userland applications can be run that call bcm_host_init.(Running raspistill with no parameters wil call bcm_host_init). So essentially nothing in the userland tree can be run.
3. modprobe bcm2835-v4l2 must be run.

At this point, the RPI will start having catastrophic failures and systems will all go down including USB.  It may be power is being pulled to the hardware or some clock is messed up. I don't know.

But booting from a SD card with a complete firmware will work just fine.   And like I said, running anything that calls bcm_host_init is enough to put the firmware into a happy enough state that the failure doesn't happen if modprobe bcm2835-v4l2 is run.

The failures all happen rather soon after bcm2835-v4l2 finishes the modprobe sequence even if nothing actually opens the camera.

I saw Eric Anholt( @anholt ) complain to someone either through e-mail or on his blog that the network went down on him during his initial tests after submitting the camera driver to upstream.  So he may have actually been network booting as well(or perhaps using u-boot) and may have also hit the same issue.

Hope this helps!

6by9, 2017-03-09:
If you're runnning upstream then I'm assuming all of Eric's code is active, including controlling the QPUs from the ARMs. That's going to be an issue in itself as part of the camera stack is using the QPUs from the VPU for stats analysis. I need to revisit that and probably drop to an alternate algorithm if using vc4-fkms-v3d or vc4-kms-v3d.

As to the modprobe issue, from a quick look there appears not to be an equivalent call to bcm_host_init from the V4L2 driver initialisation. I'd need to research what that is actually setting up on either ARM or VPU side.
Why it requires net boot is also a curious one. That may implicate clock or power domain issues. I haven't set myself up with net booting yet, so it will be a while before I can investigate.

anholt, 2017-03-09:
@6by9 would you be interested in building a messaging channel for the firmware to request that Linux do a QPU job?  That could then get us working H264 as well.  https://github.com/anholt/linux/tree/rpi-4.4.y-qpu-execute has some code for running a QPU job on the Linux side, and we'd need to hook something up in the mailbox driver (I guess?  Or just use VCHI?) for passing a message to Linux describing the job to be run.

Electron752, 2017-03-09:
@6by9: Actually, none of Eric's code is running.  Since I have a RPI 3 and HDMI hotplug doesn't work, I simply disable VC4 from the build configuration.   This tricks the firmware into using simplefb, which surprisingly isn't much slower on a RPI 3 in 64 mode mode then the downstream frame buffer driver.

Electron752, 2017-03-09:
@anholt and @6b9 - I would personally like to see a third transport built.  Something much smaller then vc04_services but a bit more complex then mbox. I'm thinking something that could be written in say 2,000 lines of clean C code. Then phase out both vchi and mbox.

I know many of people that are active on the RPI are or were associated with Broadcom at one time.  Hopefully I'm not personally insulting someone who was personally involved in designing vchi.

6by9, 2017-03-09:
@anholt I need to talk to Gordon and others about the best approach. A VPU->ARM QPU job dispatcher would be one option. I need to look at what the algorithm is actually doing - it was only on the QPU as the VPU doesn't have vector floating point, but it may be practical to do fixed point. I honestly don't know at the moment. I'll open a new issue as a reminder, but it may end up being just a firmware job.
The short term fix is to make sure that the VPU doesn't try to dispatch the QPU job, but we do need it to do something instead.

@Electron752 What I'm trying to say is that I'm not convinced that you're going to get an easy run with the camera anyway. 
I'll try to grab Phil or Gordon tomorrow to try and understand what is going on with the relevant calls.
There should be a release tomorrow that should fix expgpio to read the raw GPIO values and fix hotplug.

Electron752, 2017-03-09:
@6b9 I was able to get the camera to work on an arm64 kernel last night.  I'm still tracking an oops which I think is happening in V4L2, but the basic functionality works just fine.

Electron752, 2017-03-09:
And series 1 of my fixes(which I thought would be the hardest to get in), was merged by the stagging maintainer last night. I just have to polish up series 2.

Series 1 uses a mapping table for that context field that gets passed back and forth.  If It turns out to be too slow for downstream on 32 bit, either I or your guys not no-op out the mapping table operations.  

Adding the mapping table also fixes a few of the Todo comments.

6by9, 2017-03-09:
> I would personally like to see a third transport built.

VCHI may have faults but replacing it requires rewriting all the services that use it. There was an RPMSG based server for MMAL, but even that would require extra work as it didn't support any form of bulk transfer. All the other services would need porting for relatively little gain.

Electron752, 2017-03-09:
That the problem.  No end user gain.  It's just that VCHI is a maintenance nightmare.

Electron752, 2017-03-10:
@6by9 

FYI: I submitted the remaining necessary changes to get the change to work with arm64.  Since linux-next and the staging tree seem to have a bunch of extra debugging checks turned on by default and I'm using simplefb, it hard to judge what kind of performance I should be getting.  

I used a very simple canned mapping table implementation that already exists in the kernel tree to increase the chance of the changes being accepted.   If this turns out to be way too slow on 32 bit in downstream, either the table can be optimized or it can be no-oped out very easily to just assume that pointers and handles are identical.

Electron752, 2017-03-17:
@6by9 and @anholt 

Thinking longer term, perhaps such a vchiq dispatcher from the firmware isn't such a a bad idea.  The kernel registers a service with VCHIQ that accepts requests from the firmware(I think VCHIQ is bidirectional) Ideally the firmware should have a compatible service that gets reqistered on it's version of VCHIQ. 

Thinking longer term, I could image a day when through the DT it is actually selectable which side is handling the service.  And the kernel version of VCHIQ could have a short circuit path that bypasses the firmware if the service is either a kernel driver or user mode application.  

Depending on various things, for some RPI it may be better to let the firmware handling things and for some RPI it may be better to let the kernel or user mode handle things.

Electron752, 2017-03-17:
@anholt I noticed @popcornmix has been checking in some changes for a QPU dispatcher based on VCHIQ.  I don't know if it's functional or not, but I wonder if a really easy way to make progress would be to write an upstream service that registers with VCHIQ that accepts the QPU dispatch jobs and somehow kicks then over to the VC4 driver.

popcornmix, 2017-03-17:
The vchiq gpu server runs on the firmware and allows the arm to request VPU and QPU jobs to be executed by the GPU. It is basically an enhancement to the mailbox interface intended to support HEVC acceleration in kodi.
Obviously in the vc4 graphics driver world the QPU jobs won't be submitted to the GPU, but to the arm (e.g. using @anholt's qpu ioctl).
In the vc4 driver world there aren't many users of the QPUs from firmware. There is the AWB algorithm of camera driver, which we have now added a fallback to a simpler, non-QPU scheme when unavailable.
There is also a QPU deinterlacer which will need a new way of dispatching. But apart from those I can't think of any firmware side users, so an arm side vchiq service may be overkill

Electron752, 2017-03-17:
I think @anholt's original intention was that maybe someone wants to use VC4 most of the time, but they occasionally loaded up a movie player.  Obviously, the video decoding is going to stay on the firmware for quite a long time, but it would be nice if it worked on the few occasions that it's loaded.  One way to do that would be for the firmware to kick the QPU jobs back onto the ARM side.

I'm thinking of a very simple VCHIQ based service that would just send the stuff to @anholt's qpu ioctl.

Terminus-IMRC, 2017-04-07:
@popcornmix Could you tell me how to enable non-QPU-based AWB?

6by9, 2017-04-07:
It's automatic on firmware since 18th March / https://github.com/raspberrypi/firmware/commit/384559354762f36aa55584560d8749fc66a4cfd0. If either the v3d device tree node is enabled, or you've otherwise disabled the VPU interrupt for v3d through config.txt, then it will use the non-QPU algorithm.

Terminus-IMRC, 2017-04-07:
Thanks. But I couldn't find a config.txt option to disable the VPU interrupt for V3D. (`mask_gpu_interrupt[01]`?)
Could you please provide an option to do that more easily?

6by9, 2017-04-07:
```mask_gpu_interrupt0=1024```

But you're approaching this from the wrong side. This is disabling ALL access to the QPUs from the VPUs, including the QPU mailbox service. It is not a user-selectable AWB algorithm switch.

Please stick to #793 for discussion rather than spreading between here and there.

Terminus-IMRC, 2017-04-07:
I see. Thank you.

Ruffio, 2017-04-09:
@Electron752 I can't figure out the status of this issue. Can it be closed?

closed 2018-10-29 by JamesH65