raspberrypi/firmware #761: Fallback config needed while overclock in unattended mode.

kirgene, 2017-03-11:
There are several RPi  v1 B devices that are working in headless unattended mode (without physical access to them, only ssh).
Now I need to set some overclock options that will include `force_turbo=1` and I'm afraid that some of devices won't boot at all after such operation.
It would be great to have a fallback config that device could load in case of previous boot fails.

Is there any way to achieve such behavior now or are there any plans to develop such feature?

pelwell, 2017-03-11:
Detecting a boot failure sounds tricky, but I can think of an alternative - a one-time boot configuration. This could be implemented in a number of ways - a section ("[once]") or entire config file ("config_once.txt"). The only way to reliably prevent them being used a second time is to rename them - either the section or the file - just before launching the kernel, which is slightly bothering because the firmware doesn't normally modify the boot partition. This scheme would also fail if the failure is so catastrophic that the firmware won't run.

There used to be a mechanism to select an alternative configuration using a GPIO, which might have been useful with external hardware support, but that feature is no longer supported.

popcornmix, 2017-03-12:
I'm not keen on firmware writing to sdcard - it never has previously.
This sounds like it could all be handled on the arm side:
-  don't use force_turbo
- default to powersave cpufreq governor (the default in raspbian kernel)
- remove switch to ondemand governor (in ``/etc/init.d/raspi-config``)
- Enable the watchdog.
- Switch to performance governor (same effect as force_turbo) if previous boot wasn't a watchdog reset

You can detect a watchdog reset with ``vcgencmd get_rsts``

JamesH65, 2017-12-22:
@kirgene Did the suggestion above work? Can this be closed?

closed 2019-11-05 by JamesH65