raspberrypi/firmware #762: HDMI and Analog audio at the same time through firmware

Electron752, 2017-03-13:
Does the firmware by any chance support playing back audio through HDMI and Analog at the same time?  I have no idea why anybody would want this, but pluseaudio supports combining multiple independent devices into a virtual audio devices.

The reason I'm asking is that with ARM based HDMI Audio and Analog audio drivers being submitting this would now be possible.

I'm looking into how to implement the necessary interlocks between the various devices(HDMI, PWM, Analog Audio, and Firmware).  Things are just easier if the firmware output devices could be treated as logically independent.

Does any software actually support the route property on the firmware based driver?  I thought something like alsamixer or pulseaudio should support selecting the output devices, but I couldn't get it to work.

popcornmix, 2017-03-13:
Yes, firmware supports output of HDMI and Analogue at the same time. e.g.
```
sudo amixer cset numid=3 2
speaker-test # will output through hdmi
/opt/vc/src/hello_pi/hello_audio # will output through analogue
```
This uses alsa for hdmi and openmax for analogue.
The problem with alsa is the amixer switch only allows one destination at a time, but the audio service would support both if it were requested.

Electron752, 2017-03-13:
Ok, I think I understand now.

Electron752, 2017-03-14:
OK, I submitted the change to make have alsa expose multiple virtual sound cards.  I'm hoping it will be accepted.

The way I setup the defaults actually does break the customized mixer and pulseaudio in Raspbian since it expects to see the old style card otherwise it assumes audio is disabled.

To add back the old card add as a module parameter:
snd-bcm2835.enable_compat_alsa=1

To disable the HDMI and Headphone specific virtual cards:
snd-bcm2835.enable_hdmi=0
snd-bcm2835.enable_headphones=0

I assuming either the default cmdline.txt can be modified, the firmware/bootloader can pass this, or as a last resort the defaults in the source can be patched in the downstream version.

Please let me know if this is a problem before the changes get merged in upstream.

pelwell, 2017-03-14:
Having the default concussion break the biggest OS on Raspberry Pi is an interesting choice, but I'm sure @spl237 may have more to say about it.

pelwell, 2017-03-14:
s/concussion/configuration/ - gesture keyboards for the win.

Electron752, 2017-03-14:
It's not written in stone yet, and I can always send a V3 or additional changes.  

I wasn't sure what the default should be.  Upstream seems to be more targeted toward third party OSs, and pulse an plain debian works best with the defaults I used.  I was thinking it wouldn't be hard to change the defaults in the downstream version.

Electron752, 2017-03-14:
On a related topic, I also noticed today that the firmware is turning off some of it's features when using the simplefb trick.  This means using a pure upstream built on the pi3 in it's current form doesn't work very well.

pelwell, 2017-03-14:
That sounds like a new issue.

Electron752, 2017-03-14:
Minecraft_pi doesn't work through simplefb even though all the needed vchiq layers are now working.  I was wondering if that was intentional.

Electron752, 2017-03-14:
But then it could just be the version of the firmware that happened to be on the Raspbian image I loaded up.

Electron752, 2017-03-14:
Anyway, if the default audio settings are an issue let me know what they should be and I'll send out a V3 before it gets applied.

I originally had the setting as a compile time option but a few people didn't like it that way.

pelwell, 2017-03-14:
That's a "New issue". This one's unrelated, and closed.

Electron752, 2017-03-14:
Should I reopen the issue so that people can tell me what the defaults should be for the audio?

pelwell, 2017-03-14:
Issues usually remain open until they have been completely addressed. It makes it easier for people to find current issues, and stops them dropping off the radar prematurely

Electron752, 2017-03-14:
OK, I'll reopen it.

popcornmix, 2017-03-14:
The simplest option to have raspbian use the updated driver is to make the default behaviour the same.
If you believe a different behaviour is beneficial (e.g. to allow concurrent analogue and hdmi) then we can consider it, but it is likely to (temporarily) break a large number of user's installations, so there will be some reluctance.

Can you describe what the planned behaviour is, e.g.
Where does audio go with the default device?
How should a user switch between devices?

Even if ``snd-bcm2835.enable_compat_alsa=0`` is not the default, but is an option, then supporting the virtual cards with raspbian/pixel would be desirable. Can you link to a patch that applies downstream (rpi-4.11.y should be okay - the upstream staging driver did appear to work on a quick test) and I'll try to get @spl237 to look into supporting it.

Electron752, 2017-03-14:
This is the latest version I posted, but since the driver has been going through heavy cleanup through small changes, I doubt it will apply directly.
http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-March/005771.html

Raspbian does actually work the way you would expect if all 3 cards are present(HDMI, Headphones, and Old Combo Style).  In fact, you can even manually select the destination which I don't believe worked well before.

The issue is that if the card "bcm2835 ALSA" isn't around, everything assumes audio has been disabled in config.txt even though the other cards are perfectly selectable and in fact someone could have even plugged in a USB audio device to get to this point.

Like I said I have no problem changing the defaults.  I just wasn't sure what the defaults should be.

popcornmix, 2017-03-14:
> In fact, you can even manually select the destination which I don't believe worked well before.

I believe it has always worked fine. The default option automatically uses the "obvious" device - hdmi for a hdmi display which reports audio capability in its edid and analogue otherwise:
```
amixer cset numid=3 0
```
You can force analogue:
```
amixer cset numid=3 1
```
and you can force hdmi
```
amixer cset numid=3 2
```
(or use the gui options in raspbian/pixel).
This implementation treats the output device as a switch. You can't drive both at once.

Electron752, 2017-03-14:
I guess that was the problem.  Most people won't use amixer directly.   Having the extra cards means in Raspbian someone can quickly click the audio icon in the top right and use the popup to switch.  I don't believe it worked before.

Perhaps I should just send out a V3 that keeps things they way they were by default for now, and people can enable the extra cards manually.

BTW, pulseaudio on other OSs was completely broken before.  It only worked with auto.  I couldn't get either pulse or alsamixer to easily change the settings.  That all works quite well now.

Electron752, 2017-03-14:
And of course throwing ARM based devices in the mix really complicates things.  The ALSA people were ignoring the driver I sent, but like 10 minutes after I enhanced the firmware based driver it the ARM based driver quickly got people's attention.

spl237, 2017-03-14:
There's no way I'm accepting any changes that break the current PulseAudio/ALSA interaction for something which is only a minority requirement, in that it has not been possible to have simultaneous HDMI and analogue audio for five years now, and I can't think of anyone who's asked for it in that time...

I'm investigating a possible solution to having to use PulseAudio for Bluetooth audio output - if that works, we might be able to consider this thereafter.

popcornmix, 2017-03-14:
I've added your patch to rpi-4.11.y. ``With snd-bcm2835.enable_compat_alsa=1``
I now get:
```
pi@domnfs:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [bcm2835 HDMI], device 0: bcm2835 HDMI [bcm2835 HDMI]
  Subdevices: 3/3
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 2: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 2: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I don't think raspbian/pixel detects it (volume control is shown as muted and can't be adjusted).

Electron752, 2017-03-14:
Let me think  on that.  It was working for me.

So I can accept that the old behavior is required. So that leaves two questions:

1. Should I change the default to be compatible with Raspbian by default?
2. Or should the downstream version be patched/cmdline.txt modified to provide the old style behavior?

Electron752, 2017-03-14:
BTW, this all leads to the more complex issue of if a pure upstream really should work well for Raspbian.  I would think most of the people that Raspbian is intended for arn't going to even think or care about running Raspbian.  The complexity happen when adding embedded and advanced uses like the compute module.

spl237, 2017-03-14:
"I don't think raspbian/pixel detects it (volume control is shown as muted and can't be adjusted)."

No, it won't - it is only aware of the two cards with device 0 on both for the internal hardware; I'm not sure what it'll do with device 1 on an internal card, but I doubt it'll work...

Electron752, 2017-03-14:
@spl237 I think I had it working with all 3 after I manually enabled the other cards in the advance audio options dialog.

popcornmix, 2017-03-14:
@spl237 This isn't really about us accepting something.
This is what is going on in the upstream kernel, and at some point in the future we'll have to decide whether to work the way upstream works, or patch it to maintain the same behaviour as before.
Obviously finding a solution that both are happy with is ideal.

@Electron752 Patching the actual cmdline.txt file would be undesirable - users may have customised that and so so adjustments are tricky. The firmware can (and does) add to cmdline.txt file for additional options, although the firmware doesn't know about the version of kernel being run, so may be adding these options when not required. Is device tree not a better way of configuring this?

popcornmix, 2017-03-14:
Ah, adding all the options:
``snd-bcm2835.enable_compat_alsa=1 snd-bcm2835.enable_hdmi=0 snd-bcm2835.enable_headphones=0``
does make the raspbian/pixel volume control usable again.
```
pi@domnfs:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I guess it's the virtual devices appearing before the ususal ones that is causing the problem.

Electron752, 2017-03-14:
The device tree would also work, just that the device tree for sound/vchiq is undocumented in upstream at the momement and I was advised not to try to document it.  Add more options is going to raise issues.

Like I said maybe the default should just be the old behavior for now.

popcornmix, 2017-03-14:
Defaulting to old behaviour would be simplest for us.

Electron752, 2017-03-14:
I think that's what I'm going to do for now.

I still think it would be useful to know if pure upstream really should work well though.  Because like I said most people that are going to use upstream are not going to be using Raspbian.

popcornmix, 2017-03-14:
In an ideal world raspbian would just be using the upstream kernel.
Realistically that probably won't completely happen (e.g. fiqs and dwc_otg), but we are a lot closer with 4.9 (and 4.11 is closer still).
If upstream drivers do work with raspbian then we can use them without downstream patches, which is good news for downstream and upstream.
Sure, the number of raspbian users using an untouched upstream kernel may be small, but the hope is there will be very many raspbian users using a downstream kernel that is only lightly touched compared to upstream.

Electron752, 2017-03-14:
OK, I'll send V3 using the excuse that it seems more testing before it can be made the default.  But since GregK can apply whatever version he feels like applying or someone else can submit a change that changes the default, I can't make guarantee anything.  

I'll open a separate issue in awhile on the issue of why simplefb doesn't work with vchiq based graphics.

closed 2017-03-14 by Electron752