raspberrypi/firmware #763: Firmware possibly disabling features/services when using simplefb

Electron752, 2017-03-15:
Since VC4 on the PI3 doesn't work great at the moment without modifying VC4 source code to force HDMI to be enabled(due to the expander issue), I've often been testing patches that I've been submitting to upstream  by removing VC4 from the build configuration and letting the firmware patch the device tree to load simplefb.  The only problem is that it appears that when using simplefb, the firmware is disabling a bunch of features making Raspbian not particularly usable on a pure upstream kernel.

I'm wondering if this makes sense.  I realize that very few people who are interested in Raspbian are going to use a pure upstream kernel, but at the same time I understand it is a goal to use as much upstream code as possible and the video driver is one of the last pieces.

I know about a year ago when I first started tinkering with the Linux kernel on the RPI, frame buffer based video was completely unusable in 1920x1080 without the DMA acceleration.  Today, simplefb appears to have advanced and when using it on a RPI 3 either in 32 bit or 64 the performance is quite good. And for the lower end RPI, I'm wondering if a better solution would be for the upstream code to have general DMA support added as the dmaengine framework does now support memory to memory copies in some cases.

I'm also wondering if perhaps the general DMA copy functionality of the downstream frame buffer video driver could just as well be moved to a separate small driver.

pelwell, 2017-03-15:
>  the firmware is disabling a bunch of features making Raspbian not particularly usable on a pure upstream kernel

Which features are being disabled? Please give more details about your configuration, what you expect to see and what actually happens.

Electron752, 2017-03-15:
Specifically I'm talking about minecraft_pi.  When I run the application from an upstream build using the upstream device tree, the application doesn't start and I see a message in dmesg about a videocore service not being available.   The same thing happens when running most of the /opt/vc applications such as the raspicam based applications.  All this works if I run from a downstream github tree with a github device tree+upstream stagging patches.

I'm doing most of my testing on arm64 and yes I know that isn't supported at all, but it seems to me the only difference is the DT and the video driver. Granted I only have surface knowledge, but it would seem to me that all the pieces are in place to allow all of these apps to just work from upstream now.  

All that seems to still be broken at this point in upstream is graphics rendering related.  Video Camera works with V4L2 driver, audio works, etc.  It's just rendering to the screen that appears to be disabled.

pelwell, 2017-03-15:
Run `dtc -I fs /proc/device-tree > upstream.dts` and upload the resulting file to a gist, pastebin etc.

Electron752, 2017-03-15:
https://gist.github.com/Electron752/97eb5d5d81f09c8d7eff1949049b64a1

Electron752, 2017-03-15:
An important point is the VC4 is not in the kernel build, but simplefb is.

pelwell, 2017-03-15:
Your dtb doesn't include a "serial-number" property, which makes you firmware older than October last year. Before I dig any deeper can you update to current firmware and repeat the test - if it still fails, dump a new upstream.dts.

Electron752, 2017-03-15:
OK, give me a few minutes.

Electron752, 2017-03-15:
Sorry, it took a long time to get the latest firmware.  Like I said it's exiting with a message about a service not being available.

https://gist.github.com/Electron752/97eb5d5d81f09c8d7eff1949049b64a1

pelwell, 2017-03-15:
Can you include the precise error message?

The firmware is getting information about the ARM-implemented services from the dtb. It looks for the interrupts claimed by certain non-disabled nodes, and if found it disables the corresponding interrupt handling on the VPU (and in some cases does a bit more than that). In your dtb you have enabled nodes for `v3d`, `hdmi`, `pixelvalve`, and `hvs`, so the firmware will not handle those interrupts.

If you don't want the vc4 driver to be loaded then disable the v3d node.

By the way, the GPIO expander driver in the (very) latest kernel and firmware is working now, so hotplug shouldn't be an issue in the future.

Electron752, 2017-03-15:
OK, thanks. That makes sense.

The piece I'm missing here though is that if VC4 does actual load and work, I wouldn't be able to run minecraft_pi or as far as I know the other RPI specific apps because they are going to be talking to videocore graphics components which wouldn't be compatible with VC4.

I'm simply foreward thinking a bit in what if it were possible to get these apps to work in the case that VC4 isn't enabled. Shouldn't simplefb provide enough of what the downstream video driver does to make these apps work.  If I disable VC4 in the DT, wouldn't simplefb not load?

pelwell, 2017-03-15:
> The piece I'm missing here though is that if VC4 does actual load and work, I wouldn't be able to run minecraft_pi...

I am not saying that VC4 is running, I'm saying that the firmware thinks that the ARM has drivers for v3d, hdmi, pixelvalve, and hvs. If you have disabled any of these drivers, you have done so in a way that the firmware can't detect - you need to disable the corresponding nodes in the dtb.

Minecraft_pi uses the dispmanx interface, which is disabled if the hvs driver is running on the ARM. If you want Minecraft and the VC4 driver you need to use the firmwarekms driver instead which works through dispmanx rather than replacing it.

Electron752, 2017-03-15:
I guess I'm making this more complex then it is. firmwarekms is not available upstream and I'm guessing it probably won't be for some time if ever.

What I'm wondering if why couldn't simplefb be a easy replacement for the downstream video driver that is upstream only.  Yes it isn't as fast, but I would think it would also be getting it's fb from dispmanx just like firmwarekms or the traditional downstream video driver.

Electron752, 2017-03-15:
BTW, was anybody considered upstreaming the expander driver?  I know Eric is typically busy and I'm still learning things.  It would make it possible to have VC4 in upstream though.

Electron752, 2017-03-15:
Finally thought, I was doing some thinking today on how something similar to firmwarekms could be written in upstream.  Perhaps something like Tinydrm with custom frame buffer ops rather then going through the typical helpers.  But that's really looking ahead.

pelwell, 2017-03-15:
Can you state clearly what is stopping you using simplefb as a replacement for bcm2708_fb?

The firmware currently creates and initialises the `framebuffer@...` node if the ARM-side HVS node is enabled., i.e. the real kms/no dispmanx case. If this isn't sufficient, it would be simple and safe to get the firmware to also check for the presence of a `framebuffer@` node and initialise it if found.

Electron752, 2017-03-15:
That's exactly what I was thinking.

Someone loads a DT with no ARM-side HVS node(or it's disabled) but a framebuffer node is present.  The firmware detects this, initialize the fb with whatever compat string(either use the one already in the dt or create one), but yet still keeps all of dispmanx active in the case someone loads up one of the RPI specific applications.(They are running Raspbian).

I have discussed with Eric about the potential of hacking on the real VC4 to get a very simple display on say the RPI zero that is only occasionally used, but he didn't think that would be easy to do with the existing code base(this is sort of an aside).

pelwell, 2017-03-15:
I think the firmware logic should be extended to also create the simple framebuffer if an enabled `fb` node is not found (i.e. it is disabled or absent). This will automatically result in a frame buffer with an upstream dtb.

Electron752, 2017-03-15:
That makes sense to me. The only issue I can think of is if someone is intentionally trying to disable the disable on say an embedded system for security reasons.  I'm not sure how that should be handled.

pelwell, 2017-03-15:
Blacklisting the driver would take care of that requirement.

pelwell, 2017-03-15:
...except it's compiled in.

Electron752, 2017-03-15:
Sounds like a plan.  I may even be able to see about getting an alternative DT for very simple systems documented and submitted to upstream.

Electron752, 2017-03-15:
Yah, I forgot about that.  You can't blacklist something until root is mounted.

pelwell, 2017-03-15:
Try this: if you don't want a frame buffer, create a disabled `/soc/framebuffer`. You can also create a disabled `/soc/fb` for the sake of appearances but it isn't required.

Electron752, 2017-03-15:
I think that would work.  If framebuffer already exists but is explicitly disabled, then leave it disabled.

Electron752, 2017-03-15:
So if things are ok, I'll go ahead and close out the issue.  I've been using my e-mail for patches and I know people how been checking the e-mail list so someone can ping me when something is available to try out.

You guys could probably just use the already existing simplefb driver for testing and debugging until a DT can be created and documented.

closed 2017-03-15 by Electron752

pelwell, 2017-03-15:
I have a firmware patch that implements the proposal with one difference - rather than checking for a `framebuffer` node it looks for a `display0` alias, which if present will prevent the simple framebuffer from being created. The intention is that any existing display referred to by the alias, or an explicitly empty alias, will disable simplefb.

Electron752, 2017-03-15:
Cool. Can I get a test version to try out?

pelwell, 2017-03-16:
You can grab the elfs and dats here: https://drive.google.com/file/d/0B_P-i4u-SLBXLVFINGZZeWp3bGs/view?usp=sharing

Electron752, 2017-03-19:
I'll give it a whirl.  Just to let you know Eric is against an enhanced simplefb, so I would consider this a short term solution that gets things going.

I opened another topic on the firmware which is a better long term approach.  If the purpose of the firmware is to protect IP(which is a completely valid reason), I think too much munching of the DT isn't good.

Just look at what a disaster the whole virtual expander is becoming....

pelwell, 2017-03-19:
Too late - it's already hit the firmware repos.

I don't think the gpio expander driver is anything close to a disaster (although some of your upstreaming modifications look unnecessary), and language like that will gain you no friends.

Electron752, 2017-03-19:
Sorry,

I don't think the version on the firmware repos will do any harm. I'm planing to test it out soon.  I think it's a step in the right direction, but I don't think it's a good long term solution.

The expander has been submitted, we'll have to see what happens to it. That's something I have no control over.

If the goal of the firmware is to protect IP, that needs to done properly.  Trying to add obfuscation layers in public source code is not the way to go.

pelwell, 2017-03-19:
Please give an example of obfuscation in the expander driver, as you see it.

Electron752, 2017-03-19:
This is totally my view, but I think adding +128 to the offset was sort of an obfuscation.  It wouldn't take people very long to figure out that the mailbox API was a general mechanism for controlling pins.  In fact, I found out the version Eric originally submitted added most of the exact same mailbox defines to the header and he was also calling it a general firmware based pin control mechanism.

If the intention was to make it only control the expander, then a new mailbox API should have been created that was specific for the expander and only accepted number 0 - 7.

pelwell, 2017-03-19:
The way you describe the driver interface sounds less abstracted (less obfuscated) than numbering from 0 to 7 - 
It has exposed an implementation detail of the firmware in a way that it didn't have to. How is that obfuscation?

Do you mean that calling it an expander driver is an obfuscation? Yes, you could use that interface to drive any gpio, but why would you when the ARM can access the onboard controller directly.

Electron752, 2017-03-19:
Actually, Eric already exposed the implementation detail of the firmware last year.  So the cat was already out of the bag to speak.

The API as it's written already leads people down a path were you may not want things to go.  Which is a general pins sharing mechanism.   And the magic numbers that get passed through mailbox raises questions as well.

I'm just saying if the point is to hide things, doing that in public code is not the best way.  I have no problem with things being hidden.  I look at alot of the so called SOC sound drivers that are being submitted where all the interesting things are happening in hidden firmware. I don't think upstream has any problem with that.

Electron752, 2017-03-19:
I don't want to know the answer to this actually, but the mailbox property numbers makes people think the firmware isn't even really parses those numbers.  Perhaps it's just a general peek and poke kind of deal.  Not good for protecting IP.

pelwell, 2017-03-19:
There is no hiding! I can see that you have a bee on your bonnet about the closed-source firmware, but we are not being secretive about the GPIO expander interface.

Electron752, 2017-03-19:
No I'm not Eric, closed source had a purpose.  If it's closed, then maybe it needs to really be closed rather then half closed.

pelwell, 2017-03-19:
So now you are saying we aren't hiding enough? There is a short circuit in your logic - I'm not going to argue with you.

Electron752, 2017-03-19:
Yes, that's sort of what I'm saying. Things are closed need to be closed.  Things that are open need to be open.  Half open doesn't really work because people want to know more about what isn't open.

I agree of having some of the RPI driver open sourced.  Or perhaps even running in a mixed mode.  But as I found out the hard way with the whole vcdbg tool being incompatible with VC4, it's isn't always possible to have it both ways.

Electron752, 2017-03-19:
I'll leave the discussed at this.  Their is a famous quote from the movie "The Wizard of Oz".  "Pay no attention to the man behind the curtain." That's something you really don't want to be telling people some day.

I'll run some tests on the current public firmware on the repo and let you know shortly if it's working.

Thanks.