raspberrypi/firmware #766: Documentation of mailbox API for virtual expander

Electron752, 2017-03-16:
Upstream really needs a expander driver for upstream VC4 testing on the RPI 3.  I don't want to submit the expander driver as is since it's still being worked on.

What I'm thinking of is creating a very simple stripped down RPI 3 specific expander driver that would be upstream only just to unblock progress of VC4 on the RPI 3.  I was going to submit the driver to stagging and keep it in stagging until someone else works on a longer term solution.  The driver would be a read only expander. I'm thinking the longer term solution could include read/write, but I don't have time for that now.

So I was wondering if someone could put some brief documentation of the mailbox API for the expander in a header, in the documentation tree of the downstream linux source, or  in the documentation tree for the RPI devices on github.

I would specifically like to know the definition of the pin offsets as specified in the mailbox api.

Looking at the current driver source on github, I see the pin numbering starts at 128.  I was wondering if this is correct?

6by9, 2017-03-16:
> Looking at the current driver source on github, I see the pin numbering starts at 128. I was wondering if this is correct?

https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts#L1187 for a Pi3
```
            pin@p128 { function = "output"; termination = "no_pulling"; }; // BT_ON
            pin@p129 { function = "output"; termination = "no_pulling"; }; // WL_ON
            pin@p130 { function = "output"; termination = "no_pulling"; }; // ACT_LED
            pin@p131 { function = "output"; termination = "no_pulling"; startup_state = "active"; }; // LAN_RUN
            pin@p132 { function = "input";  termination = "no_pulling"; polarity = "active_low"; }; // Hotplug
            pin@p133 { function = "output"; termination = "no_pulling"; }; // Camera shutdown
            pin@p134 { function = "output"; termination = "no_pulling"; }; // Camera LED
            pin@p135 { function = "input"; termination = "no_pulling"; polarity = "active_low"; }; // Power low
```
Yes you could use it with GPIO numbers 0-53 to read/write/alter the GPIO state of onboard GPIOs too, but the ARM may as well do that directly.

Electron752, 2017-03-16:
Yes, I just noticed that as well when browsing the documentation.  Now I understand where the numbers come from.  I read the documentation on the document tree and I can kind of see where this is going.  But I'm not sure that is the best way to go.

I would think a better way to do this would be to expose model independent numbers that get documented and have the firmware do some mapping internally to the real pin numbers. The firmware could also provide some basic protections such as ensuring that an input only pin doesn't get set to an output. Such as the example I gave on the other thread of frying a $2,000 TV set.

That way it would be easy to add a new RPI module with subtle changes and just swap out the dt-blob or firmware image.

As it is now, the interface is way to generic and exposes too much of the hardware to the end user.

pelwell, 2017-03-16:
The problem with that argument is that RPis are in use by low-level hackers as well as complete novices, potentially for applications not involving HDMI where the pin has been borrowed for some other purpose. Remember that CM has no HDMI connector, so if a custom carrier was using virtually all of the GPIOs for 24-bit DPI it might need to use the HPD pin for something else.

As I said in the other thread, a GPIO driver should implement mechanism, not policy.

Electron752, 2017-03-16:
This problem is very deep. 

I do see a problem though of exposing the raw numbers to the kernel driver in that some it defeats it's own purpose.  Since the kernel level DT may well be distributed independently, you would also have to update the kernel level DT to keep it in sync with the videocore DT.

Electron752, 2017-03-16:
Because now you would be adding a 128 offset, so you would be starting out with out of sync numbers.

Electron752, 2017-03-16:
And if the point is to keep things hidden, why have the simple translation of adding 128.  That sounds to me like this is all going down the path of the fixed address translation that was part of the RPI from the start.  I thought the DT was going to solve that.

If your planning to automatically generate the DT for the kernel in the bootloader, then why not have a 1-1 mapping?

6by9, 2017-03-16:
In the firmware code there are a set of defines
```
{
   GPIO_DEFINE_INTERNAL = 0,
   GPIO_DEFINE_EXTERNAL = 128,
   GPIO_DEFINE_VIRTUAL  = 512,
```
Internal (0-127) goes to the base GPIO driver. External (128-511) goes to the expander driver. Virtual isn't used.
Those defines aren't going to change, so any expander using the firmware will have an offset of 128. It is not dependent on the pin allocation, that is just defining the modes of the pins.

If really desired then that 128 could be written into the exp_gpio device tree bindings (everyone loves writing and documenting bindings), or moved out to a #define.

Electron752, 2017-03-16:
Like I said, if your going to automatically generate the DT for the kernel then why not have a 1-1 mapping? And possible not document any of the numbers.

Electron752, 2017-03-16:
This is going to be very difficult to get into upstream...

Electron752, 2017-03-16:
Like I said, if the purpose is to hide things this doesn't accomplish that.  It took me about 5 seconds to figure out where the 128 was used.   It would only take someone a few seconds of looking at the /proc/device-tree node to figure it out.

Electron752, 2017-03-16:
I'm not sure I want to be part of upstreaming this expander.  Perhaps it's better it if stays in the downstream tree or someone else tries to send it upstream...

Electron752, 2017-03-17:
Guys, I think I over reacted on this without a good reason.  I need to keep a more open mind.

I think this topic may actually be related:
https://github.com/raspberrypi/firmware/issues/760

Basically, my biggest issue at this point is the lack of symmetry in the API. I can image in some cases it may actually be better for the kernel to handle the expander(I understand on the RPI 3 a completely working open source expander driver always exists).    

So I can image that for some RPI it may actually be better for the kernel to handle the expander.  Why couldn't the firmware have an symmetrical system on it's side that kicks the expander requests back into the firmware.  The mailbox system or this driver does not have that symmetry.

Ideally, it should be the boot loader or the user through devices overlays that munges both DTs to determine which side is better for the job.

Electron752, 2017-03-17:
Basically, what I'm saying is is.  Instead of generating the kernel DT of the videocore DT at boot time, perhaps it would make sense for the bootloader to arbitrate between the two and generate both the kernel DT and the videocore DT off the template DT.

And if things were done correctly, I see no reason why their couldn't be some headers or other code that gets shared between the firmware and the kernel(That's really thinking out of the box).

Electron752, 2017-03-17:
So I submitted an enhanced version of this driver to staging.  By default it should be compatible with the original driver so you shouldn't have the need to maintain both.

closed 2017-03-17 by Electron752