raspberrypi/firmware #767: RPI_FIRMWARE_FRAMEBUFFER_GET_GPIOVIRTBUF: What happens if it's called multiple times

Electron752, 2017-03-16:
So I'm trying to understand the mailbox API for the expander, and I'm looking at RPI_FIRMWARE_FRAMEBUFFER_GET_GPIOVIRTBUF.  I'm thinking a bit ahead here but what happens if this mailbox call is called multiple times(say from multiple drivers).   Would it always return the same address each time it's called.

And I have a second question.  Since we know mbox can lockup if called rapidly, I'm wondering if it's possible to use this mailbox call for monitoring the HDMI connections for hotplug.

What I'm thinking is perhaps this is a better firmware API since multiple small, independently written drivers can all be loaded at the same time and share the same buffer. Especially for something like the MMC activity led which could potentially really hit the mbox calls rapidly.

pelwell, 2017-03-16:
Repeated calls will be given a pointer to the same buffer.

> Since we know mbox can lockup if called rapidly

Is that true? I know there is an issue with the heartbeat LED pattern causing a lockup, but without investigation it feels more like a deadlock than a crash.

But yes, the same thread that polls the buffer to set the ACT LED could also poll the HPD line and update the buffer.

Electron752, 2017-03-16:
Interesting...

Since VC4 on upstream is currently blocked, I'm also writing an expander that I plan to submit to upstream.  I was considering building the expander out of libraries.  One for the GPIOVIRBUF approach and one for the new approach.

The idea would be to create a very small framework where someone could easily pump out platform specific expander drivers out of the two competing library blocks.

pelwell, 2017-03-16:
I've not commented on your other post yet, but I think we now have a perfectly good expander driver, so I don't think there is much to be gained except confusion and loss of goodwill be writing another.

Electron752, 2017-03-16:
I don't want to write a third competitor either. I really have better things to do, but VC4 on RPI 3 is blocked and somehow an expander needs to get into upstream.

That's why I thinking of the platform specific driver approach.  I write it quickly and submit it to upstream, but it doesn't get used anywhere else.  Basically, it's throwaway code.

pelwell, 2017-03-16:
Have you written off the downstream driver? It looks eminently upstreamable.

Electron752, 2017-03-16:
Net yet.  It looks rather good but the firmware interface isn't quite stable yet.  I'm also concerned about performance issues if it's used for mmc.

The biggest issue is who would submit it.  Are you guys willing to?

Electron752, 2017-03-16:
Also Eric told me he had trouble getting generic expanders like that into upstream.  You basically have to explain what each of those offsets are connected to.

That was I was thinking in my version I would only actually support the GPIO that I know actual work.  And it would prevent people some doing silly things with setting the HDMI sens pin to an output.

pelwell, 2017-03-16:
We wouldn't use it for the MMC activity LED - that's what the GPIOVIRTBUF driver is for.

I'm sure that between us and Eric we should be able to get it upstreamed, but it's not my driver and I'm not going to promise anything.

Electron752, 2017-03-16:
How will the new driver prevent the user mode interface from setting HDMI sense to output?

pelwell, 2017-03-16:
The upstream on-board GPIO driver places no restrictions on GPIOs, including the HDMI hotplug sense, and I don't think it should - this should be mechanism, not policy.

Electron752, 2017-03-16:
In this case I kind of think it should. The RPI is all about learning and education, so it's going to be used by people that don't know what they are doing.  It seems to me the kernel or firmware should be providing a min amount of protection.

pelwell, 2017-03-16:
Wouldn't using DT to claim the GPIO for the HPD pin prevent a change of direction via the API?

HDMI HPD is hardly mission-critical - people learn from making mistakes - and to repeat myself, the upstream driver doesn't do it. Somebody not a million miles from here once wrote:

> I'm like @msperl, why have a hack like this. Perhaps a better solution is to have a real driver simply claim the clock, even if it's a dummy placeholder driver that doesn't actual do anything. The real driver could be the Raspberry PI firmware driver that talks to mbox, it could be the video driver, or perhaps it should be something new.

Surely that's inconsistent with respect to your attitude re: the GPIO expander?

Electron752, 2017-03-16:
I kind of agree, but I'm not sure of the scope of the physical damage we are talking about he.  Could it perhaps fry a $2,000 TV set?

One option I'm considering is submitting an extremely simple expander that just does one thing. Read the HDMI sense pin.  Then it can be thrown away once some figures out the best solution.

pelwell, 2017-03-16:
Please save your effort - rather use it to help get the existing GPIO expander driver upstreamed.

Electron752, 2017-03-16:
How should I help exactly.  We would also need to upstream the other expander for mmc eventually, and I'm not sure it's quite upstream quality yet.

pelwell, 2017-03-16:
Well, you could squash the commits (the initial one and the followup removing the double addition of the base) and send the initial email to the relevant mailing lists. Both commits are already signed off by Dave.

Electron752, 2017-03-16:
Let me think on that.  

Just one final question.

Would I burn any bridges by submitting a throw away expander that uses the new firmware API but only supports HDMI sense and that's it.  Nothing more.

Current it's hard to even explain to people how I'm doing any kind of graphics or testing that requires a display at all, since the story is that RPI 3 doesn't support video at all in upstream.  The idea would be to at least unblock people.

Electron752, 2017-03-16:
And I'm sure people don't like hearing people say remove VC4 from the built config to get working video.

pelwell, 2017-03-16:
Given that the existing driver already does that, you would effectively be saying that it isn't fit for purpose, which isn't true.

I can't imagine that your driver will be substantially different, given that it has to go through the firmware->mailbox interface, so restricting it to just HPD only reduces the chance of hitting an issue. The drivers will be going into staging, so change is expected, so lets get the ball rolling and we'll push a followup when we find the cause of the lockup (unless of course it is completely in the firmware side).

Electron752, 2017-03-16:
OK, let me think about that.  I really don't have the hardware tools to debug firmware lockups and they are somewhat annoying.

What's the long term story on the mmc expander?

pelwell, 2017-03-16:
You mean the GPIOVIRTBUF driver? TBD.

Electron752, 2017-03-16:
Yes I mean the GPIOVIRTBUF.  Since the mailbox interface is at least going to have performance issue.  And I know that one would be hard to upstream in it's current form.

closed 2017-03-16 by Electron752