raspberrypi/firmware #768: Reset 7" display on disconnect/reconnect

6by9, 2017-03-16:
From https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=173891

At the moment if power is lost to the display, it never comes back.

Seeing as the display board is being polled in order to read the touchscreen, is it worth noting NACKs and flagging the display as offline when received. If it starts ACKing again, send the startup sequence to the board again (but not the DSI peripheral init) so the Toshiba chip is reset. 

@popcornmix thoughts?
I've lost track of whether Eric's code also does any control of the display directly over I2C, but if so then the touchscreen thread must be disabled anyway so the above wouldn't trigger.

popcornmix, 2017-03-16:
My view is that the LCD display is not hot-pluggable, so dropping power is not supported.

You could probably detect if the display is not responding as expected and try to re-initialise it, but if this is a frequent occurrence I suspect there will be many edge cases to consider (e.g. power dropping in the middle of an I2C transaction during re-initialisation).

6by9, 2017-03-16:
I think the request is more that they are pulling the power to it rather than actually unplugging the ribbon cable (agreed that that is a very bad idea).

I guess you would have to handle failure during reinitialisation, otherwise a comms only blip during the reinit would result in the touchscreen comms being happy so not detecting the need for a reset, but the controller still not being in a usable state as some registers aren't set. Hmm.

JamesH65, 2017-12-22:
I'm not convinced we should do any work on this - seems of very little benefit. Closable?

6by9, 2017-12-22:
Leave this open and assigned to me. It's the sort of thing I'll pick up if I get an hour spare.

A relatively simple reset on an I2C NAK at any stage should handle the display power getting cycled, although a little care to stop the thread spinning and hogging the I2C bus would be needed.