raspberrypi/firmware #770: Firmware subsystem in kernel: Using it to protect firmware intellectual property

Electron752, 2017-03-19:
I was thinking today about why the firmware is around at all, and I think the core of it is to protect intellectual property.  The world has secrets and it always will. Somethings are best not in the public.  All other electronics companies have secrets, what's different about the RPI. I've already thought that from the start of this ARM64 stuff.

It turn out that GregK who is the maintainer of the current tree that vc04_services is located in doesn't like platform drivers and trying to put firmware based proxies in the device tree kind of defeats the whole purpose of the DT of which the whole point is to document the hardware. Also the device tree is visible through /proc/device-tree so trying to put protected information in the DT doesn't accomplish anything. 

So what I'm thinking is perhaps a firmware subsystem is needed that works like ACPI on X86.  The DT has a firmware node in it that just contains enough to bootstrap the whole process. Some of these parameters could be hard coded in a static DT and some could be automatically generated by the boot loader. The firmware subsystem would query the firmware about which proxy services to activate on the kernel side.  So the firmware would have absolute power of what proxy services get activated and to what extent. 

I'm thinking this would also be a good solution to marking certain resources on the kernel side as being critical to the firmware.  The firmware proxy could query the firmware about critical resources like clocks and such and mark them as being in use on the kernel side of things by claiming them.

I think this may be also a good solution to get a thermal driver and cpu speed switching drivers upstreamed, since the exact method of doing that probably can't or won't every being in the public domain.

Any thoughts.

Electron752, 2017-03-19:
BTW, 

If this is extended more, perhaps this would be a good way to get a display driver property upstreamed.  It turns out that UEFI already has a display system, so maybe a firmware based display isn't use a bad idea that just has a very simple kernel proxy. Essentially something firmware-kms.

I could also image a day when people may want to use a firmware based clock manager rather then the open source one.  Perhaps the firmware based solution could offer a different set of features then the open source clock manager.

But I still think it should be the end user that decided if they want to run their RPI in closed source mode or open source or perhaps some mix of the two.

feherneoh, 2017-03-19:
Make your own open-source CPU, and we won't need closed-source binaries. Until that be glad, that we only need this much of them. Just look at how Qualcomm does things, and you'll love what we have here.

Electron752, 2017-03-19:
I don't actually like Qualcomm much.  I once bought a Netgear wireless AP that was built on Atheros and the thing was a total piece of crap. I've had nothing but trouble with their garbage.

closed 2017-03-21 by Electron752