raspberrypi/firmware #771: simplefb: What's exactly needed in the DT to get it created/populated by the firmware

Electron752, 2017-03-19:
I'm testing the latest firmware binaries, and I'm finding that I can't get simplefb to be created at all now.  Even if I completely remove all the VC4 nodes from the device tree.  I have no display0 alias in the DT either.  This is an upstream DT with an upstream build.
[temp-dts.txt](https://github.com/raspberrypi/firmware/files/853209/temp-dts.txt)

I attached a dtb to dts conversion of the bcm2837-rpi-3-b.dtb file that I'm pointing the config.txt at.  This is from the build tree.

Electron752, 2017-03-19:
So I got it to work! Everything is awsome!!!

It appears the DT needs to have a node like this:
       soc {
                fb: fb {
                        status = "disabled";
                };
       };

That's good enough to get simplefb to work.  And if I manually disable a few VC4 nodes in the DT, I can get minecraft_pi to work just fine from Raspbian when running a arm64 upstream build based on the current staging tree.

Everything is awsome!!!!

Should I send an announcement on the linux-rpi mail list that this works now.  Some people are going to have a out and out fit that this works without VC4 being enabled.

Electron752, 2017-03-19:
I sent an announcement out.  I've go ahead and close the issue since as far as I'm concerned, it's resolved.

closed 2017-03-19 by Electron752

pelwell, 2017-03-20:
The firmware is expecting /soc/fb to exist and to be disabled, rather than to not exist or be disabled. I'll fix the check, then you can simplify the DT.

Electron752, 2017-03-20:
I'm assuming you are still going to be leaving the videocore graphics components enabled if and only if vc4 is disabled in the DT.  That's what I thought the original intention of this change was.   Ideally I think the end user should be the one that choses between the two.

pelwell, 2017-03-20:
In the original issue () I said:

> The firmware is getting information about the ARM-implemented services from the dtb. It looks for the interrupts claimed by certain non-disabled nodes, and if found it disables the corresponding interrupt handling on the VPU (and in some cases does a bit more than that). In your dtb you have enabled nodes for v3d, hdmi, pixelvalve, and hvs, so the firmware will not handle those interrupts.

This is still the case - individual services are disabled in the firmware when the DT indicates that the ARM is handling the relevant interrupts.

Electron752, 2017-03-20:
Yes I know, I've been putting too much time and effort into this stuff.  Sometimes things get more complicated then they need to be.

Anyway, it sound's like we are on the same page here.  And yes I think it's really cool that upstream has some offical way to get working video until some of the others deal with what needs to be dealt with to get VC4 working.  That's something I'm planing to wash my hands of at this point.  I'll let them deal with it.