raspberrypi/firmware #772: Does firmware support multiple outstanding mbox calls?

Electron752, 2017-03-19:
I'm looking at the upstream mbox driver and I'm noticing that nothing about it forces a single outstanding mbox call at a time.  It's really just the raspberry pi firmware driver that's serializing the calls. So I'm wondering if this in an limitation of the current firmware, or it's just the current raspberrypi firmware driver?

What I'm thinking is that we have an upstream expander submitted that could potentially be banging on the mbox channel constantly.  So perhaps some things should go through an alternative mbox channel.

Also, upstream desperately needs thermal and cpu throttling drivers. I think the issue is that people have been fighting with the correct DT nodes for a very long time.  So I'm thinking of using the existing downstream thermal and cpu throttling drivers as a information reference and writing some upstream replacements.  If downstream actually uses the upstream drivers or not is a orthogonal question.

I'm thinking the upstream thermal and throttling drivers would either call mbox directly through the alternative channel, or they would call through an alternative firmware driver.

popcornmix, 2017-03-19:
There should be a lock on the arm side that is acquired before sending the mailbox message and is released after the response is read. Otherwise there is no guarantee that the response is for the correct message.

Electron752, 2017-03-19:
I'll have to look at the mbox code some more.  I wondering if a solution could be pounded out that could keep things paired.  Is an empty or unused slot available in the mbox messages?  If multiple requests are pending would they be at least responded to in order?

Electron752, 2017-03-19:
Perhaps including a dummy property in the transaction that is a no-op that could be used for context?

popcornmix, 2017-03-19:
There is no point. The firmware executes mailbox requests from a single thread. You won't get any advantage (apart from breaking things) from submitting a second request before the first has responded.

Electron752, 2017-03-19:
OK, just thought I would ask.

Is that something you guys have considered changing?  I'm a bit concerned with the expander driver banging on the mbox channel too much.  And I do see firmware lockups on arm 64 with the cpu throttling driver.  I don't know if a second thread would help much or not.

Anyway, if this isn't something that you've considered I'll close the issue.

popcornmix, 2017-03-19:
The mailbox service will always be single threaded.
vchiq would be used is asynchronous requests are required.

I don't believe the mailbox task itself locks up (we have used it for HEVC decode with thousands of requests a second used by very large numbers of users with libreelec and osmc).

What I suspect you are seeing is that crashing the firmware will result in mailbox service no longer responding. I'm not sure what is causing that, but the upstream clock driver has been observed to disable the GPU's clock which would be an obvious way of crashing the GPU. See: https://github.com/raspberrypi/linux/pull/1846

Electron752, 2017-03-19:
Interesting, I've been batting around the idea of a firmware proxy mirror that's in the kernel.  Could be a very simple driver I think.  The idea would be to claim some of the resources that we know the firmware absolutely can't live without.  I haven't gotten any interest in it though.

One thing it could do is send some test messages through the mbox channel from time to time and log a message in the kernel log if the firmware isn't responding. Or it could force a system reboot.

Electron752, 2017-03-19:
I probably already have most of the information I would need to write such a beast.  It sounds like PLLD is very important.  For a test message I could use something simple like a query on the firmware version or board version info.

If you don't have anything more to add, I'll close the issue shortly.

popcornmix, 2017-03-19:
We provide a simple mailbox utilty in raspbian. This reads firmware version:
```/opt/vc/bin/vcmailbox 0x00000001 4 4 0```

If that doesn't respond I suspect you have killed the mailbox task on GPU (quite possibly the whole GPU). If you can provoke the crash on raspbian with downstream kernel by hammering this command, we can investigate.

If it only occurs with upstream kernel, then it might be interesting to work out what kernel did just before the command stopped responding.

Electron752, 2017-03-19:
OK.  That's a good place for me to get started.

I'll go ahead and close out the issue then.

closed 2017-03-19 by Electron752