raspberrypi/firmware #773: Potential support for a boot.tar

Electron752, 2017-03-21:
I've been noticing some kernel image checkins for bootfs from @Dom.

I was wondering if you guys had any plans for something like a boot.tar facility in the firmware? The idea would be that I do a tar cfv boot.tar /boot of my boot partition files on my RPI and I get something that could be downloaded as a single image off the network or wherever. 

Any chance something like that has been planned or could happen?

It's doesn't need to be a tar file(that's just an example).

Ferroin, 2017-03-21:
Just a general thought, but for this type of thing, CPIO or ar might be better options since they're significantly easier to parse than a tar file (and smaller).  The downside is that pretty much everything has tar, but not all distros ship cpio or ar by default.

Also, while it won't help much for generic usage, it is possible to bundle an initramfs (and I think dtb's also) into a Linux kernel image.  It ends up making the kernel image bigger, but it does reduce the total number of files to be downloaded.

Electron752, 2017-03-21:
That makes sense.  What I was thinking is that perhaps this image could actually be mounted as a read only file system in some way. So I'm not sure what the best format would be.  Tar was just an example, but something optimized for read only access(so the index could be presorted).

I know of two issue with the existing initramfs that I would be attempted to solve here:
1. The initramfs gets tossed after initial boot.  I want it to actually stay around so that I can include the kernel modules as well.  Those may not get loaded until they are actually needed, say the user plugs in a usb device of some kind.

2. The image may need to include DT for multiple RPI machine types, so something in the boot process would have to select which of those DTs to load after identifying the machine model.

Electron752, 2017-03-21:
Sorry, I know double posting is bad...

This file system could have firmware stuff inside it some how.  How this would get packed I don't know.  Maybe that part should be visable to the kernel, or maybe it shouldn't. Perhaps the firmware part would be it's own inner filesystem with it's own format.

But if you take this to it's logic conclusion, the big package should be in a format sort of compatible with existing firmware infrastructure in that enough of it is in the correct places to bootstrap the whole RPI.

JamesH65, 2017-12-22:
I'm not aware of any plans to change this, and manpower demands would dictate it being very low priority anyway. I'm inclined to close this issue. Thoughts?

Electron752, 2017-12-22:
I say just close the issue.  It's low priority for me as well...

closed 2017-12-22 by Electron752