raspberrypi/firmware #774: Sync issue in files and folder after reboot

djrocktarun, 2017-03-22:
I am using RPi 2 and Linux raspberrypi 3.18.7-v7+ #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015 armv7l GNU/Linux.

Actually i am facing this issue since 1 month. Files and folder not able to sync after reboot.

Also sudo sync not helping in that case.

Here what i did

- Do changes in some file or create a new file or folder.

- Then run **sudo sync**

- Then run **sudo reboot**

After restart no changes retain in RPi.

Kindly help me if any one also facing this issue and find the solution.

XECDesign, 2017-03-22:
If you do the same thing on another computer, do you see the same problem? I had a card which started ignoring writes suddenly. It appeared to work, since the data was cached by the OS, but a reboot or a remount would ensure all the 'written' data was discarded, no matter what computer or OS I used.

djrocktarun, 2017-03-22:
Yes i tried using with different Raspberry pi 2 as well with the same SD card. Issue still present.

Yes i know the cached data will be discarded after reboot or a remount. But i am not able to understand

- Why any files changes still be in cache for so long period of time ?

- If file changes are in cache, How can i confirm that all data is permanently write to the memory before system restart. Because **sudo sync** is not working in my case

XECDesign, 2017-03-22:
I would guess since sync only flushes the write buffers, it doesn't give the OS a reason to drop the read cache.

In any case, I think it's a dead card and you will see the same behaviour on a desktop PC or in a camera, It's very unlikely that this is a raspberry pi firmware issue.

Ferroin, 2017-03-22:
>Why any files changes still be in cache for so long period of time ?

Because there is no reason for the OS to not keep them cached if it has enough room.  If it's active data, it's likely to get used again soon, so there's no point in purging it from the cache if there's room for it.

>If file changes are in cache, How can i confirm that all data is permanently write to the memory before system restart. Because sudo sync is not working in my case

If your hardware is lying about writes completing, then there's not really anything you can do except get better hardware.  You might try checking the card with [f3](https://github.com/AltraMayor/f3), that will tell you whether or not the card has just gone bad, or if it's actually a counterfeit (which can display the same kind of behavior as this after a while), but keep in mind that that need to wipe the card to test it properly with f3.

Ferroin, 2017-03-22:
Oh, I forgot to mention, the quick way to see for certain if the data is being read from cache and the writes are just getting dropped by the card is to call sync then do:
`echo 2 > /proc/sys/vm/drop_caches`
as root and verify if any files you had changed got reverted as if you had rebooted.

Make sure if you do that that you have almost nothing running, if the SD card is dropping writes, then that will likely cause most things you have running to lose their mind.

djrocktarun, 2017-03-22:
I tried the steps you suggest above @Ferroin

- Change a file using nano editor

- Then run **sudo sync**

- Then run as superuser  **echo 2 > /proc/sys/vm/drop_caches**

After that i open the file and found no data in file 

file content are look like **@@@@@@@@@@@@@@@@@**

After **sudo reboot** my SD card not able to boot.

When i tried to check the boot logs using FTDI 

the logs stops after this line

Uncompressing Linux... done, booting the kernel.

Any suggestion to avoid this sync issue ?

Ferroin, 2017-03-22:
It's definitely a hardware issue given what you've said (although I have to admit I didn't expect that to nuke the SD card, that should have just dropped everything from the cache), so the only thing that's going to fix it is getting a new SD card.

feherneoh, 2017-03-25:
I'm interested in the brand of your SD card. My Samsung Evo card bought with my first Pi does not write any data, and still reports write succes. Samsung eMMC chips in lots of phones I know do exactly the same too. Is it by any chance a Samsung card too?

djrocktarun, 2017-03-27:
No its not Samsung Sd card. My Sd card is of Strontium class 6 oemid : 0X5446

JamesH65, 2017-04-20:
@djrocktarun Did a new SD card fix the issue? Is so please close. If not please add any more information.

djrocktarun, 2017-04-26:
Yes i changed my SD card manufacturer and new firmware version (4.4.5), 
new SD card not showing this issue. 

I am closing this issue, as its like to be an hardware issue not the firmware issue.

closed 2017-04-26 by djrocktarun