raspberrypi/firmware #775: Possibility of killing/shutdowning down firmware during boot sequence

Electron752, 2017-03-22:
I opened an issue on the bare metal forum on raspberrypi.org.

I'm exploring an interesting option for an open firmware.  Instead of writing everything from bootcode.bin upward, perhaps a better option is a "nuke firmware" ability one the base kernel image is loaded.

The idea would be to have an option for the firmware to shutdown at some point after the core kernel image loads.  This could include reseting parts of the VPU and such.  The ARMs would completely take over the system(including the infamous expander on the RPI 3), and reload any necessary VC bits from the lib/firmware directory such as is typically done on other platforms such as X86.

This may be a very good way to solve the expander issue on the RPI 3, and I think would be completely accepted on upstream linux.

pelwell, 2017-03-22:
You can't reset the VPU without resetting everything else, include the ARM, DRAM controller etc., which would land you in a boot loop, and blocks tend not to have individual resets.

Electron752, 2017-03-22:
What about the possible solution of having the VPU jump into on new code which is loaded from /lib/firmware and then nuking out what the bootloader loaded.

The idea is that at some point the old stuff gets turned off and new stuff gets loaded.

Electron752, 2017-03-22:
I sent the link to Eric.  He may be able to contribute to this as well.

This would be a very good way to unblock VC4 on upstream.  The idea being that upstream could do things the open source way and use VC4 to it's full extent and downstream could continue with the existing methods for maximum compatibility.

Electron752, 2017-03-23:
I was just thinking of this other simple alternative.  What if the existing boot loader loaded a new file say firmware.bin during the boot sequence and the existing firmware jumped to it just before the arm jumps into the kernel. This new file if it existed would completely take over the functions of the existing firmware that was loaded during the boot sequence.

The new firmware could be built with the free/open source VC toolchain and the source for the new firmware could be published.  I would actually be interested in writing such as beast.

popcornmix, 2017-03-23:
You have three choices for running code on the vpu.
You can replace bootcode.bin.
You can replace start.elf.
You can submit vpu jobs through the mailbox interface.

No plans to add additional mechanisms.

If you are interested in open source firmware, you may want to look here:
https://github.com/christinaa/rpi-open-firmware
(I haven't studied this, I am just aware that it exists)

Electron752, 2017-03-23:
I've actually been looking at redoing the armstub8 code and attaching it to a custom Linux kernel.  It may actually be good enough to just take over the existing firmware after it's already been loaded from the arm by moving the VPU exception/interrupt vectors and forcing an exception.  Essentually, "killing" the existing firmware.

I'm think that may actually be alot easier then starting from bootcode.bin or needed to use something undocumented such as how start.elf hooks into bootcode.bin.

Electron752, 2017-03-23:
@popcornmix 

After thinking about it perhaps it's best that things just stay the way they are.  With the existing firmware are such.  I'm not sure I really have the time to go done the path this is going in.

closed 2017-03-23 by Electron752