raspberrypi/firmware #776: EDID can be read for RPi3B, but not CM3L+CMIO3

JinShil, 2017-03-23:
If using a RPi3B with an HDMI display and the `vc4-kms-v3d` overlay, we can read the EDID of our display with either `xrand --props` or `xxd /sys/devices/platform/soc/soc:gpu/drm/card0/card0-HDMI-A-1/edid` as illustrated below:

```
pi@raspberrypi:~ $ xrandr --props
Screen 0: minimum 320 x 200, current 800 x 480, maximum 2048 x 2048
HDMI-0 connected 800x480+0+0 (normal left inverted right x axis y axis) 108mm x 68mm
        EDID:
                00ffffffffffff000481040001000000
                01110103800f0a000a00000000000000
                00000000000001010101010101010101
                010101010101800c208030e02d102830
                d3006c44000000180000001000000000
                00000000000000000000000000100000
                00000000000000000000000000000010
                00000000000000000000000000000017
   800x480       65.68*+
```

```
pi@raspberrypi:~ $ xxd /sys/devices/platform/soc/soc:gpu/drm/card0/card0-HDMI-A-1/edid
0000000: 00ff ffff ffff ff00 0481 0400 0100 0000  ................
0000010: 0111 0103 800f 0a00 0a00 0000 0000 0000  ................
0000020: 0000 0000 0000 0101 0101 0101 0101 0101  ................
0000030: 0101 0101 0101 800c 2080 30e0 2d10 2830  ........ .0.-.(0
0000040: d300 6c44 0000 0018 0000 0010 0000 0000  ..lD............
0000050: 0000 0000 0000 0000 0000 0000 0010 0000  ................
0000060: 0000 0000 0000 0000 0000 0000 0000 0010  ................
0000070: 0000 0000 0000 0000 0000 0000 0000 0017  ................
```

However, if we use the same LCD and configuration (`vc4-kms-v3d` overlay)  with the CM3L+CMIO, the EDID cannot be read, and our LCD will not show anything.  If we VNC to the device, we can see the desktop, but it is the wrong resolution, and the mouse and keyboard input are really slow and laggy.  I'm assuming this is because the CM3L firmware is not properly reading the EDID from the display.  

We have not made any changes to the device tree.

* Is this difference between the RPi3 and CM3L by design?  
* If it is by design, is there any configuration or boot parameters we can use to workaround the problem?  
* If it is not by design, can we expect an upcoming fix?

pelwell, 2017-03-23:
I can't think of any obvious reason why CM3 should behave any differently than Pi3 in this respect. The HDMI interface uses I2C2 on dedicated pins so it can't be a GPIO mapping issue, and to my software engineer eyes the two circuits, aside from the CM3 version being split across the CM and the CMIO board, look the same.

I'm interested to see whether the firmware can read the EDID. Can you run `sudo vcdbg log msg >& vcdbg.txt` and upload the output somewhere?

JinShil, 2017-03-23:
I've uploaded the output for both Pi3B and CM3L here: https://gist.github.com/JinShil/c1bb305ff0cdb5b44ee103ee2b185cf4

ghollingworth, 2017-03-24:
Is this because the hotplug detect is not working and therefore it's
defaulting to 720P HDMI?

Electron752, 2017-03-25:
I'm using kms on my PI3 to test ARM64 and I see that hotplug still appears to be broken on my RPI 3 as well.   The VC4 driver tries to switch from simplefb and it hangs, just like what happens if hotplug is broken.

I noticed some things that were a bit funky with Eric's GPIO handling code, and I sent an email but nobody paid attention.

I'm building out of this tree so are the drivers should be available.

6by9, 2017-03-26:
Whilst testing https://github.com/raspberrypi/linux/pull/1879 I had HDMI hot plug working fine with dtoverlay=vc4-kms-v3d
You do need the matching firmware update (509beaa / 13th March or more recent). Firmware from 15th Feb - 13th March supported the relevant commands, but had an inversion that on review it was decided was inappropriate.

JinShil, 2017-03-29:
I can confirm that the problem disappears after applying https://github.com/raspberrypi/firmware/commit/509beaab0e102cf6decf209922669700c9ac5576.  Thank you for your valuable feedback. Closing.

closed 2017-03-29 by JinShil