raspberrypi/firmware #780: Official 7" display + console blanking prevents any other dispmanx layer from being visible

dividuum, 2017-03-27:
I got a bug report for info-beamer and have been able to reproduce the problem with the latest Raspbian release (userland/firmware/kernel) and omxplayer. The setup looks like this: Only the official 7" display is connected (so no HDMI is active). Booting up the system shows the console output on the 7" display. Video playback with omxplayer works as expected until the console blanks (I think after 5 minutes in the default settings?). Once the console goes blank, the omxplayer output is gone as well. It only returns when pressing a key on the connected keyboard as this forces the console to unblank.

I'm not entirely sure if how console blanking works under the hood, but the following commands result in the same problem without having to wait (Commands taken from [this bug report](https://github.com/raspberrypi/firmware/issues/407#issuecomment-88908289)):

```
root@raspberrypi:~# echo 1 > /sys/class/graphics/fb0/blank
# output gone
root@raspberrypi:~# echo 0 > /sys/class/graphics/fb0/blank
# output back
```

Oddly enough, everything works as expected (omxplayer keeps being visible) the first time I run those commands. But omxplayer disappears when repeating them.

Any idea what might cause this? Everything works as expected when using HDMI instead.

```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux

root@raspberrypi:~# vcgencmd version
Mar  3 2017 13:45:32 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
```

popcornmix, 2017-03-28:
So, I've booted with LCD display attached (no HDMI).
I run omxplayer from one ssh.
I connect with another ssh as root and run:
```
echo 1 > /sys/class/graphics/fb0/blank
sleep 1
echo 0 > /sys/class/graphics/fb0/blank
sleep 1
echo 1 > /sys/class/graphics/fb0/blank
sleep 1
echo 0 > /sys/class/graphics/fb0/blank
```
Video is still on screen and playing. Are you seeing something different?

dividuum, 2017-03-29:
Sorry for the delay: When I do exactly that, the video disappears twice the second time I run them. The first time, the video only disappears once.

I thought I've used a Pi3, but I just noticed that I'm using the Model B+ (Revision 0010).  I've now connected a Pi3 (Revision a02082) and it's exactly the same: the video disappears once the first time and twice the second time I run those commands.

`/boot/config.txt` is completely commented out, except for the `gpu_mem=192` line. The Pi is powered by the official Pi power supply.

popcornmix, 2017-03-31:
I've had a look. There seem to be two unrelated issues here.

The first is the first blanking command is ignored.
This is because ``bd->fb_bl_on[node]`` is initialised to false so this code is skipped:
http://lxr.free-electrons.com/source/drivers/video/backlight/backlight.c#L67

but once you unblank, everything is back in sync. I think we can
initialise this to true in rpi_backlight.c to describe the fact the backlight is initially on.

The second is whether the default framebuffer blank command should just remove the framebuffer console or if it should power off the backlight/display.
This occurs by default after a few minutes and can be annoying when using omxplayer or other apps that don't render to the framebuffer console.

When using hdmi, we only remove the framebuffer console on a blank request. However for users who want to stop outputting over hdmi (for DPMS) the config.txt option ``display_blanking`` will enable this.

We're still debating what the best behaviour here is, but unifying the HDMI, composite, dsi, dpi displays is desirable.

For now, entering ``setterm -blank 0`` in the console before starting omxplayer should stop the blanking. If you are running X you may want:
```
xset -dpms
xset s off
```

dividuum, 2017-03-31:
First of all: Thanks for looking into this.

> We're still debating what the best behaviour here is, but unifying the HDMI, composite, dsi, dpi displays is desirable.

I agree. I'd prefer using the current behaviour for HDMI connected screens, as that's how it worked in the past on the majority of Pis (I guess?). And having a way to disable the framebuffer so it doesn't require a 
 HVS layer is pretty useful. Of course having the option to additionally disable output (on HDMI)/backlight (on the 7" display) sounds like a nice plus for people that need the "deep" blank. I haven't heard of `display_blanking` before. Good to know.

popcornmix, 2017-03-31:
> I haven't heard of display_blanking before. Good to know.

Actually it is currently called ``hdmi_blanking``. ``display_blanking`` would be a better name.

JamesH65, 2017-12-22:
@dividuum @popcornmix Did anything come of this? Sounds like a real fault.

dividuum, 2017-12-22:
I can't reproduce the problem with a newer kernel/firmware. Looks solved to me. Closing for now.

closed 2017-12-22 by dividuum