raspberrypi/firmware #781: Limited memory on RPi2 after kernel/firmware upgrade

ebugakov, 2017-03-27:
Hi!

I've tried to upgrade my RPi2's kernel and firmware to latest available

raspberrypi-bootloader_1.20170303-1_armhf.deb
raspberrypi-kernel_1.20170303-1_armhf.deb

and got the same issue which was described here - #725.

root@pi:~# uname -a; vcgencmd version; vcgencmd get_mem arm; vcgencmd get_mem gpu
**Linux pi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux**
Mar  3 2017 13:42:34 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
**arm=128M
gpu=128M**

Switching back to previous version of firmware and kernel fixed the problem:

raspberrypi-bootloader_1.20170215-1_armhf.deb
raspberrypi-kernel_1.20170215-1_armhf.deb  

root@pi:~# uname -a; vcgencmd version; vcgencmd get_mem arm; vcgencmd get_mem gpu
**Linux pi 4.4.48-v7+ #964 SMP Mon Feb 13 16:57:51 GMT 2017 armv7l GNU/Linux**
Feb 15 2017 17:14:25
Copyright (c) 2012 Broadcom
version b53088af3eae5f17d809e37578f1cb40dd555ed8 (clean) (release)
**arm=944M
gpu=64M** 

As far as I understand, 20170303 is the latest version at the moment, so the question - do we have workaround available at the moment?

Thanks in advance!

popcornmix, 2017-03-27:
Are you running NOOBS or a direct raspbian image?
Do you have any custom settings in config.txt (e.g. start_file or fixup_file)?
Can you report output of:
```
md5sum /boot/start.elf
md5sum /boot/fixup.dat
```

ebugakov, 2017-03-27:
this is direct raspbian image.

root@pi:~# md5sum /boot/start.elf                                                                                                                                                03d13d6e525b0e540690cbdd05708be6  /boot/start.elf
root@pi:~# md5sum /boot/fixup.dat
4ee3c1e3edaf9fadf5dbed1953867dcd  /boot/fixup.dat 

I have only one uncommented parameter in config.txt:
dtparam=audio=on
and I really don't have a clue why :)

popcornmix, 2017-03-27:
This is what you should have:
```
pi@domnfs:~ $ vcgencmd version
Mar  3 2017 13:42:34 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
pi@domnfs:~ $ md5sum /boot/start.elf /boot/fixup.dat 
1cc21227f2d66d520966d95719fa5810  /boot/start.elf
5886c96cfcf6939a1c761694f3294124  /boot/fixup.dat
```
Can you report output of
```
cat /proc/partitions
mount
```

ebugakov, 2017-03-28:
root@pi:~# vcgencmd version
Feb 15 2017 17:14:25
Copyright (c) 2012 Broadcom
version b53088af3eae5f17d809e37578f1cb40dd555ed8 (clean) (release) 

root@pi:~# cat /proc/partitions
major minor  #blocks  name

   1        0       4096 ram0
   1        1       4096 ram1
   1        2       4096 ram2
   1        3       4096 ram3
   1        4       4096 ram4
   1        5       4096 ram5
   1        6       4096 ram6
   1        7       4096 ram7
   1        8       4096 ram8
   1        9       4096 ram9
   1       10       4096 ram10
   1       11       4096 ram11
   1       12       4096 ram12
   1       13       4096 ram13
   1       14       4096 ram14
   1       15       4096 ram15
 179        0   15670272 mmcblk0
 179        1      64512 mmcblk0p1
 179        2   15601664 mmcblk0p2 

How can I update these (start and fixup) files? As far as I understand, they are not port or firmware or kernel packages?

ebugakov, 2017-03-28:
I've found these files on github:
https://github.com/raspberrypi/firmware/tree/master/boot
but they are referring to kernel 4.9.17.

Can I use these files with the kernel version 4.4.50-v7 (2017030)?

ebugakov, 2017-03-28:
So, right now, I've upgraded packages (kernel and firmware) to latest available for my distributive and they are absolutely the same checksum which you mentioned above:

5886c96cfcf6939a1c761694f3294124  ./fixup.dat
1cc21227f2d66d520966d95719fa5810  ./start.elf
039c01c34596f5db8130cd41af2c4b5b  ./kernel.img
d7e30bb56d92781505306116307b2f15  ./kernel7.img 

And in this particular configuration I've got the issue with memory which was reported initially:

root@pi:~# uname -a; vcgencmd version; vcgencmd get_mem arm; vcgencmd get_mem gpu
Linux pi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
Mar 3 2017 13:42:34
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
**arm=128M
gpu=128M**

popcornmix, 2017-03-28:
This feels like an sdcard problem.
The fact you have had the same "vcgencmd version" result with two different md5sums on start.elf means data is not being read/written consistently to sdcard.

I'd suggest backing up anything you want to keep from sdcard.
Try a reinstall. Ideally on a different sdcard.
I'd suggest testing sdcard with something like [H2testw](https://www.raymond.cc/blog/test-and-detect-fake-or-counterfeit-usb-flash-drives-bought-from-ebay-with-h2testw/). If it fails bin the sdcard. If it passes then you could try using the sdcard again, but I'd treat it with suspicion (i.e. make sure you back up regularly).

ebugakov, 2017-03-28:
Hmm, I suppose we have misunderstanding here.

With the latest versions of kernel and firmware (md5sums are the same, which you mentioned before) - I face with this memory issue.
Rolling back to previous versions on kernel and firmware (with the different from the latest files checksums of course) solve this problem.

It seems the issue was introduced in kernel and/or firmware files version 20170303 (4.4.50). 
Rolling back to version 20170215 (4.4.48) fix the issue.

Where is the sd-card problem here? :)

popcornmix, 2017-03-28:
In your first post you reported:
Linux pi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
Mar 3 2017 13:42:34
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
03d13d6e525b0e540690cbdd05708be6 /boot/start.elf
```
In your recent post you reported:
```
Linux pi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
Mar 3 2017 13:42:34
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
1cc21227f2d66d520966d95719fa5810 ./start.elf
```
i.e. the same version of the firmware but with two different md5sums. That shows sdcard corruption must have occurred. Therefore sdcard is not reliable.

ebugakov, 2017-03-28:
Ok, now I got this. The information regarding kernel version (9ae30f71c7ef4239e9d5b56346c0842f3ef56736) was captured during issue was observed.
Later I've downgraded kernel and firmware to previous version (20170215) and md5sums of these files were reported.

So, what do we have on the table:

1. Issue observed (20170303):
kernel 9ae30f71c7ef4239e9d5b56346c0842f3ef56736
5886c96cfcf6939a1c761694f3294124  ./fixup.dat
1cc21227f2d66d520966d95719fa5810  ./start.elf

2. No memory issues observed (20170215):
kernel b53088af3eae5f17d809e37578f1cb40dd555ed8
03d13d6e525b0e540690cbdd05708be6 ./start.elf
4ee3c1e3edaf9fadf5dbed1953867dcd ./fixup.dat

Are we on the same page now?

popcornmix, 2017-03-28:
> Are we on the same page now?

Ah so md5sums reported in post 3 were from older "Mon Feb 13" firmware? That wasn't clear. Okay, understood.

popcornmix, 2017-03-28:
Can you post your config.txt file here, just to be sure there is nothing funny there.

ebugakov, 2017-03-28:
root@pi:/boot# grep -v "#" /boot/config.txt
dtparam=audio=on

that's all.

popcornmix, 2017-03-28:
The 20170303 firmware has been the stable version available for any updaters over the last couple of weeks. As this is the only report it is almost certainly something unique to your install, rather than a general problem.

Do you have a spare sdcard? I'd be quite interested in result of making a backup copy of the sdcard (e.g. with win32diskimager). First does the copy behave any differently? If not then I have some tests that can be tried.

ebugakov, 2017-03-30:
Hi!

reinstalled debian system on new flash card and it seems current kernel version works fine. Probably it was indeed a sd-card issue and you can close this discussion.

Thanks for help!

closed 2017-03-30 by popcornmix