raspberrypi/firmware #782: Model B+ wont boot with invalid total_mem value.

paul-1, 2017-03-28:
With firmware pulled today (March 25th commit), it appears as issue #717 has cropped back up

A model B+ fails to boot with this setting 
[PI1]
total_mem=513

This works, but how to distinguish between 256M and 512M models.
[PI1]
total_mem=512

Does a pi with less than 1024 memory benefit from this setting, or is total_mem=1024 providing extra memory for only rpi2B and rpi3B

popcornmix, 2017-03-28:
Can you test this: https://drive.google.com/uc?id=0B-6zmEDJwxZES1lhRmJhUWZkcmc&export=download
It should fix the ``total_mem=513`` case.

The Pi SoC has a 30-bit address space (top 2 bits are used for caching aliases).
This means you can access a phyisical address range of 1G.
The ARM accesses this memory through a coarse MMU (64 pages or 16M each).
But the ARM also need to access the GPU's peripherals which makes the highest page unusable.
As a default the top 16M of actual sdram is unused.

However recent updates try to put GPU memory that the ARM doesn't need to access in the top page. We have provided a way for vcdbg to get GPU debug info from top page using DMA.

As this is still undergoing testing you must enable it manually with total_mem=1024.

As Pi0/Pi1 only used half the pages of the MMU anyway, they never had a missing 16M or sdram, so setting total_mem=512 will provide no benefit.

paul-1, 2017-03-29:
Yes,  That took care of the issue.
Also set total_mem=1024 in an [ALL] section. All of my easy access boards booted fine
Zero, Zero-w, B+, rpi2B rev1.2

And thanks for explanation.  If I am using this on a headless box, will the extra memory be usable to the normal kernel?

popcornmix, 2017-03-29:
Yes, you will have 16M of extra usable memory.
If gpu_mem is not specified you will get 12M extra gpu_mem and 4M extra arm mem.
The extra gpu_mem allows improved chromium hardware video performance.

However you are free to set gpu_mem=64 and you will get 16M extra on the arm side.
Just boot up and run ``free -h`` and you should see an increase in free memory.

paul-1, 2017-03-29:
Yes, I was only seeing 4MB extra.  Thanks.

popcornmix, 2017-03-30:
Fix should be in rpi-update firmware now. Okay to close?

paul-1, 2017-03-30:
Did a quick test with the version from github.  Everything seems fine.

closed 2017-03-30 by paul-1