raspberrypi/firmware #784: BCM43438 Firmware location

YOUN3SS, 2017-03-29:
Hi,

Can someone tell me where is the firmware for 43438 Broadcom chip?
In /lib/firmware/brcm I found only for 43430 (.bin)

Can you also tell me if it uses other generic firmwares (since it is Wi-Fi and Bluetooth), like brcmfmac or brcm80211?

Thank you.

pelwell, 2017-03-29:
Although the device is the 43438, it identifies itself as 43430 and the firmware is /lib/firmware/brcm/brcmfmac43430-sdio.{bin,txt} .

YOUN3SS, 2017-03-29:
Thanks for the answer @pelwell.

When I remove these files,  I don't see the error saying that firmware is not loaded.
(dmesg | grep bluetooth)

That's my doubt that Raspbian uses other paths for this chip...
http://youness.net/raspberry-pi/bluetooth-headset-raspberry-pi-3-ad2p-hsp

pelwell, 2017-03-29:
The Bluetooth firmware is completely separate - that is `/lib/firmware/BCM43430A1.hcd`.

YOUN3SS, 2017-03-29:
I didn't see this file in Raspbian Jessie by default, is it installed separately also?
Thanks.

pelwell, 2017-03-29:
What do you mean by "by default"? If you install Raspbian Jessie (even Jessie Lite), that file is there.

YOUN3SS, 2017-03-29:
Ok I checked in wrong folder.
We can close the issue. Thank you again.

closed 2017-03-29 by YOUN3SS