raspberrypi/firmware #785: Raspberrypi zero W support

telphan, 2017-03-30:
The yocto BSP layer for raspberrypi pulls the firmware as a tar archive that gets generated from the tags in the firmware repo. 
Now we are trying to fully support the raspberrypi zero W and we mainly need this commit (https://github.com/raspberrypi/firmware/commit/f5446ba0b3bbd4241bf0ad35a2c9f299f5c447d5).
When are you guys planning on adding this commit to the stable branch and tagging?

lurch, 2017-03-30:
Does https://github.com/raspberrypi/firmware/tree/1.20170303 not include the required changes? AIUI that's the firmware included in the latest Raspbian release, which AFAIK does fully support the Pi Zero W.
I suspect https://github.com/raspberrypi/firmware/commit/f092ca8e39b137277a63ce4dfb56514e51bdf404 might be the commit of interest?

agherzan, 2017-04-03:
PIng @telphan

Ruffio, 2017-04-09:
@telphan what do you say to @lurch question?

telphan, 2017-04-13:
@lurch Sorry for the late response, I was on a small holiday. It seems like you are right, that is the commit I am looking for. Thank you very much for your help.

closed 2017-04-13 by telphan