raspberrypi/firmware #787: Decoding of 1080x1270 videos does not seem to work

carlonluca, 2017-04-01:
I tried to play a 1080x1270 video and it seems the output is corrupted. I then tried to scale to 1080x1270 with ffmpeg a regular video file and the result is similar. It seems like decoding of 1080x1270 videos is not working. The same happens with my app using OpenMAX and omxplayer. Can someone confirm? Is this a resolution not supported by the Pi maybe?

6by9, 2017-04-02:
What firmware version are you running?
https://github.com/raspberrypi/firmware/commit/2ef9cb622f0587e9a0fc4826a9a2d7d34c092d95 (21 March 2017) included a fix for decoding videos with heights between 1264 and 1366 (if I've remembered the numbers correctly).

carlonluca, 2017-04-07:
Updating the firmware seems to fix the issue. I think you can close this, thanks.

closed 2017-04-07 by popcornmix