raspberrypi/firmware #788: Could not set interface wlan0 flags (UP): Input/output error

djrocktarun, 2017-04-04:
I am using RPi 3, and my kernel version is 

Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

Here are my dmesg

[  874.959311] w1_master_driver w1_bus_master1: Family 0 for 00.480000000000.84 is not registered.
[  903.602943] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  903.602956] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  905.602989] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  905.603011] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  907.602983] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  907.602997] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  909.603019] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  909.603033] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  911.603052] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  915.603154] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-52)
[  917.603179] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[  919.603213] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  919.603234] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  921.603217] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  921.603230] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  922.290152] w1_master_driver w1_bus_master1: Family 0 for 00.c80000000000.08 is not registered.
[  923.603245] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  927.603340] brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-52)
[  929.603370] brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
[  931.603415] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  931.603437] brcmfmac: brcmf_cfg80211_get_tx_power: error (-52)
[  933.603402] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110

**ifconfig** result not showing wlan0

eth0      Link encap:Ethernet  HWaddr b8:27:eb:20:49:a6  
          inet addr:10.42.0.194  Bcast:10.42.0.255  Mask:255.255.255.0
          inet6 addr: fe80::7a55:6ca:d998:6cbf/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:313 errors:0 dropped:0 overruns:0 frame:0
          TX packets:250 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:31187 (30.4 KiB)  TX bytes:38635 (37.7 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:224 errors:0 dropped:0 overruns:0 frame:0
          TX packets:224 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:18672 (18.2 KiB)  TX bytes:18672 (18.2 KiB)

**ifconfig -a** shows the wlan0

eth0      Link encap:Ethernet  HWaddr b8:27:eb:20:49:a6  
          inet addr:10.42.0.194  Bcast:10.42.0.255  Mask:255.255.255.0
          inet6 addr: fe80::7a55:6ca:d998:6cbf/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:352 errors:0 dropped:0 overruns:0 frame:0
          TX packets:278 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:34838 (34.0 KiB)  TX bytes:43405 (42.3 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:236 errors:0 dropped:0 overruns:0 frame:0
          TX packets:236 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:19689 (19.2 KiB)  TX bytes:19689 (19.2 KiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.12.226  P-t-P:10.8.12.227  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:75:1c:f3  
          inet6 addr: fe80::1607:9b6d:f19:c4a3/64 Scope:Link
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B

pelwell, 2017-04-04:
1) What do you have in config.txt?

2) What does `vcgencmd get_throttled` report?

djrocktarun, 2017-04-04:
**Here is my config.txt**

# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

dtoverlay=w1-gpio,gpiopin=17

**Output of vcgencmd get_throttled**

throttled=0x50000

pelwell, 2017-04-04:
> throttled=0x50000

This says that your Pi has suffered an under-voltage situation since booting, and tells me that your power supply is inadequate. This is most likely the cause of your problems.

The Pi3 requires a 2.5A supply - potentially more if you are powering external devices.

JamesH65, 2017-04-20:
@djrocktarun Did a better power supply fix the issue? If so please close.

djrocktarun, 2017-06-29:
Issue not resolved even after changing the power supply.

I am closing this issue, as no one able to reproduce

Thanks for help !!

closed 2017-06-29 by djrocktarun

eliliam, 2023-05-30:
I'm having the same issue(along with the same throttled value), when running `sudo ifconfig wlan0 up` I am met with `SIOCSIFFLAGS: Input/output error`. This is odd because you can clearly see the wlan0 interface in the output of `ip a`:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:e0:5f:11 brd ff:ff:ff:ff:ff:ff
    inet 10.1.1.5/22 brd 10.1.3.255 scope global dynamic noprefixroute eth0
       valid_lft 59037sec preferred_lft 45476sec
    inet6 fe80::6d81:1d3c:83cf:eeab/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:b5:0a:44 brd ff:ff:ff:ff:ff:ff
```

pelwell, 2023-06-05:
Same problem, same solution - check your power supply and cable. An under-voltage while booting can easily upset the WiFi controller.