raspberrypi/firmware #789: OpenMAX resizer slow

atic-atac, 2017-04-05:
Hi all,

Is anyone doing any work on improving the performance of the openmax resizer component ?

I'm feeding YUV frames in and spitting out RGB, but fullscreen frames can take 60ms through the resizer, and waiting for the buffer fill done to fire.

6by9, 2017-04-05:
No, no work being done on it, and no intention of doing any.

Some details of your pipeline would be useful if you want any suggestions on optimisations. Are you working in stripes or full frames? Actually resizing or just format converting? Tunneled or not? Other load on the VPU/VC4?

If you're only after YUV to RGB with no crop then you could look at "vc.ril.isp" as a component. That is wrapping the hardware ISP block to do the actual pixel processing (resize is just VPU vector code). It was written with MMAL in mind instead of IL, but the basic functionality should work.
The one thing I do NOT expect to work is tunneling on the input port, but it should work on the output port. It does require full frames on both input and output (nSliceHeight >= nFrameHeight).

That component is being used under MMAL for video_decode in Chrome, taking 1080P YUV in and producing fullscreen RGBA32 out. Overall it is pretty close to 30fps.
Keep the hardware fully utilised and it should be able to munch >150MPix/s. The issue in Chrome not quite hitting 30fps sometimes are down to shifting data around more than raw conversion power.

Is there a particular reason you're using IL instead of MMAL? There are a number of optimisations available in MMAL that aren't in IL that can make a big difference. The main one I'm thinking of is the option to zero copy buffers by mapping the VPU buffers into ARM memory. That saves a fair chunk of memory bandwidth and DMA time.

atic-atac, 2017-04-05:
Tunneled from video_decode, just format convert, no resize. No other load on the GPU.

MMAL, looks like an approach, but last time I looked at it, it wouldn't spit out frames as soon as the the frame was complete like OpenMAX using the ENDOFFRAME. Has that changed ?

6by9, 2017-04-05:
> Tunneled from video_decode, just format convert, no resize. No other load on the GPU.

As a simple test, try video_splitter instead. IIRC resize will still run the resize algorithm even if 1:1. video_splitter will just do the conversion to the specified output format with no resize.

> MMAL, looks like an approach, but last time I looked at it, it wouldn't spit out frames as soon as the frame was complete like OpenMAX using the ENDOFFRAME. Has that changed ?

MMAL and IL both use the same underlying components. MMAL was written to remove a load of the headache inducing state transition and buffer allocation stuff that made IL client code so tricky to get right. The behaviour of the components when presented with data and flags should be identical.

atic-atac, 2017-04-05:
Reading video_splitter and it mentions that the input/output ports will have the same format. No format conversion is done.

http://home.nouwen.name/RaspberryPi/documentation/ilcomponents/video_splitter.html

Is that false ?

6by9, 2017-04-05:
> Is that false ?

Yes. Looking at that documentation it is badly out of date :-(

video_splitter accepts proprietary or regular comms on input and output.
Proprietary in/regular out has always supported format conversion, mainly as you don't know the image format on the proprietary comms.
Regular in/regular out has supported format conversion since June 2016. https://github.com/Hexxeh/rpi-firmware/commit/4468c4c063c3d034821dfef4e4da00f21ea72645 "firmware: video_splitter: support format conversion on buffers in and out"

The only issue you may hit is that video_splitter does not queue up incoming proprietary comms buffers, so if the conversion and buffer handling is much slower than video_decode is supplying frames then you may get some drops. You may view that as an appropriate place to drop things when running behind, or it may not be acceptable. You can force ports to use regular comms by using ```OMX_IndexParamBrcmDisableProprietaryTunnels```, and then buffers will get queued up.

BTW Probably better to refer to the HTML files at https://github.com/raspberrypi/firmware/tree/master/documentation/ilcomponents - those should get updated with every release that changes the relevant headers, although they seem not to have been touched for a while whilst I know the headers have been. I'll nudge Dom to see if he'll add it to the release scripts.

atic-atac, 2017-04-05:
O.k. is there any example of video_splitter, seems as though it's not moving the the execution state here.

6by9, 2017-04-05:
> O.k. is there any example of video_splitter

No public ones that I'm aware of in IL.
I do have the source to an old test app that is setting up
```
/*                                                          --- video_render
 *                                                         /
 *   read_media -- video_decode -- [scheduler] -- vsplitter --- video_render
 *                                /                       | \
 *                               /                        |  --- video_render
 *                       clock--                          \
 *                                                          --- image_fx -- video_render
 */
```
read_media is not available on Pi but you can feed the raw stream in from the ARM. 
Beyond that there is nothing unusual in the implementation.

>  seems as though it's not moving the the execution state here.

One of the reasons I hate IL - it was always pretty counter-intuitive as to what was wrong.
You have disabled all the splitter output ports you aren't using I assume? That would block a state transition as it would be waiting for buffers to be allocated.

atic-atac, 2017-04-05:
Yes, one thing I was checking was that. I guess I'll try the opposite and allocate buffers for all ports regardless of disable/enable.

atic-atac, 2017-04-05:
Nope, didn't work either. Mmmm.

6by9, 2017-04-05:
Can you post your code somewhere? I'm happy to switch to email if it needs to remain private (dave.stevenson@raspberrypi.org).
Alternatively "vcgencmd set_logging level=0xC0", run your code, then "sudo vcdbg log msg" should log all IL level stuff. It is a circular log buffer, so you may loose early stuff.

atic-atac, 2017-04-05:
Even though the log suggests it goes to executing, something is broken here as I don't get it as the userlevel....

1046116.423: video_decode:6:RIL:resolution changing
1046116.446: video_decode:6:cb:callback parameter changed 131,0
1046117.059: video_decode:6:SendCommand(2,131,3e7a9974) Disable 131
1046117.077: video_decode:6:sending cmd 2 to 131
1046117.117: video_decode:6:starting command 2
1046117.139: video_decode:6:processingCommand=0
1046117.305: video_splitter:7:SendCommand(2,250,3e7a9974) Disable 250
1046117.328: video_splitter:7:sending cmd 2 to 250
1046117.357: video_splitter:7:starting command 2
1046117.377: vx_set_port_state(250): 1
1046117.398: vx_get_parameter(0x2000001)
1046117.417: vx_get_parameter(0x2000001)
1046117.602: video_splitter:7:SendCommand(2,251,3e7a9974) Disable 251
1046117.626: video_splitter:7:sending cmd 2 to 251
1046117.654: video_splitter:7:starting command 2
1046117.670: vx_set_port_state(251): 1
1046117.690: vx_get_parameter(0x2000001)
1046117.710: vx_get_parameter(0x2000001)
1046117.859: video_splitter:7:SendCommand(2,252,3e7a9974) Disable 252
1046117.877: video_splitter:7:sending cmd 2 to 252
1046117.903: video_splitter:7:starting command 2
1046117.925: vx_set_port_state(252): 1
1046117.939: vx_get_parameter(0x2000001)
1046117.960: vx_get_parameter(0x2000001)
1046118.106: video_splitter:7:SendCommand(2,253,3e7a9974) Disable 253
1046118.129: video_splitter:7:sending cmd 2 to 253
1046118.160: video_splitter:7:starting command 2
1046118.180: vx_set_port_state(253): 1
1046118.196: vx_get_parameter(0x2000001)
1046118.220: vx_get_parameter(0x2000001)
1046118.374: video_splitter:7:SendCommand(2,254,3e7a9974) Disable 254
1046118.398: video_splitter:7:sending cmd 2 to 254
1046118.433: video_splitter:7:starting command 2
1046118.457: vx_set_port_state(254): 1
1046118.478: vx_get_parameter(0x2000001)
1046118.500: vx_get_parameter(0x2000001)
1046118.535: video_splitter:7:completed state transition to OMX_StateIdle
1046118.560: video_splitter:7:processingCommand=0
1046119.073: video_decode:6:GetParameter(7f000003)
1046119.108: video_decode:6:ComponentTunnelRequest(131,setup)
1046119.140: video_decode:6:CTR: Output port 131 is not supplier
1046119.323: video_splitter:7:GetParameter(7f000003)
1046119.350: video_splitter:7:ComponentTunnelRequest(250,setup)
1046119.372: video_decode:6:GetParameter(2000001)
1046119.396: video_decode:6:RIL:GetParameter(3e7b5984)
1046119.422: vx_get_parameter(0x2000001)
1046119.449: vx_verify_format(local=250,remote=131)
1046119.472: video_decode:6:SetParameter(7f000009)
1046119.495: video_decode:6:RIL:SetParameter(7f000009)
1046119.523: video_decode:6:GetParameter(2000001)
1046119.545: video_decode:6:RIL:GetParameter(3e7b5984)
1046119.568: video_decode:6:SetParameter(2000001)
1046119.589: video_decode:6:RIL:SetParameter(2000001)
1046119.613: video_decode:6:RIL:PortDef, port 131
1046119.633: video_decode:6:RIL: set output format 20
1046119.662: vx_set_parameter(0x2000001)
1046119.720: video_splitter:7:cb:callback parameter changed 251,0
1046119.754: video_splitter:7:cb:callback parameter changed 252,0
1046119.781: video_splitter:7:cb:callback parameter changed 253,0
1046119.810: video_splitter:7:cb:callback parameter changed 254,0
1046119.841: vx_get_parameter(0x2000001)
1046119.868: video_splitter:7: buffer count 1 to 0
1046119.887: video_decode:6:SetParameter(2000002)
1046119.912: video_decode:6:SP: Output port 131 is not supplier
1046119.940: video_splitter:7:CTR: Input port 250 is supplier
1046120.413: video_splitter:7:GetParameter(2000001)
1046120.428: vx_get_parameter(0x2000001)
1046120.607: video_splitter:7:SendCommand(0,1,3e7a9974) State to loaded
[... 42 lines cut ...]
1046196.423: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046196.465: decode: display frame 3e7bfd10 T=0 sts=0
1046196.490: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046216.423: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046216.466: decode: display frame 3e7bfd10 T=0 sts=0
1046216.490: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046236.424: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046236.466: decode: display frame 3e7bfd10 T=0 sts=0
1046236.491: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046256.426: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046256.469: decode: display frame 3e7bfd10 T=0 sts=0
1046256.495: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046276.652: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046276.692: decode: display frame 3e7bfd10 T=0 sts=0
1046276.716: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046296.425: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046296.466: decode: display frame 3e7bfd10 T=0 sts=0
1046296.489: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046316.425: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046316.468: decode: display frame 3e7bfd10 T=0 sts=0
1046316.492: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046336.428: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0
1046336.469: decode: display frame 3e7bfd10 T=0 sts=0
1046336.494: video_decode:6:RIL: disp_cb fail 3e7bfd10/0/0x10
1046356.428: video_decode:6:RIL:output dec_frames_full:3e7b6da0 out:0

atic-atac, 2017-04-06:
After fiddling around with things I've got the splitter to work, it is indeed slightly faster than the resizer too.

Thanks!

6by9, 2017-04-06:
Could you give a few details?
Something just wrong in the setup, or something that appears to be a bug?
Any numbers on "slightly faster"?

Please close the issue when you're happy.

atic-atac, 2017-04-06:
Just correcting the appropriate states. Seems that the resize is happy without going into a Loaded state, yet the splitter needs to be told.

We're at about 40ms now instead of 60ms.

Ruffio, 2017-04-06:
I wouldn't call a reduction of 33% slightly ;-)

atic-atac, 2017-04-06:
I can do CSC on the CPU with NEON faster than 40ms on the GPU.

Ruffio, 2017-04-06:
"Everything else equal" how much faster?

6by9, 2017-04-06:
> We're at about 40ms now instead of 60ms.

Still a touch higher than I'm expecting, but it is a slightly weird conversion required and the copy back to the ARM gets painful for 1920x1080 RGB(A) at 6 or 8 MB per frame - the dma for that has been seen to be as high as 18ms if the ARM side buffer is very fragmented or other higher priority SDRAM activities. That is where MMAL wins by being able to use VCSM and map the GPU buffer into your ARM process.

Converting all the hello_pi apps to MMAL is on my list of tasks. I'll see if I can get hello_video done quickly and see what performance gains we can make there.

atic-atac, 2017-04-06:
| "Everything else equal" how much faster?

13ms on the CPU.

pelwell, 2017-04-06:
13ms taken or 13ms saved?

atic-atac, 2017-04-06:
taken.

6by9, 2017-04-06:
In many ways not surprising as the ARM is running at 1.2GHz compared to the VPU at 250MHz (although probably with wider vector operations).

The ISP should munch through a 1080P YUV to RGB format conversion in about 8-10ms, but you will need to be using MMAL and zero copy to actually realise the full benefit.

JamesH65, 2017-12-22:
@atic-atac Can this be closed, it does appear you have the required answers?

closed 2017-12-22 by atic-atac