raspberrypi/firmware #790: Multiple regressions introduced by bf5201e9682bf36370bc31d26b37fd4d84e1cfca firmware

SylvainGarrigues, 2017-04-05:
Hello,

The bf5201e9682bf36370bc31d26b37fd4d84e1cfca breaks a few things, and **does not allow BSD images to boot anymore**. I tested with previous commits up to 4e156d5 and can 100% confirm all these regressions were introduced in bf5201e9682bf36370bc31d26b37fd4d84e1cfca.

I believe changes in the DTB provided to the non-Linux kernel has changed in non-backward compatible way.

As I said, since the MMC driver no longer recognize the SD card, FreeBSD images do not book anymore (see point 2 below).

Here are the changes I'm not happy with:

1/ MAC address is no longer patched in the DTB tree:
```
                                ethernet {
                                        compatible = "net,ethernet", "usb,device";
                                        reg = <0x1>;
                                        mac-address = [00 00 00 00 00 00]; ==> commit 4e156d5 reported [b8 27 eb 3d xx yy] there
```
It's fine to pass the ethernet address a kernel parameter, but why no longer patch the DTB? The firmware did it for me, and since I don't use u-boot, I don't have any more MAC address in the ethernet tree. This was a serious regression to me.  

2/ Much more important, the SDHCI / MMC driver does report any longer that a miniSD card is inserted in the slot. What can possibly cause this? This is also a serious regression, FreeBSD images don't boot anymore now since this commit. Is it related to the change below?
```
                dma {
                        compatible = "broadcom,bcm2835-dma", "broadcom,bcm2708-dma";
                        reg = <0x7000 0x1000 0xe05000 0x1000>;
                        interrupts = <0x18 0x19 0x1a 0x1b 0x1c 0x1d 0x1e 0x1f 0x20 0x21 0x22 0x23 0x24>;
                        interrupt-parent = <0x2>;
                        broadcom,channels = <0x0>; ==> commit 4e156d5 reported 0x7f35 there
                };
```

3/ The following part also completely disappeared from the DTB.
```
                vc_mem {
                        reg = <0x3dc00000 0x3f000000 0xc0000000>;
                };
```

pelwell, 2017-04-05:
With the switch to 4.9 the rpi tree is getting ever closer to upstream Linux. In particular, we are using the upstream dts files, with a few patches. This has resulted in some changes and tidying up of the device tree and command line processing performed by the firmware.

0) In general, the "/axi" tree is deprecated - almost everything now exists under "/soc" or "/chosen".

1) The MAC address is now passed in the recommended way - via the `local-mac-address` property of the node pointed to by the `ethernet` alias. See https://github.com/raspberrypi/firmware/issues/613

2) I have no idea why the FreeBSD driver doesn't work any more. The DMA channels mask is fixed in the DTB as 0x7f35. We won't be changing it.

3) See 0. The memory layout and mapping is described by /soc/ranges and /soc/dma-ranges.

SylvainGarrigues, 2017-04-05:
Thank you very much Phil for your quick feedback.

The  ```broadcom,channels``` property of the ```dma``` node is now set to ```0x0``` in the DTB the FreeBSD kernel gets as of this commit, while it indeed used to be 0x7f35 as per previous commit.

I am gonna try to check if that is what makes sdhci fail to probe the mmc interface. Also got your point about /axi vs. /soc, thanks.

pelwell, 2017-04-05:
> The broadcom,channels property of the dma node is now set to 0x0 in the DTB the FreeBSD kernel gets as of this commit, while it indeed used to be 0x7f35 as per previous commit.

My guess is that in the DTB used by FreeBSD the property is initialised to 0, and the firmware used to overwrite it but no longer does.

pelwell, 2017-04-05:
By the way, OS's other than Linux aren't really on my radar, so I didn't take them into account when making these firmware changes. I'm happy to restore some of the removed functionality if there is a need for it, but I want to have a reason other than "we've always done it that way".

SylvainGarrigues, 2017-04-05:
Thanks again @pelwell for such quick feedback. I have reported and started to discuss the issue on FreeBSD's arm bug tracker, trying to figure out the proper way to fix this.

Right now, I can see the DMA driver is looking at the 0x735 mask in the ```brcm,dma-channel-mask ``` property first, then in ```broadcom,channels```. It gets 0x0 starting from bf5201e9682bf36370bc31d26b37fd4d84e1cfca like you guessed.

What would you recommend to FreeBSD folks to get this mask (apart for parsing command line) now that it is no longer filled by the firmware in the axi / dma property?

pelwell, 2017-04-05:
I think they should use the `brcm,dma-channel-mask` property of the dma controller node, for which the downstream dts creates a `dma` alias (although the location shouldn't matter if the driver is passed a DT node pointer when it is probed). However, since the mask is now essentially fixed it would seem acceptable as an interim measure to just patch their current location to read `0x7f35`.

SylvainGarrigues, 2017-04-05:
Thanks, will patch FreeBSD's own rpi2.dts to 0x7f35 for now.

Just as a side note, using https://github.com/raspberrypi/firmware/blob/master/boot/bcm2709-rpi-2-b.dtb:
```
# dtc -I dtb bcm2709-rpi-2-b.dtb | grep brcm,dma-channel-mask
			brcm,dma-channel-mask = <0x7f34>;
```

But using https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/bcm283x.dtsi:
```
# grep brcm,dma-channel-mask bcm283x.dtsi
			brcm,dma-channel-mask = <0x7f35>;
```
Is that normal?

pelwell, 2017-04-05:
Yes it is. Upstream uses 0x7f35, but the downstream bcm2708_fb driver uses channel 0 for blitting via a dedicated driver - the Linux DMA framework doesn't have the concept of memory-to-memory transfers. Unless FreeBSD has a similar need to reserve channel 0 (which is the most capable DMA channel) then you should use 0x7f35.

SylvainGarrigues, 2017-04-06:
Hardcoding ```0x7f35``` as the DMA channel mask in the DTS fixed FreeBSD, thanks @pelwell.

For MAC address, replacing ```mac-address``` placeholder with ```local-mac-address``` in the DTS also fixed the retrieval of the MAC address from the FDT.

I'm seeing discrepancies over and there about these 2 properties :-) [U-boot](http://git.denx.de/?p=u-boot.git;a=blob_plain;f=common/fdt_support.c;hb=HEAD) fills both. The RPI firmware now fills only ```local-mac-address```. The [upstream linux tree](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/drivers/of/of_net.c?id=refs/tags/v4.10.8) checks for ```mac-address``` first then for ```local-mac-addr```. The [FreeBSD driver](https://github.com/freebsd/freebsd/blob/master/sys/dev/usb/net/if_smsc.c) does the opposite.

closed 2017-04-06 by SylvainGarrigues

pelwell, 2017-04-06:
Regarding the mac address properties, I agree that there seems to be some confusion.

I'm treating the ePAPR v1.1 specification as gospel (but I'm open to correction). It says:

> Property: local-mac-address
> ...
> Description: Specifies MAC address that was assigned to the network device described by the node containing this property.
> ...
> Property: mac-address
> ...
> Description: Specifies the MAC address that was last used by the boot program. This property should be used in cases where the MAC address assigned to the device by the boot program is different from the local-mac-address property. This property shall be used only if the value differs from local-mac-address property value.

That is pretty clear to me, and is supported by the Linux implementation, i.e. use mac-address if present, otherwise local-mac-address.