raspberrypi/firmware #791: Security issue with Broadcom firmware?

goobering, 2017-04-05:
Is the security issue detailed [here](https://googleprojectzero.blogspot.co.uk/2017/04/over-air-exploiting-broadcoms-wi-fi_4.html) and [here](https://bugs.chromium.org/p/project-zero/issues/detail?id=1047) relevant to the BCM43438, and if so is that something that should be addressed here? It looks like a swathe of Broadcom WiFi chips are affected by vulnerabilities that leave them open to attacks via TDLS.

pelwell, 2017-04-05:
We will update users when we know more ourselves.

goobering, 2017-04-27:
Given that I only understand about one word in ten of the writeups I linked to, I appreciate it may take some time to roll out a fix if it's needed. It's not ideal to keep potentially insecure devices running sensitive applications up and running in the meantime, when it would be relatively trivial just to switch them to wired networks. Are you in a position yet to confirm whether the BCM43438 is affected by this exploit? Would you recommend turning off WiFi for more sensitive applications?

hellyberry, 2017-07-06:
This issue is oft high importance!

lategoodbye, 2017-08-09:
Please look at https://github.com/raspberrypi/linux/issues/1342#issuecomment-321221748 for a new firmware prerelease, which should address 2 issues.

dividuum, 2017-08-17:
The linked [preview firmware](https://drive.google.com/file/d/0B_P-i4u-SLBXUnAwLU9RSm9EMGc/view?usp=sharing) in the issue linked above seems to have ended up in the Stretch release. The package _firmware-brcm80211_ includes the file _brcmfmac43430-sdio.bin_ (md5sum 5f520a38ab4e943bfa1ba102f80fb2a0) from the above preview. Can we assume this preview works well enough? Will there be a final version?

pelwell, 2017-08-17:
It isn't "preview firmware", it was a release candidate for testing. The testing didn't show up any new problems so it is the new release version. There will be a final version if and when we stop finding problems or Cypress stops fixing them.

dividuum, 2017-08-17:
Thanks for clarifying and fixing the problem! I guess this issue can be closed then?

pelwell, 2017-08-17:
We normally leave that to the original reporter, but it will probably be closed during the next issue cull.

By the way, your [Info Beamer multi-monitor video wall](https://www.raspberrypi.org/blog/info-beamer/) is very cool.

JamesH65, 2017-12-22:
Please update to the latest kernel which contain a fix for this issue.

Closing.

closed 2017-12-22 by JamesH65