raspberrypi/firmware #792: Specifying hdmi_group=1 in config.txt prevents later use of hdmi_timings

Phlamethrower, 2017-04-05:
When using `vcgencmd hdmi_timings` and tvservice to define and use custom modes at runtime, it appears that there's a bug in the firmware where specifying hdmi_group=1 in config.txt causes the tvservice mode switch to fail. With hdmi_group=1 in config.txt it's possible to use tvservice to switch to any of the standard group 1/2 modes, but when trying to switch to group 2/mode 87 (i.e. the custom mode) my monitor loses signal and the response portion of the tvservice VCHIQ request doesn't appear to be received. With hdmi_group=2 in config.txt, or with the default hdmi_group left undefined, everything appears to work OK.

Although there's the obvious workaround of not specifying hdmi_group=1 when you're planning on using custom mode timings, this issue can still catch out people who upgrade their OS without knowing to check the contents of config.txt (e.g. recently the RISC OS video driver was updated to make use of the custom mode timings feature).

I've tested this with the current firmware and firmware from a few months ago.

popcornmix, 2017-04-06:
Just to be sure I'm looking at the same issue, can you list any hdmi config.txt settings
and list the tvservice commands you run to provoke the issue.

Phlamethrower, 2017-04-06:
Starting from a clean install of the latest stable raspbian image:

```
vcgencmd hdmi_timings 640 0 16 64 120 480 0 1 3 16 0 0 0 75 0 31500000 1
tvservice -e "DMT 87"
fbset -depth 8 && fbset -depth 16
```

With the default config.txt (i.e. with just the `dtparam=audio=on` line active) the above commands will work. If you enable the `hdmi_group=1` line (you don't even need to specify a mode number) then the hdmi_timings won't "stick" - the output from vcgencmd will always show the previous timings (I didn't notice this before since the RISC OS video driver just throws away the response message). Since mode 87 starts as undefined this then explains why the following tvservice command causes the monitor to lose signal.

`vcgencmd hdmi_cvt` appears to work with hdmi_group=1, so it looks like the problem is local to hdmi_timings.

The same behaviour is seen after rpi-update.

popcornmix, 2017-04-07:
Okay, I think the issue is that you can create a custom CEA mode or a custom DMT mode.
By default the custom mode will be DMT 87.
If you use ``hdmi_group=1`` the custom mode will be CEA 65.
Try:
```
vcgencmd hdmi_timings 640 0 16 64 120 480 0 1 3 16 0 0 0 75 0 31500000 1
tvservice -e "CEA 65"
tvservice  -s
```

Phlamethrower, 2017-04-07:
Yes, CEA 65 works - thanks. Do you know if there's any practical difference between CEA and DMT when using hdmi_timings, or does the hdmi_group setting only affect the list the mode ends up in? I've had a read through the [original hdmi_timings forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=24679) (which does mention CEA mode 65, doh) but can't spot any obvious mentions of differences between the two.

popcornmix, 2017-05-04:
I believe the timings are identical.
The CEA modes default to hdmi drive (rather than dvi) and using the blanking interval for audio and AVI infoframe signalling.
But I believe the actual pixel timings are the same.

Phlamethrower, 2017-05-04:
OK, thanks for the help!

closed 2017-05-04 by Phlamethrower